arendst/Tasmota #2272: adding support for Ws2801 LEDs

Rockiii87, 2018-03-28:
Hi Theo. 
I'm using Tasmota on several Sonoffs, but i need it also on my NodeMCU to light up a stripe of WS2801 and get input from a DHT11 Sensor at the same time.

Would it be possible to get it in the code for the NodeMCU's?

Jason2866, 2018-03-28:
Why does it not work?
Tasmota works the same way as on a sonoff device.

Rockiii87, 2018-03-28:
yes, but not with the WS2801, as i think. Because the WS2801 is a 4 Pin Stripe.

Jason2866, 2018-03-28:
Your description is missunderstanding for that case.
Just write in headline feature request for WS201 and nothing else...

arendst, 2018-03-29:
The WS2801 interface is fundamentally different from WS2812. The WS2801 is an older version led strip than WS2812 so I do not expect much demand for it.

I currently have no plans to implement unless someone comes up with a pull request.

Jonn3y, 2020-12-30:
Would love to see this being done some time. NeoPixelBus supports the WS2801 since Jan. 2020 (v2.5.6)
https://github.com/Makuna/NeoPixelBus/issues/324

Jason2866, 2020-12-30:
@Jonn3y  As Theo wrote if someone provides a PR he will probably merge