arendst/Tasmota #2273: Cannot add AM2301 sensor to Sonoff Basic

kajmaj, 2018-03-28:
Hello,
I am trying to add sensor AM2301 to Sonoff Basic, but if is selected for GPIO14, there is no temperature and humidity displayed on web interface. It is the same with DTH11, the only sensor working (what I've tried)  is sensor DS18x20
It is the same for 5.12.0 and 5.12.0.i as well.
Any advice please?

reloxx13, 2018-03-28:
status 0?
connection of sensor?

> Does it mean that Sonoff Basic does not support these sensors?
nope, they work.

kajmaj, 2018-03-29:
Still no connection.
Whatt is even worst - the same behavior is on all my sonofs after upgrade to "i"
There must be sthg wrong with my 5.12.0i build even if  I did everything as several times before - disabled Domoticz, changed wifi in `user_config.h`, save+build and done. And upgraded via OTA.
I will try to compile it again and will see.

arendst, 2018-03-29:
No problems what so ever:
```
00:00:00 Project sonoff Ring2 (Topic ring2, Fallback DVES_83BB10, GroupTopic sonoffs) Version 5.12.0i-2_3_0
00:00:00 WIF: Connecting to AP2 indebuurt2 in mode 11N as ring2...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on ring2.local with IP address 192.168.2.223
00:00:06 MQT: Attempting connection...
00:00:06 MQT: Connected
00:00:06 MQT: tele/ring2/LWT = Online (retained)
00:00:06 MQT: cmnd/ring2/POWER = 
00:00:06 MQT: tele/ring2/INFO1 = {"Module":"Generic","Version":"5.12.0i","FallbackTopic":"DVES_83BB10","GroupTopic":"sonoffs"}
00:00:06 MQT: tele/ring2/INFO2 = {"WebServerMode":"Admin","Hostname":"ring2","IPAddress":"192.168.2.223"}
00:00:06 MQT: tele/ring2/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: homeassistant/switch/ring2_1/config =  (retained)
00:00:06 MQT: homeassistant/light/ring2_1/config =  (retained)
09:52:06 MQT: stat/ring2/POWER = {"POWER":"OFF"}
09:52:06 MQT: stat/ring2/POWER = OFF
09:52:06 MQT: domoticz/in = {"idx":172,"nvalue":0,"svalue":"19"}
09:52:13 MQT: tele/ring2/STATE = {"Time":"2018-03-29T09:52:13","Uptime":"0T00:00:14","Vcc":2.731,"POWER":"OFF","Dimmer":19,"Color":"14301A","HSBColor":"133,58,19","Channel":[8,19,10],"Scheme":2,"Width":1,"Fade":"ON","Speed":1,"LedTable":"ON","Wifi":{"AP":2,"SSId":"indebuurt2","RSSI":90,"APMac":"00:22:6B:FE:8E:20"}}
09:52:13 MQT: tele/ring2/SENSOR = {"Time":"2018-03-29T09:52:13","AM2301-12":{"Temperature":20.8,"Humidity":45.1},"SI7021-14":{"Temperature":20.6,"Humidity":43.6},"TempUnit":"C"}
```

kajmaj, 2018-03-29:
Theo, thank you for assurance.
I know that it is supported. But it simply does not work and I've tried everything I am able to do to have it back.
It is like "not active" . And those 2 lines "temperature" and "humidity" on the web interface are not there.
If I select DS18x20, line with temperature is immediately  there. 
The same it is wirh Sonoff TH16 where I used to have connected AM2301, without any problems. Now it works with DS18x20 only.
It started after upgrade to ""i" and it remains even after return back to 5.12.0 OTA from the web.
Is there anything what you can advice  to have them back?

arendst, 2018-03-29:
The problem is that this is a Tasmota Issue log and Not a user's forum.

There are currently over 1000 issues here which should have been handled by a forum as they have no relation with Tasmota software errors or bugs.

Sorry, you are the first to get hit.

kajmaj, 2018-03-29:
Somebody must be hit first 😪
Hmm. It means no forum, no advice.

arendst, 2018-03-29:
Cheer up.

There is this: https://groups.google.com/d/forum/sonoffusers for forum and more user experience.
And soon there is this: : http://forum.digital-pig.com/

ornolio, 2018-04-20:
could you confirm the AM230x has a pull-up resistor (4.7 to 10kohm)?