arendst/Tasmota #2274: New version of Sonoff B1

timosntr, 2018-03-28:
Hardware: Sonoff B1 "V1.0.3"

Today I got my second pair of Sonoff B1, I opened it and got this:
[image]
As you can see the layout is a bit different, but it is also possible to flash Tasmota to it. The problem is, that if I configure the module as an Sonoff B1, it will do nothing if I want it to power on.
I got two of the new ones and both have this problem.

Maybe someone else got the new ones and can help? Otherwise it would be a good future improvement in Tasmota to support those.

arendst, 2018-03-29:
Have a look at module AiLight and see if that works.

timosntr, 2018-03-29:
Thank you very much, that worked! Mysterious, from the outside it looked like an Sonoff B1 and i ordered it as one. The board itself looks like a AiLight, as i found out on google.

closed 2018-03-29 by timosntr