arendst/Tasmota #2275: New model SONOFF T1 US 1-2-3 gang

cirolaferrara, 2018-03-28:
New model SONOFF T1 US 1-2-3 gang is out

emontnemery, 2018-03-28:
@cirolaferrara It's a wiki, if you can create issue you can also edit the wiki..

emontnemery, 2018-03-28:
Sorry, I put a comment in wrong issue!

PiOver2, 2018-05-18:
I have flashed T1 R2 US TOUCH BOARD V1.0 successfully by touching pad P3 while pressing switch S1 to enter flash mode.

dcazarez05, 2018-05-25:
Hi @PiOver2 Did you mean you can install another firmware like Tasmota or something like that?
If yes, Can you share how you do it please?

gonzunigad, 2018-06-19:
It also worked for me touching P3 while pressing S1. 
https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK-and-T1-EU/

Connect the serial adapter to J3 
Module - Serial USB
- 3v3 -> 3v3
- Rx -> Tx
- Tx -> Rx
- GND -> GND

Then, once connected and powered up, touch pad 3 (the one next to to esp8265) and the, without releasing pad 3, press Switch 1 (above). 

Once I was on flashing mode, I compiled tasmota on arduino using these settings (Generic esp8265. Not 8266)

[image]
I think you can just upload the sketch, but what I always do, is to program it using [esptool](https://github.com/espressif/esptool). 

[image]

That worked for me.

MYLE-01, 2018-06-20:
I have did it also 

What did got the bin file only used espeasy to flash it then used termite  to send all info I needed and order 3 more next time i'm try the backlog thing

zomars, 2018-07-05:
Have anyone had luck using SonOTA?

cirolaferrara, 2018-07-13:
Yes. I flashed it with sonota

zomars, 2018-07-16:
@cirolaferrara what firmware did you have when you flashed? I'm on 1.8 and it's stuck before 
FinalPhase.

cirolaferrara, 2018-07-17:
I'm sorry. I don't rembember. Anyway remember to disable firewall

closed 2018-08-19 by cirolaferrara