arendst/Tasmota #2276: sonoff pow ds18b20 on gpio 14

andyd999, 2018-03-28:
I have a pow v2.0.  do these have the issue with mains voltage on the ground serial header? i wish to connect a ds18b20 sensor to it. using the board where is gpio 14

Also i wish to use the pow to check power usage for my aquarium heater, or am i better having the pow in parallel with a th10?

thanks

curzon01, 2018-03-28:
> I have a pow v2.0. do these have the issue with mains voltage on the ground serial header?

see [Wiki Sonoff Pow - Special Pow Attention](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Pow#%EF%B8%8F%EF%B8%8Fspecial-pow-attention---%EF%B8%8F%EF%B8%8F)

For this reason if you select module type `08 Sonoff Pow` within Tasmota there is no possibility to select additonal sensors.
Suggestion is to get the temperature with an extra device.

Frogmore42, 2018-03-28:
All of the ones that measure power usually connect the low voltage ground directly to one of high voltage lines.  While you can hook up a sensor to the esp8266, there is no way to it safely.  I find it much safer easier and better to us a separate wemos D1 mini or nodemcu to handle an sensors I need.  I have found the connectors on th10 and th16 are very low quality, so much prefer taking the time to do it right myself.  The sonoffs are great when you need to switch or measure power.

AlexTransit, 2018-03-28:
[https://github.com/arendst/Sonoff-Tasmota/issues/1941](url)

andyd999, 2018-03-29:
Thaanks guys. i am new to tasmota and sonoffs but in principle will tasmota work on a d1 mini? is that even possible? if so that would be a better solution for me and how many sensors can i attach to 1 d1 mini?

Frogmore42, 2018-03-31:
Yes, Tasmota works on a D1 mini.  The code supports 8 DS18B20 sensors all on the same pin as I recall.  Take a look at the wiki and other issues and you will find a lot of information that might be useful.

closed 2018-06-20 by stale[bot]