arendst/Tasmota #2280: NTP.ORG and MEZ SUMMER TIME (UTC+2)

on-off-on, 2018-03-28:
Hi,
could someone give me a hint what I have to do that the european summer /normaltime change automatically?  The NTP Server NTP.ORG runs, but my S20 has still the winter time utc+1.  Restarts dosn´t solve the issue and the systemtime of my s20 is actual utc+1 instead +2.

RaymondMouthaan, 2018-03-28:
Command Timezone 99 helpt me to set the correct time in the Netherlands.

on-off-on, 2018-03-28:
00:00:00 Projekt sonoff Sonoff (Topic sonoff, Fallback DVES_1288E1, GroupTopic sonoffs) Version 5.12.0i-2_4_0
00:00:00 WIF: verbinden mit AP1 XYZ in Modus 11N wie sonoff-1111...
00:00:03 WIF: verbunden
00:00:03 DNS: initialisiert
00:00:03 HTP: Web-Server aktiv bei sonoff-2273.local mit IP-Adresse 192.168.100.99
00:00:03 RSL: INFO1 = {"Module":"S20 Socket","Version":"5.12.0i","FallbackTopic":"DVES_1288E1","GroupTopic":"sonoffs"}
00:00:03 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-1111","IPAddress":"192.168.100.99"}
00:00:03 RSL: INFO3 = {"RestartReason":"Power on"}
00:00:05 RSL: RESULT = {"POWER":"ON"}
00:00:05 RSL: POWER = ON
20:50:13 RSL: STATE = {"Time":"2018-03-28T20:50:13","Uptime":"0T00:00:12","Vcc":3.436,"POWER":"ON","Wifi":{"AP":1,"SSId":"XYZ","RSSI":72,"APMac":"XX:XX:XX:XX:XX:XX"}}
20:50:40 CMD: Timezone 99
20:50:40 RSL: RESULT = {"Timezone":99}
21:50:46 CMD: command Timezone 99
21:50:46 RSL: RESULT = {"Command":"Unknown"}
21:50:58 CMD: # define timezone 99
21:50:58 RSL: RESULT = {"Command":"Unknown"}
21:51:04 CMD: timezone 1
21:51:04 RSL: RESULT = {"Timezone":1}
20:51:27 CMD: Timezone 99
20:51:27 RSL: RESULT = {"Timezone":99}

not with my config :-/ 
1 =   utc+2 correct time
99 = utc+1 wrong time

RaymondMouthaan, 2018-03-28:
Check this post : https://github.com/arendst/Sonoff-Tasmota/issues/851

Phrede, 2018-03-28:
in user_config.h..........

// -- Time - Start Daylight Saving Time and timezone offset from UTC in minutes
#define TIME_DST               North, Last, Sun, Mar, 2, +120  // Northern Hemisphere, Last sunday in march at 02:00 +120 minutes

// -- Time - Start Standard Time and timezone offset from UTC in minutes
#define TIME_STD               North, Last, Sun, Oct, 3, +60   // Northern Hemisphere, Last sunday in october 02:00 +60 minutes

// -- Application ---------------------------------
#define APP_TIMEZONE           1                 // [Timezone] +1 hour (Amsterdam) (-12 .. 12 = hours from UTC, 99 = use TIME_DST/TIME_STD)

Timezone 99 in sonoff command line = use TIME_DST/TIME_STD that are set in firmware.

on-off-on, 2018-03-29:
@Phrede 
I know about the feature 1 / 99 -please check my listing above. The unit is on "99" -but the time is not correct. Strange -I will check the behavior next week.