arendst/Tasmota #2281: Publishing MQTT messages only using GPIO1 and GPIO3 for HA automations

tatunts, 2018-03-28:
Hello everyone!

I was wondering: is it possible to use GPIOs TX (01) and RX (03) just to publish MQTT messages using a regular 3 button switch?

This is my setup:

[image]

[image]

I’ve used the GND wire on all three buttons. This works if set relay on GPIOs 14, TX and RX (it turns the one and only relay on and off successfully) but no luck setting other switch type like this:

[image]

My idea is when I press buttons 2 or 3 (GPIO TX/RX) a message is published via MQTT so I can use it with an automation (MQTT trigger) to turn on/off a ceiling fan, for example, and so on.

Is it possible?

I wish I could just publish a MQTT message when the contacts are shorted (TX or RX and GND).

The odd part is if I set TX and RX to switch1, they both work just fine! Exactly like GPIO14 does. If I change to anything else, nothing is shown on the console screen. I’ve tried almost every option there.

I really think this could open a lot of possibilities with this already incredible device.

Thanks!

emontnemery, 2018-03-29:
Yes, this is possible, check the wiki here:
https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic

roguestreak, 2018-03-30:
I do hope you're going to ensure those power lines are pushed all the way in and not hanging in the breeze?

tatunts, 2018-04-09:
Thanks a lot @emontnemery! I will check it out. 

No worries @roguestreak, this is only a test prototype.

Thanks!!

ascillato, 2018-06-06:
Hi,

I think that rules should solve your request. Can you share the rules you used?

I'm very interested on this because it should be a very common **use case** of a sonoff basic.

tatunts, 2018-06-09:
Hello!

I could not make this work. Have not tried to do it with rules, if anyone could share an example it would be very appreciated!

Thanks.

ascillato, 2018-06-09:
Hi,

Can you make something like [this](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#2-execute-any-mqtt-message-when-a-button-is-pressed) example of the wiki?

tatunts, 2018-06-11:
Hello @ascillato!

No, sorry. The odd thing is that even with seriallog off, nothing is shown on console. I've set the GPIOs to button, switch, relays, nothing happens.

Thanks!

ssinseeme, 2018-06-17:
have you tried NodMCU it has more GPIOs. or you want just tried it on Sonoff?
I am trying to do the same thing actually with NodeMCU

tatunts, 2018-06-17:
I have not tried with a NodeMCU. Hope you can share your results!

ascillato2, 2018-07-21:
Hi,

Any updates?

tatunts, 2018-07-22:
Hello! Yes! Got it working by setting the board as generic, like this:

[image]

These are the results on logs: 

[image]

Here you can see it working: https://photos.app.goo.gl/JdnB76mZkuPMPzTU7

Here is the backlog used: `Backlog Module 18; GPIO0 9; GPIO1 10; GPIO2 0; GPIO3 11; GPIO4 0; GPIO5 0; GPIO12 17; GPIO13 52; GPIO14 9; GPIO15 18; GPIO16 19; SetOption13 1; SetOption26 0; SwitchTopic 1; switchmode1 3; switchmode2 3; switchmode3 3; switchmode4 3; SerialLog 0`

closed 2018-07-23 by tatunts

ascillato, 2018-07-23:
Great! Thanks for share :+1: