arendst/Tasmota #2284: OTA - Not enough space

camdiep2, 2018-03-29:
Hi all,

I updates some codes in this fw. It works well but when I try to use OTA to update the fw in my device. The system always displays

"Upload Failed
Not enough space"

What should I check now ? 
my fw info as the bellow information
Flash Size =	4096kB
Program Flash Size	= 1024kB
Program Size	= 378kB
Free Program Space =	624kB
Free Memory	= 26kB

I think my free program space is good enough to do OTA but don't know what happens. which methos I should check ? what's problem ?

Thank all and have a nice day.

arendst, 2018-03-29:
> I updates some codes in this fw. 

Check your changes.

earth08, 2018-03-29:
i dont know what have u changed,
but if you upgrade it with minimal and than again with your latest version, it does work.