arendst/Tasmota #2287: Support integrating AM2320 humidity/temperature sensor I2C

florty, 2018-03-29:
Hardware: ESP8266

I do have the necessary code read the sensor's data, but I would like some help integrating it. 
 Any hints will be appreciated. 

Thank you.

roguestreak, 2018-03-30:
I'd like this integrated too, I found a thread from a while ago discussing it but it seemed to just get dropped?

arendst, 2018-03-30:
Well the following from Adafruit is a good reason to not include it:

[image]

roguestreak, 2018-04-01:
So you won't support it because adafruit reckon it's probably not as accurate as the datasheets say it is??

Frogmore42, 2018-04-03:
I think he's saying if you want a inaccurate sensor get one of the already supported ones, ie the DHT11 or 22.  Or for only a little, sometimes not even, more get an accurate one like the SHT, BME, HTU sensors that are also already supported.  

If you really want to use the AM2320, it is not too hard to use the new user sensor capability.  There was some discussion in another issue and someone had some sample code for proportional control, which would help, as would looking at the implementation of the other humidity sensors.

ascillato2, 2018-07-10:
Hi,

Seems that your question has been addressed. Closing this issue. Please, reopen if needed. Thanks.

closed 2018-07-10 by ascillato2

lrswss, 2019-01-10:
https://github.com/lrswss/tasmota-am2320-i2c-driver

nightphobos, 2021-06-24:
https://github.com/nightphobos/tasmota-am2320-i2c-driver - fixes & binary for 9.5.0

ascillato, 2021-06-24:
@nightphobos

Hi, please, can you make a PR with your new driver to Tasmota? Thanks

nightphobos, 2021-06-24:
@ascillato I'll be glad to, wait for pull request this week

nightphobos, 2021-06-26:
@ascillato added as optional functional, because this i2c address has conflict with more popular modules.
https://github.com/arendst/Tasmota/pull/12485

PR #12485 (2021-06-27): add optional support for am2320 temperature/humidity sensor
## Description:

**Related issue (if applicable):** fixes #2287

## Checklist:
  - [V] The pull request is done against the latest development branch
  - [V] Only relevant files were touched
  - [V] Only one feature/fix was added per PR and the code change compiles without warnings
  - [V] The code change is tested and works with Tasmota core ESP8266 V.2.7.4.9
  - [V] The code change is tested and works with Tasmota core ESP32 V.1.0.7
  - [V] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contri [... cut]