arendst/Tasmota #2291: /cmdn/POWER dont return STAT

blizz7979, 2018-03-29:
Make sure these boxes are checked before submitting your issue - Thank you!

- [X] What hardware you are using
Sonoff Basic, Dual and 4Ch
- [X] Provide the output of command ``status 0``
Tried to enter status 0 in the console (web gui),, returned nothing... But these are the versions Im running

Program Version | 5.12.0i
2018-03-26T17:50:30
2_4_1/2.2.1(cfd48f3)

The problem that Im experiencing is that when i Toogle a device trough web gui I get an MQTT Message of something like ......stat/POWER1 Payload: ON but when I sent a command like [topic: ......cmdn/POWER1 Payload: ON] the device turnes on but the Sonoff dont return any "stat/POWER1" at all.
I tried to subscribe to "#" to see if it sent the return stat to another topic then the specified one but it sends nothing in return to the power commands.

So everything seems to work fine besides that I dont get any STAT returned when the commands is sent through MQTT but only when using the Web GUI.

Is this by design or am I doing something wrong?

/Björn

curzon01, 2018-03-29:
Check the Web GUI Console log in parallel when you send the commands. Tasmota is logging what's tried to send to MQTT when you send your command. Your stat topics must be there.

blizz7979, 2018-03-29:
Hmm,, Ok,,,
Log when sending POWER1 OFF through MQTT:
18:17:04 RSL: Received Topic /KARI/UNITS/ESP/9bacdd/cmnd/POWER1, Data Size 3, Data OFF
18:17:04 RSL: Group 0, Index 1, Command POWER, Data OFF
18:17:04 MQT: stat/9bacdd/RESULT = {"POWER1":"OFF"}
18:17:04 MQT: stat/9bacdd/POWER1 = OFF

Log when sending POWER1 OFF through WebGUI (The "Toogle Button" on MainPage):
18:18:39 MQT: /KARI/UNITS/ESP/9bacdd/stat/RESULT = {"POWER1":"ON"}
18:18:39 MQT: /KARI/UNITS/ESP/9bacdd/stat/POWER1 = ON

It seems like that the STAT-topic dont use the settings defined in the user_config.h when I send cmnd and just when using the Web GUI or Buttons.... Wierd.

#define MQTT_FULLTOPIC         "/KARI/UNITS/ESP/%topic%/%prefix%/" // [FullTopic] Subscribe and Publish full topic name - Legacy topic

// %prefix% token options
#define SUB_PREFIX             "cmnd"            // [Prefix1] Sonoff devices subscribe to %prefix%/%topic% being SUB_PREFIX/MQTT_TOPIC and SUB_PREFIX/MQTT_GRPTOPIC
#define PUB_PREFIX             "stat"            // [Prefix2] Sonoff devices publish to %prefix%/%topic% being PUB_PREFIX/MQTT_TOPIC
#define PUB_PREFIX2            "tele"            // [Prefix3] Sonoff devices publish telemetry data to %prefix%/%topic% being PUB_PREFIX2/MQTT_TOPIC/UPTIME, POWER and TIME
                                                 //   May be named the same as PUB_PREFIX

Any thoughts?

/Björn

blizz7979, 2018-03-29:
Hmm,, changed the "Client ID" MQTT Setting to DVES_%06X and everything started to work.

Thanks for the help, didnt know that the WebGUI was putting out logs this way :)

/Björn

closed 2018-03-29 by blizz7979