arendst/Tasmota #2293: AiLight (TYWE3S) Issue. Bulb only show white

helmar74, 2018-03-29:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ WIFI LIGHT BULB AILIGHT TYWE3S] What hardware you are using
- [X ] Provide the output of command ``status 0``
22:36:47 CMD: status 0
22:36:47 MQT: stat/sonoffB1TYWE3S/STATUS = {"Status":{"Module":27,"FriendlyName":"Sonoff","Topic":"sonoffB1TYWE3S","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
22:36:47 MQT: stat/sonoffB1TYWE3S/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffB1TYWE3SStandardGruppe","OtaUrl":"http://192.168.1.50:8080/static/firmware/sonoffB1TYWE3S.ino.bin","Uptime":"0 00:00:53","Sleep":0,"BootCount":15,"SaveCount":77,"SaveAddress":"F7000"}}
22:36:47 MQT: stat/sonoffB1TYWE3S/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018.03.29 21:41:02","Boot":4,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
22:36:47 MQT: stat/sonoffB1TYWE3S/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"192.168.1.254","LogPort":514,"SSId1":"XXXXX","SSId2":"sonoffap","TelePeriod":300,"SetOption":"00008009"}}
22:36:47 MQT: stat/sonoffB1TYWE3S/STATUS4 = {"StatusMEM":{"ProgramSize":490,"Free":512,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
22:36:47 MQT: stat/sonoffB1TYWE3S/STATUS5 = {"StatusNET":{"Hostname":"sonoffB1TYWE3S-3281","IPAddress":"192.168.1.161","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"BC:DD:C2:E0:2C:D1","Webserver":2,"WifiConfig":3}}
22:36:47 MQT: stat/sonoffB1TYWE3S/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.50","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E02CD1","MqttUser":"DVES_USER","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
22:36:47 MQT: stat/sonoffB1TYWE3S/STATUS7 = {"StatusTIM":{"UTC":"Thu Mar 29 20:36:47 2018","Local":"Thu Mar 29 22:36:47 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99}}
22:36:47 MQT: stat/sonoffB1TYWE3S/STATUS10 = {"StatusSNS":{"Time":"2018.03.29 22:36:47"}}
22:36:47 MQT: stat/sonoffB1TYWE3S/STATUS11 = {"StatusSTS":{"Time":"2018.03.29 22:36:47","Uptime":"0 00:00:53","Vcc":3.438,"POWER":"ON","Dimmer":2,"Color":"00050000","Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"XXXXX","RSSI":24,"APMac":"CC:CE:1E:AB:0A:3F"}}}
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

Hi sucessfully flashed a "WIFI BULB" which is a AILIGHT.
I can acces the webinterface and the devices responses to commands, but it simply does nothing.
I'm not able to change color, dim or simply switch off or on.
The bulb just lights white.
Do I have to change some parameters in the main config screen?
Here's what I set:
[image]

I tried every webconsole command but nothing happened.
Any help would be great! Thanks a lot

helmar74, 2018-03-30:
Hi,

I managed  to get white led's dimming. with this settings and setoption15 1:
[image]

I also found out, that a LED Driver (SM16726) is soldered. The Clock PIN of SM16726 ist connected to GPIO4 and the Data PIN is soldered to GPIO14.

Has anyone an idea, what to set to get the RGB LED's running?

Thanks a lot in advance

helmar74, 2018-03-30:
Wenn I try this setting, the white LEDs can be dimmed and RGB are blue, but I'm not able to control color:
[image]

Dieter2000, 2018-04-02:
You need to have fastled.h inside Tasmota. For my understanding this is different to I2C.
As described in https://community.home-assistant.io/t/cheap-uk-wifi-bulbs-with-tasmota-teardown-help-tywe3s/40508/39, I was successful with the colors using an Arduinio instead of ESP8266:

For the RGB-LEDs using the SM16726 (same protocol as SM16716) …

#include “FastLED.h”
#define NUM_LEDS 1 // as 1 LED with 3 colors.
#define DATA_PIN 4
#define CLOCK_PIN 14

CRGB leds[NUM_LEDS]; // this is the array of leds used by FastLED

in the setup:
FastLED.addLeds<SM16716, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS); 

in the loop with R, G, B being declared as Byte:
leds[0] = CRGB(R,G,B); // prepare the data
FastLED.show(); // update the color LEDs

33 PWM 1 on GPIO5 works fine as the white LEDs are connect directly to GPIO5.

helmar74, 2018-04-10:
Moved as pull request

helmar74, 2018-04-10:
Change to Feature Request.
Would it be possible to support the device by inserting the code above?

arendst, 2018-04-10:
What code above? Provide a working PR and I will evaluate.

helmar74, 2018-04-10:
Code from @Dieter2000 above..
Maybe Dieter2000 can assist in providing working code....
 https://community.home-assistant.io/t/cheap-uk-wifi-bulbs-with-tasmota-teardown-help-tywe3s/40508/39

helmar74, 2018-06-06:
With this code the RGB LEDs can be used. Maybe someone can implement?

`int CWpin = 9; // the PWM pin is attached to D9
byte CW = 0; // brightness of the LEDs, initally zero

and for the RGB-LEDs using the SM16726 (same protocol as SM16716) …

#include “FastLED.h”
#define NUM_LEDS 1 // as 1 LED with 3 colors.
#define DATA_PIN 3
#define CLOCK_PIN 13

CRGB leds[NUM_LEDS]; // this is the array of leds used by FastLED

in the setup:
FastLED.addLeds<SM16716, DATA_PIN, CLOCK_PIN, RGB>(leds, NUM_LEDS); //SM16716 is said to be equiv. to SM16726

in the loop with R, G, B being declared as Byte:
leds[0] = CRGB(R,G,B); // prepare the data
FastLED.show(); // update the color LEDs

analogWrite(CWpin,CW); //update the cold white`

closed 2018-08-05 by stale[bot]

kueblc, 2019-01-19:
Hi, I'd like to reopen this request. I have a couple cheap Tuya based RGBW bulbs from Walmart I just flashed OTA.

After some digging I've found that the W channel is controlled independently of RGB as PWM on GPIO5. The color is managed by a [SM726EB](http://chinaasic.com/product.jsp#item=series#style=21#id=104) which has a protocol compatible with the SM16716, (supported by FastLED) on GPIO4 and 14 (clock and data respectively). GPIO13 seems to gate the RGBs collectively.

I gave it a brief attempt at hacking support in by modifying the AiLight profile to use pins 4 and 14 but no luck yet.

epicurean70, 2019-03-06:
I put in these settings but I cannot get any RGB control, all I get is white light regardless of my slider settings in Tasmota
GPIO 2 PWM 4 
GPIO 4 PWM 5 
GPIO 12 PWM 2 
GPIO 13 PWM 3 
GPIO 14 PWM 1