arendst/Tasmota #2295: Support OpenThermGateway

renne, 2018-03-30:
European boiler and air-conditioning manufacturers started about the year 2000 to switch to the[ Opentherm](https://www.opentherm.eu/) [bus/protocol](http://www.domoticaforum.eu/uploaded/Ard%20M/Opentherm%20Protocol%20v2-2.pdf) to control their devices.

[NODO-SHOP.NL](https://www.nodo-shop.nl/en/) offers an [OpenThermGateway](https://www.nodo-shop.nl/en/opentherm-gateway/188-opentherm-gateway.html) with a socket for the NodeMCU.

It uses a [serial interface](http://otgw.tclcode.com/features.html#serial) to the PIC 16F88. Support to convert the [proprietary protocol](http://otgw.tclcode.com/firmware.html) by Tasmota would allow to control boilers and air-conditioning units via MQTT.

renne, 2018-04-13:
[Opentherm Arduino IDE library](https://github.com/jpraus/arduino-opentherm), circuits and PCB.

ascillato2, 2018-11-06:
Hi, there is someone working on a Tasmota's driver for this sensor?

ascillato2, 2018-11-11:
Closing as nobody seems to be interested on making a driver for this.

Added the label hold over to take it into account for future reference.

closed 2018-11-11 by ascillato2