arendst/Tasmota #2297: Timer does not work

wunderbaum, 2018-03-30:
- [Sonoff Basic ] What hardware you are using

I just tried the new timer function. Unfortunately, it seems to do nothing on my sonoff basic. I defined a timer in slot 1 to switch it on. At the reached time, nothing happend.
What have I done wrong?

arendst, 2018-03-30:
First of all provide output of command ``status 0`` as requested. Then also provide the output of command ``timers``.

wunderbaum, 2018-03-30:
Thanks!
I found out that its messed up with DST local time, UTC and non-DST. It seems that its not using the DST.

14:58:44 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Fri Mar 30 13:58:44 2018","Local":"Fri Mar 30 14:58:44 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}

And this is the timezone:
15:06:40 MQT: stat/sonoff/RESULT = {"Timezone":1}

The device is one hour behind.

arendst, 2018-03-30:
So execute command ``Timezone 99`` should fix it.

wunderbaum, 2018-03-30:
Great, some things are much easier now!

closed 2018-03-30 by arendst