arendst/Tasmota #2299: sOnOff TH16 config

xMadelain, 2018-03-30:
Hi,

Flash it with "Sonoff-Tasmota 5.12.0 by Theo Arends"
Manage it on a RaspPi 3B with
-Mosquitto
-homebridge-mqtt-switch-tasmota

T°,Humidity ok in the console (not already tested in Homebridge)

BUT
what is the setting to have leds blinking or something else (Wifi Ok, On/Off)

Merci

Dickesplumpel, 2018-04-01:
I have never used the "morse-code" - maybe you are searching this?
Section main at  https://github.com/arendst/Sonoff-Tasmota/wiki/Commands
I only use the savedata, sleep, switchmode and wificonfig commands.

xMadelain, 2018-04-02:
I explain.....
On sOnOff TH16 you have 2 LEDS
With stock firmware LED are "blinking" or are "On" for WIFI or POWER . 
With tasmota where do I set LED in the config
Currently I Have
GPIO1 Serial Out = 00 
GPIO3 Serial In = 00 
GPIO4 = 00 
GPIO14 Sensor  02 AM2301 => Give t° and humidty : OK

Dickesplumpel, 2018-04-02:
On the webui of tasmota you have "configuration" "configure module" and
then you can choose the type of the device ("Geräte Typ" in german - I have
no device in englisch). there you can choose "sonoff th" and the LEDs
schould work out of the box. Else "
https://loads.pickle.me.uk/2016/10/30/flashing-custom-firmware-to-a-sonoff-th10/16/"
says
gpio 12 - out - red led + relay - low=on
gpio 13 - out - blue led - low=on

xMadelain, 2018-04-02:
Thanks
But in tasmota gui there is only "GPIO1 Serial Out = 00; GPIO3 Serial In = 00; GPIO4 = 00; GPIO14 Sensor 02 AM2301 ")
[image]

No settings for gpio 12 or 13
How could I set them (command eline ) ?

Dickesplumpel, 2018-04-02:
Because the LEDs are fixed to the GPIOs, you can't change it without soldering, you can onlychange the behavior of the LED with the commands. When you solder s.th. other than LED to GPIO13 then you have to choose module type generic.

Frogmore42, 2018-04-03:
Look here https://github.com/arendst/Sonoff-Tasmota/wiki/Commands
For Ledstate