arendst/Tasmota #2300: wemo or hue emulation and google home

fragolinux, 2018-03-30:
hi, can anybody provide some guide to how to connect aGoogle home mini device to a sonoff basic flashed with latest tasmota? i tried wemo emulation, in HOME app put my wifi and the sonoff mac address, after 3 "strikes" of not finding it now wemo site says i've to wait 10 minutes before retry, for security...

reloxx13, 2018-03-30:
does not work anymore, only with those orginal apps/skills (dunno what they r called by google).

or use a server between e.g. IOBroker or IFTT, they got an app for google home.

harkoy, 2018-04-04:
Try with hue emulation in tasmota and yonomi app (Android)

gabryk91, 2018-04-22:
@harkoy it works! Thanks

Karosm, 2018-04-30:
Can you give some details how you made it work with hue emulation? Thank's

gabryk91, 2018-04-30:
It's pretty straightforward, on the sonoff just activate hue emulation, then install yonomi on your phone, set the account and run device discovery.
After that you can create scenes and use them in google home

Karosm, 2018-04-30:
Thank's, I will try tomorrow.

fragolinux, 2018-05-02:
@gabryk91 hi
i just tried, adding a sonoff s20 with hue emulation to yonomi (it autodetected it)...
removed all defaults favourites and created 2 new routines ("accendi luce cosenza", "spegni luce cosenza"), and created even 2 favourites for them in yonomi
added yonomi to google home, it detected my 2 routines and i added to my room
but if i try to tell those phrases to google assistant it searches for them instead of execute them via yonomi...
where am i wrong? thanks in advance

gabryk91, 2018-05-02:
@fragolinux devi creare delle scorciatoie attraverso Google assistant, metti la frase che vuoi pronunciare e in "Google home deve fare" metti activating nomescena. I nomi delle scene creali sempre senza spazi, tipo: accendilucecosenza.

fragolinux, 2018-05-02:
@gabryk91 
ehm, so in google Assistant app, NOT home... can you guide me through it? Can't find where to define custom shortcuts in assistant nor in home...
sorry to answer in english, but i think it could be useful for others who eventually read

fragolinux, 2018-05-02:
@gabryk91 
found this guide (all the others say to use IFTTT, which i don't want...): https://support.google.com/googlehome/answer/7382893?hl=en
but i've not that option in that menu... maybe available only if using app in english?

gabryk91, 2018-05-02:
Agli altri non è utile perché è una limitazione dei cell impostati su italiano, se imposti il cell in inglese puoi seguire la guida che hai trovato, altrimenti: apri Assistant, premi l'icona blu rotonda in alto a destra, seleziona un'app qualsiasi tipo "Parla con Magic 8 ball" o "Assistente salariale", vai in fondo alla pagina, premi "Gestisci scorciatoie". Premi il + in basso a destra.
in Quando dico... metti la frase che vuoi pronunciare, in "L'assistente google dovrebbe fare" metti fisso "activating" seguito da nomescena, tipo "activating accendilucecosenza". 
Una volta create le scorciatoie ti conviene dire a google assistant "sincronizza dispositivi", così aggiorna tutte le impostazioni.

fragolinux, 2018-05-02:
ah, ok, grazie, domani provo :)

fragolinux, 2018-05-02:
grazie, funziona! Se non fosse che a volte il servizio Yonomi risulta irraggiungibile... mi sa che vado avanti a completare il setup di Home Assistant, così almeno dipendo da servizi cloud solo per la parte di google e non per app di terze parti... d'altronde ho tolto il firmware stock per non dipendere da eWelink, non ha senso fare tutto 'sto giro per dipendere da altro servizio cloud, per altro con frequenti disconnessioni... grazie ancora! :)

closed 2018-05-02 by fragolinux

gabryk91, 2018-05-03:
@fragolinux in che senso completare il setup di assistant?
Che yonomi sia spesso irraggiungibile è vero, io ora l'ho tolto e ho direttamente aperto il sonoff all'esterno, così posso fare una chiamata http direttamente da iffft e funziona decisamente meglio

fragolinux, 2018-05-03:
@gabryk91 setup di HOME assistant, un sw di domotica... https://www.home-assistant.io/
purtroppo stanno facendo un casino coi nomi... google home - google assistant - home assistant

gabryk91, 2018-05-03:
Ahhh si! Conosco, ma è troppo sbattimento, serve rasperry o qualcosa su cui farla girare