arendst/Tasmota #2301: Is it possible to add a current sensor to the 18 Generic module?

PETEIOT, 2018-03-30:
I'm using a NodeMcu and I wanting to add a current sensor to it but not sure what sensor so select for the GPIO pins? I want it to work like the Sonoff POW but with more relays and functions. Also do you know what GPIO pin the current sensor is plugged into when using the Sonoff POW Module?

Thanks for you help!

Leseratte10, 2018-03-30:
The current sensor (HLW8012) on the Sonoff POW uses GPIO 5, 13 and 14 (GPIO_HLW_SEL, GPIO_HLW_CF1 and GPIO_HLW_CF). These sensor types are not declared as user-type sensors in the sonoff_template.h file, so adding them to a Generic module probably requires code changes.

PETEIOT, 2018-03-31:
Ok thanks!

arendst, 2018-03-31:
Tasmota supports the Pzem004t sensor too. It works attached to a Sonoff basic as expected.

renne, 2018-03-31:
I'm going to connect a ZMCT118F current transformer with a voltage divider to an A/D-converter for electrically isolated measurements. The 10-bit 200Hz-ADC of the ESP8266 should suffice for approximate values. If you need precise measurements you can use something like the 16-bit ADC1115.

PETEIOT, 2018-04-21:
Ok I Have tried a few different configurations to get thePZEM004T sensor to work on a NODEMCU but I can’t get it to give me any readings. I’m hooking up the (PZEMTx to D7) and the (PZENRX to D8) Using the generic board in the most recent Tasmota, can somebody please tell me what the best way to hook this up is. Thanks!