arendst/Tasmota #2302: cfg_holder value on the Information of Web UI

almostlunatic, 2018-03-30:
I started to experiment with this wonderful FW yesterday - oh, how I like this! As I build and upload very often it's sometimes hard to keep track of the cfg_holder value. It's like the SOA serial number in DNS (I'm a 30-year-experienced network engineer). Could I ask you to add the cfg_holder value on the Web UI information menu? I can code, but it would be 5 times slower compared to an everyday coder with this nice js code.

arendst, 2018-03-30:
How about trying tto use console setting commands instead of fooling around with static settings in the user_config.h file.

I see no use whaatsoever to change the config_holder other than satisfying people arriving here from Arduino like sketches.

reloxx13, 2018-03-30:
just add and subtract 1 from the holder, no need to track anything. if u r done change it back to default and flash a last time.

almostlunatic, 2018-03-31:
No problem to use the console commands. I've just been fighting with user timers that I couldn't get to work, and with the 5.12.0i build which is somewhat buggy (#2264 ). For this I needed to flash all the build many times but started to forget the serial I had used. With just one project it is not hard but with 5 parallel ones it starts to get a pain. Ok, can live with it. Thank you for your efforts!

closed 2018-03-31 by almostlunatic