arendst/Tasmota #2303: Feature request, LED state via json

MatzeJoerling, 2018-03-30:
Hi, 

wrote some node red stuff that steers my 60 Pixel LED WS2812 Ring with a Power Gauge ( Solar Power vs. used Power is shown). In a bad case i have to send 60 mqtt messages to set all LED to there right color.
It would be nice to send only a json with all information to set led color! I know all LEDs with RGB Information will extend the max mqtt message size, but then i would split the json to 3 or 4 messages, better than 40 or more ...

regards M.

arendst, 2018-03-31:
How about using the backlog command to set some more leds at once keeping in mind the max number of backlog commands (10) and the payload length (255 characters).

arendst, 2018-03-31:
Just changed max number of commands from 15 to 30. The max payload length is only valid for serial and web console but is not a problem for mqtt.

arendst, 2018-03-31:
The latest version now supports multiple colors for subsequent leds separated by a dot.

The command
```
Led2 110000.001100.000011
```
Will set led 2 to Red, led3 to Green and led4 to blue.
```
Led2 16,0,0.0,16,0.0,0,16
```
Should also work

MatzeJoerling, 2018-03-31:
Nice, will try to implement in nodered!

but you do much stuff with json (timer, ds18x20...), why now this new approach?
javascript object to json and vice versa is easy in nodered (i belive in every other tool too).

anyway, will save me a lot of mqtt messages! tnx.

closed 2018-03-31 by MatzeJoerling

arendst, 2018-04-01:
Attention: Just change the separator from dot to space to keep in sync with other commands and possible future changes.

I indeed reply with JSON status messages but I find it a lot easier to supply simple commands not using JSON. The advantage is also that the payload string can be used to it's full potential.

martin072, 2018-07-05:
Just trying it out, and I have some strange behaviour;

have set the Pixels to 3 as I have 3 WS8212 led's on a test-bed, then following happens;
(command's via the console)

Command: Color 12 (all lights to white) works OK
Command: Led1 000000 -> First led now goes to a Cyan color but result says: RESULT = {"Led1":"000000"}
Command led2 000000 -> first led now glows red and led2 dims to white, result says: RESULT = {"Led2":"000000"}
command led3 000000 -> First led now glows red, 2nd led green, 3rd led cyaan,  result says: RESULT = {"Led3":"000000"}

Dimmer is at 100 and the command HsBColor returns: {"POWER":"ON","Dimmer":100,"Color":"FFFFFF","HSBColor":"0,0,100","Channel":[100,100,100]}

Any idea what this could be? Am I missing a setting here?

See attached picture for Led1 000000 Led2 000000 Led3 000000

[image]

MatzeJoerling, 2018-07-07:
not enough voltage, bad soldering ... 
tasmota shows you only the command that was executed, not what really happens :-)

martin072, 2018-07-07:
Can’t be. Its on a 1A external 12v power with only 3 Leds, so that would be max 0.2A ?

MatzeJoerling, 2018-07-09:
you are kidding? 12v for WS2812? 
According to the datasheet:
[image]
http://www.world-semi.com/DownLoadFile/108

sorry, you bricked it ... :-(

martin072, 2018-07-09:
Ehm @MatzeJoerling , No obviously not.. the Wemos D1 R2 is running from a 12v power supply not the WS2812. Further more If I would command #000000 for the color's they should be 'off' so I can't really understand this is a power issue? As mentioned and seen on the picture, its only a 3 LED test, no full string if LEDs which cannot consume moren then 0,2 Amps can it? 
As for the soldering, 3 wire's and all the LED's do work if I give them a command with 'Color 12' for example. All colours correspond.

MatzeJoerling, 2018-07-09:
bad color with ws2812, in my case, was alwas some issue with voltage. sorry, no idea anymore.
I have some of them and also bunch of ws2812, only got wrong colors when i have to much / less voltage ... btw, how did you attach 2812 to wemos?

martin072, 2018-07-29:
Hi MatzeJoerling, small update. Used another power supply, and now it seems to work fine. Thanks again for the help.