arendst/Tasmota #2304: RGB-hexvalues not accepted

renne, 2018-03-31:
Hi,

an ESP8266-12F is configured as a generic device with the following LEDs:

```
Red    PWM1 GPIO5
Green  PWM2 GPIO0
Blue   PWM3 GPIO15
White  PWM4 GPIO2
Amber  PWM5 GPIO4

SetOption15 = on
```
Sonoff-Tasmota 5.12.0

"cmnd/sonoff/Color FF00FF0000" works fine, "cmnd/sonoff/Color FF00FF" doesn't. As most applications use standard HEX codes they cannot control the RGB-color. I suggest to either add the suffix "0000" to a standard RGB HEX-value or keep the current value for CWWW.

arendst, 2018-03-31:
Try entering rgb as three comma separated decimals like color 255,128,64 and see what happens.

renne, 2018-03-31:
Three comma separated decimals works with e.g. mosquitto_pub. [MQTT Dash](https://play.google.com/store/apps/details?id=net.routix.mqttdash) only supports RRGGBB Hex-values and decimal numbers like 15344419.

arendst, 2018-04-02:
Latest development release now also supports hexadecimal entry for RGB only.

renne, 2018-04-02:
Entering RGB values with Color/Color2 works in 5.12.0i. :)

Is there any way to return the color format of stat/sonoff/RESULT as RGB instead of RGBCW?

arendst, 2018-04-02:
Nope.Just ignore them.

closed 2018-06-02 by ascillato