arendst/Tasmota #2306: Sonoff dual and ds18b20

andyd999, 2018-03-31:
I wish to connect a ds18b20 to Sonoff dual. What's the easiest way of doing it? Is gpio 14 easily accessible ? And will I need an extra pull-up resistor?

curzon01, 2018-03-31:
Depends on Dual version you have. To connect your DS18B20 to a Sonoff Dual R2 you can use **GPIO1** (TX), **GPIO3** (RX), **GPIO0** (Button0) or **GPIO9** (Button1), select one of these - all available on board by two header - see [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Dual-and-Dual-R2#sonoff-dual-r2).
If you use TX or RX pin don't forget to disable serial logging (`SerialLog OFF`).

andyd999, 2018-03-31:
Ok so I use the power and ground from the serial header then the data line to say serial tx? Do I need a pull-up resistor?
It's the R2 I have

curzon01, 2018-03-31:
If your selected GPIO has internal pull-up enabled it should work without - why not trying it without pull-up.

Note: Internal pull-up will not have the suggested 4k7 resistor value but resistors up to 50k will also work for DS18b20

andyd999, 2018-03-31:
Do the R2 out the box have any pull-up resistors?

curzon01, 2018-03-31:
All Sonoff used ESP8285/ESP8266 has programmable internal pull-up/pull-down - see [Espressif ESP8285 datasheet](https://www.espressif.com/sites/default/files/documentation/0a-esp8285_datasheet_en.pdf) and [Espressif ESP8266 datasheet](https://www.espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf).

For board pull-up/down please see [Sonoff Dual schematic](https://www.itead.cc/wiki/images/1/1d/Sonoff_Dual_View.pdf)

andyd999, 2018-03-31:
Thanks but for me being not savvy with sonoff what's that mean in English?

closed 2018-06-20 by stale[bot]