arendst/Tasmota #2310: USE_MQTT_TLS unusable

GreyMonkey56, 2018-03-31:
- [ITEAD Sonoff Development board ] What hardware you are using
- [ impossible to issue any command] Provide the output of command ``status 0``

After making changes to user_config.h the board was flashed
Nothing happens on the MQTT
One wiki page mentions: enable mqtt through the webbrowser
With TLS there is no webbrowser
Kill:
//#if defined(USE_MQTT_TLS) && defined(USE_WEBSERVER)
//  #error "Select either USE_MQTT_TLS or USE_WEBSERVER as there is just not enough memory to play with"
//#endif
Now we have a webbrowser and TLS: enable mqtt
Lo and Behold:
20:57:23 MQT: Verify TLS fingerprint...
Alas:
dnsmasq tells: query[A] domus1 from 192.168.1.31
THERE IS NO domus1 ANYMORE IN THE user_config.h
Must be in the saved configuration I presume.
How to wipe/clean out configuration?
By webbrowser again? Which is supposed to be incompatible with TLS !

Nice Easter Egg!

reloxx13, 2018-03-31:
read the user_config.h from top to button, everything has comments and is explained in it.

you would find CFG_HOLDER to apply new configs and MQTT 1/0

alternative erase all flash with esptool: https://github.com/arendst/Sonoff-Tasmota/wiki/Esptool#optional-erase-firmware

ascillato, 2018-06-02:
Hi, @GreyMonkey56,

If your issue was solved, please close it. Thanks :+1:

closed 2018-06-13 by ascillato2