arendst/Tasmota #2311: Sonoff basic's appear dead after rdc trips until  it is removed from circuit

reeso3000, 2018-04-01:
I installed 4 sonoff basic's in my attic with the original switches connected to the gpio. A trip on the rdc causes some if not all (different ones at different times) to seem dead - physical button on the unit does not toggle the relay, no lights, no wifi connection. Removing the sonoff from the circuit for a small amount of time (powering off the lighting circuit is not sufficient) brings them back to life and they work fine until there is another trip. Does anyone have any suggestions how I can resolve this problem?

curzon01, 2018-04-02:
Long wires from original light switches connected to the gpio might cause the problem. This does not work stable enough. Each wire are also working like an antenna and it's easy to drop an ESP within sonoff which uses only 3.3V for GPIO. Solutions: Decoupling the switch wires, using a RF wall switch instead of sonoff, using a wall switch which can send MQTT (Sonoff touch/T1)...

reeso3000, 2018-04-02:
Hi Curzon01,
 Thanks for the response. I had issues with false positives on the wires until I added an additional resistor and this seemed to resolve that problem but you are saying that after a power cut they could be the cause of it not powering back on? (Completely dead temporarily). I will definitely test this in one of the rooms that I can live without a switch in and see if a power spike still kills the basic (temporarily) If you have any suggestions to make this setup work as it is I would really appreciate it - add a component etc. The basics do not have rf and I only have a switched live behind the switch so no neutral for the sonoff touch (I use these on my ground floor as they are oddly wired differently and have a neutral wire). Thanks again

curzon01, 2018-04-03:
Beside the pull-up resistore try using an additional bypass capacitor between gpio and ground to minimize the disturbance long wires causes.

Have a look on issue #484, here [DNA99 comment diagram of 8 Jul 17 17:02 MESZ](https://github.com/arendst/Sonoff-Tasmota/issues/484#issuecomment-313861088) gives a solution.

reeso3000, 2018-04-21:
Thank you for your advise. I added a capacitor but haven't had any power cuts since (a deliberate trip doesn't seem to stop them working  but it didn't before). I have decided I am going to rewire my 1st floor to allow neutral in the box and install Sonoff touch modules as I have had these downstairs working for the best part of a year with no issues.

closed 2018-04-21 by reeso3000