arendst/Tasmota #2312: sonoff with tasmota mosquitto home assistant setup

callumher, 2018-04-01:
hi 
is running a sonoff basic with tasmoto and trying to integrate with home assistant. so far i have successfully 
flashed sonoff with tasmoto
connected to my wifi 
set up sonoff
added to my config.yaml
setup mqtt mosquitto om hass.io
 i can control it with the webui. but the problem comes when i try to control my sonoff with hassio. i have the button on the overview page but have no control. i really have little idea of what im doing but have been fumbling around working things out as i go. i have been stuck like this for a number of afternoons. any help would be greatly appreciated. my pi ip is 192.168.1.96
here is the latest log on Mosquitto broker setup page on hass.io
1522542511: Socket error on client <unknown>, disconnecting.
1522542511: New connection from 172.30.32.1 on port 1883.
1522542511: Client home-assistant-1 already connected, closing old connection.
1522542511: Client home-assistant-1 disconnected.
1522542511: New client connected from 172.30.32.1 as home-assistant-1 (c1, k60, u'callum').
1522542511: New connection from 192.168.1.96 on port 1883.
1522542511: Client home-assistant-1 already connected, closing old connection.
1522542511: Client home-assistant-1 disconnected.
1522542511: New client connected from 192.168.1.96 as home-assistant-1 (c1, k15, u'callum').
1522542511: New connection from 172.30.32.1 on port 1883.

In the log in page on hass.io
Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/lib/python3.6/asyncio/tasks.py", line 180, in _step
    result = coro.send(None)
  File "/usr/lib/python3.6/site-packages/homeassistant/components/mqtt/__init__.py", line 590, in _async_perform_subscription
    _raise_on_error(result)
  File "/usr/lib/python3.6/site-packages/homeassistant/components/mqtt/__init__.py", line 676, in _raise_on_error
    'Error talking to MQTT: {}'.format(mqtt.error_string(result_code)))
homeassistant.exceptions.HomeAssistantError: Error talking to MQTT: The client is not currently connected.
my config file looks like this
mqtt: 
  broker: core-mosquitto
  client_id: home-assistant-1
  password: 
  username: 
  port: 1883
switch: 
  command_topic: cmnd/sonoffDEV1/power
  name: bedroom fan
  payload_off: "OFF"
  payload_on: "ON"
  platform: mqtt
  qos: 1
  retain: true
  state_topic: stat/sonoffDEV1/POWER

and my tasmoto mqtt set up looks like this
Sonoff Basic Module
Sonoff
 MQTT parameters 

Host (domus1)
192.168.1.210
Port (1883)
1883
Client (DVES_57CEE6)
home-assistant-1
User (DVES_USER)
callum
Password
•••••••••
Topic = %topic% (sonoff)
sonoffDev1

Im sure i have done something simple wrong but so far i cant work it out.  Any help would be greatly appreciated

Dickesplumpel, 2018-04-01:
It seems to be a hass problem. I have had hass for a year and it was good, very flexible but very cryptical (a blank-character at the wrong place in the automation can kill the whole system and sometimes the code is changed and parts don't work anymore after an upgrade) . Now i have the combination of domoticz, a pi3 (with 5 ds18b20 and 3 relays "onboard") and 15 devices with tasmota over mqtt - great, for me its best. Maybe you will give it a try.  I have had problems with mosquitto afer a reboot "sudo service mosquitto stop" and then "mosquitto -v" in a console did it and you can watch the mqtt-traffic in the console.

Dickesplumpel, 2018-04-01:
The Host in the mqtt-settings of the sonoff must be the ip of your mqtt-broker. If your mqtt-broker runs on your pi it has to be 192.168.1.96 and in the configuration.yaml  of hass you must set mqtt: broker:127.0.0.1
Maybe thats it.

mcfarlde, 2018-04-02:
I too am getting that socket error on client in the Hassio Mosquito log with a 4CH Pro.
1522697926: New connection from 10.0.1.137 on port 1883.
1522697926: New client connected from 10.0.1.137 as DVES_857B62 (c1, k15, u'mcfarlde').
1522697954: Client DVES_857B62 has exceeded timeout, disconnecting.
1522697954: Socket error on client DVES_857B62, disconnecting.
1522697999: New connection from 10.0.1.137 on port 1883.
1522697999: New client connected from 10.0.1.137 as DVES_857B62 (c1, k15, u'mcfarlde').
1522698034: Client DVES_857B62 has exceeded timeout, disconnecting.
1522698034: Socket error on client DVES_857B62, disconnecting.
In the Tasmota Console, I get:
21:26:37 MQT: Connected
21:26:49 RSL: tele/sonoff/LWT = Online
21:27:01 RSL: cmnd/sonoff/POWER = 
21:27:08 MQT: Subscribe to cmnd/sonoff/#
21:27:26 MQT: Subscribe to cmnd/sonoffs/#
21:27:38 MQT: Subscribe to cmnd/DVES_857B62/#
21:27:45 HTP: Command
21:27:52 RSL: Received Topic /Power1, Data Size 6, Data Toggle
21:27:59 RSL: Group 0, Index 1, Command POWER, Data Toggle
21:28:06 RSL: stat/sonoff/RESULT = {"POWER1":"OFF"}
21:28:13 RSL: stat/sonoff/POWER1 = OFF
21:28:30 CFG: Saved to flash at F9, Count 4368, Bytes 1648
21:28:37 MQT: Attempting connection...
21:28:44 MQT: Connected
21:28:51 MQT: tele/sonoff/LWT = Online (retained)
21:28:58 MQT: cmnd/sonoff/POWER = 
21:29:05 MQT: Subscribe to cmnd/sonoff/#
21:29:12 MQT: Subscribe to cmnd/sonoffs/#
21:29:19 MQT: Subscribe to cmnd/DVES_857B62/#
21:29:33 WIF: Checking connection...
21:29:39 WIF: Connected
21:30:00 MQT: tele/sonoff/STATE = {"Time":"2018-04-02T21:30:00","Uptime":"0T00:15:14","Vcc":3.144,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":2,"SSId":"58F","RSSI":96,"APMac":"D0:17:C2:3D:A0:20"}}
I used the light component instead of the switch as that is what worked when using emulated HUE.  The connection is failing as the HASSIO log shows:
Flame turned on
2:22 PM
Flame changed to unavailable
2:21 PM
Flame turned on
2:21 PM
Flame changed to unavailable
2:20 PM
Flame turned on
2:19 PM
Flame changed to unavailable
2:19 PM
Flame turned on
2:18 PM
Flame changed to unavailable
2:18 PM
Flame turned on
2:17 PM
Flame changed to unavailable
2:17 PM
Flame turned on
2:16 PM
Flame changed to unavailable
2:15 PM
Flame turned on
2:15 PM
Flame changed to unavailable
2:14 PM
Flame turned on
2:13 PM
Flame changed to unavailable
2:13 PM
Flame turned on

Dickesplumpel, 2018-04-03:
For me it looks like that homeassistant can't connect to mqtt - tasmota
does. It seems to me that you might better look here: "
https://www.reddit.com/r/homeassistant/comments/74qqtc/hassio_mqtt_installation_beginner_questions/
"
or start new here "https://www.home-assistant.io/getting-started/" or give
domoticz a try, much easier for beginners.

mcfarlde, 2018-04-03:
I used the Wemo emulation of tasmota instead.  Lost one of the output functions, but can get by without it.

Dickesplumpel, 2018-04-03:
As far as I can see is the problem not tasmota so my opinion is you are in the wrong place with your question - I'm out.

rivaj, 2018-05-19:
should your topics sonoffDev1 versus sonoffDEV1 be the same?

closed 2018-07-10 by ascillato2