arendst/Tasmota #2313: Trigger SwitchTopic as well as relay from stock GPIO0 button?

roguestreak, 2018-04-01:
- [ ] What hardware you are using
Sonoff S22 (and Basic, POW, NodeMCU clone)

I've tried multiple settings of both switchtopic and buttontopic with no avail, and seemingly not to the effect the docs say;  What I want to achieve is standard toggling via the onboard button (GPIO 0), but also trigger an MQTT message at the same time?
Reason being I want an obvious indicator that there's been a physical interaction with the button at the device, separate from MQTT-driven controls; manual bypass if you will.  No matter what I try it still only results in the two stat messages, stat/sonoff_S22/RESULT = {"POWER":"ON"} and MQT: stat/sonoff_S22/POWER = ON, which is no different from if it receives a command from outside.

Am I doing something wrong, or is there currently no facility for this?  If not, is there an argument as to why the button shouldn't control the relay directly AND send a message of a different topic than the norm?
I've searched throught about 20 threads but haven't found anything.

- [ ] Provide the output of command ``status 0``
Not really relevant, but here it is anyway
11:14:02 MQT: stat/sonoff_S22/STATUS = {"Status":{"Module":4,"FriendlyName":["Sonoff_S22"],"Topic":"sonoff_S22","ButtonTopic":"","Power":0,"PowerOnState":3,"LedState":8,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
11:14:02 MQT: stat/sonoff_S22/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:00:20","StartupUTC":"2018-04-01T01:13:42","Sleep":0,"BootCount":64,"SaveCount":328,"SaveAddress":"FA000"}}
11:14:02 MQT: stat/sonoff_S22/STATUS2 = {"StatusFWR":{"Version":"5.12.0i","BuildDateTime":"2018-03-29T15:38:00","Boot":6,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
11:14:02 MQT: stat/sonoff_S22/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["OPTUSFEFF922","indebuurt2"],"TelePeriod":300,"SetOption":["00000009","55818000"]}}
11:14:02 MQT: stat/sonoff_S22/STATUS4 = {"StatusMEM":{"ProgramSize":521,"Free":480,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
11:14:02 MQT: stat/sonoff_S22/STATUS5 = {"StatusNET":{"Hostname":"sonoff_S22-3886","IPAddress":"192.168.1.22","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"2C:3A:E8:0F:8F:2E","Webserver":2,"WifiConfig":1}}
11:14:02 MQT: stat/sonoff_S22/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.20","MqttPort":1883,"MqttClientMask":"sonoff_S22","MqttClient":"sonoff_S22","MqttUser":"openhabian","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:14:02 MQT: stat/sonoff_S22/STATUS7 = {"StatusTIM":{"UTC":"Sun Apr 01 01:14:02 2018","Local":"Sun Apr 01 11:14:02 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":10}}
11:14:02 MQT: stat/sonoff_S22/STATUS10 = {"StatusSNS":{"Time":"2018-04-01T11:14:02"}}
11:14:02 MQT: stat/sonoff_S22/STATUS11 = {"StatusSTS":{"Time":"2018-04-01T11:14:02","Uptime":"0T00:00:20","Vcc":3.504,"POWER":"OFF","Wifi":{"AP":1,"SSId":"OPTUSFEFF922","RSSI":100,"APMac":"BC:AE:C5:EA:C9:A7"}}}

curzon01, 2018-04-01:
Be sure `Status 0` is relevant ;-) it tells me you are using the developer release and you compiled it by yourself. This helps to answer:

disable USE_DOMOTICZ in user_config.h and switchtopic/buttontopic will send out:
`// #define USE_DOMOTICZ                             // Enable Domoticz (+6k code, +0.3k mem) - Disable by //
`

roguestreak, 2018-04-01:
Hi curzon01, I've just rebuilt and tested with USE_DOMOTICZ disabled, and no change; if I set SwitchTopic to a topic the relay still works but no additional topic is sent, and if I set ButtonTopic then that topic is sent but the relay no longer activates?

I'm setting these via the console as per below;

switchtopic s22_pressed (normal topic sent with no additional topic, relay activates)
or
buttontopic s22_pressed (sends topic but no relay)

curzon01, 2018-04-02:
Default for the module type you set the button is a _Button_, not a _Switch_. So you have to use `ButtonTopic`, not _SwitchTopic_. _SwitchTopic_ are for _Switch<x>_  (1-4), which can be set for some modules on module configution page for GPIOs.

For _ButtonTopic_ the functionality is the same as for _SwitchTopic_:

Using `ButtonTopic <topicname>` disables the auto-relation between the button function and the power and you can/have to control the Power switching by yourself.

So you have to use `ButtonTopic 1` or `ButtonTopic ON`. This will set autoset the topic for button to your Topic (here 'sonoff_S22') and the on/off functionalty will remain.

Every time you press the button, an addtional message with `cmnd` topic (!) will be send which indicates the button pressure.

### Example
#### Default behaviour (ButtonTopic OFF)
```
Topic sonoff_S22
FullTopic %prefix%/%topic%/
SwitchTopic OFF
ButtonTopic OFF
```
##### Press button once 
```
stat/sonoff_S22/RESULT {"POWER":"ON"}
stat/sonoff_S22/POWER ON
```

##### Press button again
```
stat/sonoff_S22/RESULT {"POWER":"OFF"}
stat/sonoff_S22/POWER OFF
```

#### ButtonTopic ON
```
Topic sonoff_S22
FullTopic %prefix%/%topic%/
SwitchTopic OFF
ButtonTopic ON
```
Note the first additional line, which uses the `cmnd` prefix
##### Press button once
```
cmnd/sonoff_S22/POWER ON
stat/sonoff_S22/RESULT {"POWER":"ON"}
stat/sonoff_S22/POWER ON
```

##### Press button again
```
cmnd/sonoff_S22/POWER OFF
stat/sonoff_S22/RESULT {"POWER":"OFF"}
stat/sonoff_S22/POWER OFF
```

roguestreak, 2018-04-02:
Thanks curzon01, I think the wiki needs to be updated with your info as I didn't read it like that.  I knew that when I set buttontopic to a new topic I was no longer get the relay toggling, but didn't realised that setting it to "1", or the standard topic title, would give an additionl message AND still toggle.

It's not precisely what I was looking for, as there's no blatant separation between commands coming from elsewhere and commands being sent back by the sonoff, but I might be able to nut out a method with some thought and scripting.
If I'm subscribing to "cmnd/sonoff_S22/POWER", how do you filter out the messages that you've just generated in for example node-red?  Send an ON command via a script, the subcription picks it and a listener then thinks you've physically pressed the button as it's the same command as a button press?

roguestreak, 2018-04-14:
Just a follow-up after some testing; yes a listener subscribed to the cmnd/sonoff_S22/POWER topic WILL see items generated by both the button-press and any commands coming from home automation, but a way of getting around this is by differentiating the payload; the buttonpress will result in "ON" and "OFF" payloads, so if you listen for those and end any automation commands using "1" and "0" it will be able to tell whether you've made a physical interaction or whether it's an auto-generated command