arendst/Tasmota #2314: IRsend not sending the proper data

Diode663, 2018-04-01:
I am trying to use the IRsend command via the console to make sure everything is working.  It wasnt controlling my AC as expected so I used IRscrutinizer to see what was going on.
The command to turn my AC on would look like this:

/IRsend {"protocol": "NEC", "bits": 32, "data":8166817E}

The equivalent of this command in ESPeasy would turn the AC on.

The output of a known functioning remote into ir scrutinizer nets this:
protocol = NEC1, device = 129, subdevice = 102, obc = 129

When I use the IRsend command in tasmota I get this:
protocol = NEC-f16, device = 0, subdevice = 62, obc = 185, misc = OBC2=133
And I also get random intermittent data sent out with no apparent protocol.

The setup is an ir led with an appropriate resistor on a wemos d1 mini proto board attached to pin d8.  As I mentioned before the functional equivalent in ESPeasy works just fine.  I dont want to use ESPeasy though because it doesnt support irsend through mqtt.  I would also like to have all of my devices running the same firmware.

Please let me know what other information you would need to troubleshoot this or if there is something obvious I have missed.

Thank you.

- [ ] What hardware you are using
Wemos d1 mini
- [ ] Provide the output of command ``status 0``
08:25:21 CMD: status 0
08:25:21 MQT: stat/Bedroom_AC/STATUS = {"Status":{"Module":18,"FriendlyName":"Bedroom AC","Topic":"Bedroom_AC","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
08:25:21 MQT: stat/Bedroom_AC/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.1.5:4043/data/firmwares/sonoff-minimal.bin","Uptime":"0T00:06:05","Sleep":0,"BootCount":20,"SaveCount":30,"SaveAddress":"F5000"}}
08:25:21 MQT: stat/Bedroom_AC/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-02-09T14:54:51","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
08:25:21 MQT: stat/Bedroom_AC/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"Homebase","SSId2":"","TelePeriod":300,"SetOption":"00000109"}}
08:25:21 MQT: stat/Bedroom_AC/STATUS4 = {"StatusMEM":{"ProgramSize":485,"Free":516,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
08:25:21 MQT: stat/Bedroom_AC/STATUS5 = {"StatusNET":{"Hostname":"Bedroom_AC-1610","IPAddress":"192.168.1.226","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.7","Mac":"5C:CF:7F:6E:46:4A","Webserver":2,"WifiConfig":3}}
08:25:21 MQT: stat/Bedroom_AC/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.5","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6E464A","MqttUser":"","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
08:25:21 MQT: stat/Bedroom_AC/STATUS7 = {"StatusTIM":{"UTC":"Sun Apr 01 06:25:21 2018","Local":"Sun Apr 01 08:25:21 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99}}
08:25:21 MQT: stat/Bedroom_AC/STATUS10 = {"StatusSNS":{"Time":"2018-04-01T08:25:21"}}
08:25:21 MQT: stat/Bedroom_AC/STATUS11 = {"StatusSTS":{"Time":"2018-04-01T08:25:21","Uptime":"0T00:06:05","Vcc":2.774,"Wifi":{"AP":1,"SSId":"Homebase","RSSI":98,"APMac":"80:2A:A8:97:0A:F6"}}}

arendst, 2018-04-02:
You might want to enable more debug info with command ``weblog 3`` and then try to send the NEC command again. Report the output here please.

arendst, 2018-04-02:
Ah I see. IRSend only allows decimal data entry. That's why your hexadecimal data fails.

Try this:
```
IRsend {"protocol": "NEC", "bits": 32, "data":2170978686}
```
It has been noted by others that I should support hex input too. Will have a look at it now.

Diode663, 2018-04-02:
12:16:05 CMD: /IRsend {"protocol": "NEC", "bits": 32, "data":8166817E}
12:16:05 RSL: Group 0, Index 1, Command IRSEND, Data {"protocol": "NEC", "bits": 32, "data":8166817E}
12:16:05 IRS: protocol_text NEC, protocol NEC, bits 32, data 8166817, protocol_code 3
12:16:05 MQT: stat/Bedroom_AC/RESULT = {"IRSend":"Done"}

What I find really interesting is the seemingly random bursts of 38khz data that come out of led at random.  The brightness isnt enough to be seen with my camera unlike when I try to send a real pulse but its there.  Im gonna try to get a scope on it tonight.  I wonder if its picking up emi from the wifi antenna which is about an inch away but running parallel to the diode and resistor.  Have you seen anything like this before?

arendst, 2018-04-02:
No as I'm not really using this feature at the moment.

Did you try to send the data in decimal as I noted just above?

arendst, 2018-04-02:
Try the latest development release using 0x8166817E as data value.

Diode663, 2018-04-02:
I got home from work and was able to try it.  It works with hex now using the 0x prefix.  I dont know if it was really necessary to add this if just converting the hex to decimal would have worked but I do appreciate it.  Thank you.

closed 2018-06-02 by ascillato