arendst/Tasmota #2315: Adding support for PZEM004T energy monitor for 3 phase - request

roblad, 2018-04-01:
Hi,

Is it possible add  USE_PZEM004T and add support for PZEM004T Energy monitor for 3 phase by using 
USE_SERIAL_BRIDGE and report each phase through that functionality ? May be there is possibility to use 3 phase config for USE_PZEM004T  as it is in that project:

https://github.com/apreb/eNode

roblad, 2018-04-13:
Hi,

Is it possible to add 3 PZEM004T in one node?

Rgds,
RL

0x3333, 2018-04-26:
I'm looking for the same thing.

As long as I can tell, each PZEM004T can be directly addressed, so you can hookup as many PZEM004T as you can in the same UART, when you read, you send the "ip address" of each module.

But the code in `xdrv_03_energy.ino` is not able to read from multiple sources. When I get my units I'll work on this. Don't know if this can be accomplished, but would be great.

roblad, 2018-04-26:
Hi,
It means that when I change IP for each PZM and plug 3 of themas a 3 soft serials or plug 3 of them to 1 serial ? Could you sent me a private mesage if you could with more detailed manual or some code that you hook haw it looks. I am not such expert as you it will be very usfull :-)

Thanks,

0x3333, 2018-04-26:
I mean plugging all the 3 PZEM in the same hardware/soft serial, like this:

[image]

The diode can be a BAT45, the pullup resistor can be a 10K to 5V.

But you need to program the PZEM address before using this method, the address can be sequential, like 0.0.0.1,0.0.0.2,0.0.0.3, it is not really a IP address.

But the tasmota must be changed to accommodate multiple meters, you can test using an arduino.

roblad, 2018-04-26:
Hi,

Many thanks, but I have a question, may be stupid, Master UART means tasmota serial ? Yes ?

And all of the slaves there are PZEMs Am I right ?

Have you got it on your Arduino/ESP connected ?  

I will try it ofcource but first of all I am working for MQTT bridge through serial bridge of Tasmota - I have Nodes based on it:

https://github.com/artekw/sensnodetx
https://github.com/artekw/sensnodetx

and receiver basenode which prints output to serial as JSON, I am working on www displaing all recaived data as normal sensors pluged to tasmota, next one task will be the hook of PZM004 ino.

and

0x3333, 2018-04-26:
Yes, Tasmota master, PZEM slaves. Yes, connected to ESP.

roblad, 2018-04-26:
Hi,

Could you share Your ino code for PEZM for futher testing, I have 3 already :-) after 2 months shipement :-) ?

roblad, 2018-04-27:
Hi,

Many thanks, but what do you mean 👍 
Remember to call pzem.setAddress(ip[X]); with only one connected to set it's internal address.

I did not catch you, it means that I have to set the pzem.setAddress(ip[X]) step by step with plug only one PZEM for initial setup of IP ? And next plug all of them for finale mastering am I right ?
What about Save set values in PZEM, will they disappear after power lost  ?
What do you mean "with only one connected to set it's internal address"

roblad, 2018-04-28:
Hi,

" but you should do your homework"   yes I will :-), thanks for explanation, I let you know about my results :-)

0x3333, 2018-04-29:
Great, as soon as I get my sonoffs, I'll let you know my findings.

asking23, 2018-05-28:
Any updates ? but in this way we would never be able to poll 3 Phases Simultaneously to Make Graph/Trend 3 Phases.  Each Consecutive reading will have 1 Second Interval so wont be able to correlate voltage on each phase to each other.

0x3333, 2018-05-29:
I'm looking to do the readings in the same time slice. But I'll need to change the code. I'm on vacations now, next week I'll work on it.

0x3333, 2018-06-12:
Here I'm again. I read the code and believe that create another xdrv for this is the best option to me. Change a code that is shared with other devices to add 2 more phases will require a lot of effort and will not benefit the others. I'm looking to create a `xdrv_13_energy_3phases.ino` driver, especifically to PZEM004T, this way it will not add more line of codes to other energy meters and increased complexity.

I've only an hour at night to work on side projects, but will give it some attention.

@arendst , do you believe that this is the right thing to do? What is your direction on this matter?

Thanks!

roblad, 2018-06-12:
Hi,

Where is xdrv_13_energy_3phases.ino ?

0x3333, 2018-06-12:
There is no xdrv_13_energy_3phases.ino, I'm about to write it.

roblad, 2018-06-12:
aaaaaaa :-)

0x3333, 2018-07-27:
Well, I changed to ESPurna, and in that firmware, it's simply not possible, it is too slow to read all magnitudes from all sensors. I'm working in a custom board with an Arduino Pro Mini that will read all PZEM(Max 6 devices) and will communicate using I2C, all measurements will be read in a single call. I'll let you know the progress.

rt400, 2018-08-21:
any new with this ?

0x3333, 2018-08-21:
I'm finishing the custom board, the board will read up to 6 PZEM004T and will communicate using I2C/SPI. It constantly read the PZEM and buffer the responses, when asked the response is instantaneous. In my board I also added a ESP-01 module, so I don't need a Sonoff Basic or alike. But it is not finished yet...

[image]

[image]

0x3333, 2018-08-21:
Probably I will use this repo: https://github.com/0x3333/powermeter

But the code probably will change...

rt400, 2018-08-21:
https://github.com/apreb/eNode
did you see this ?
its looks that he success to do that..
can you share the diagram of connection ?
and i understant that you connect directly to wemos or arduino board ..so why use also in spi or i2c ?

roblad, 2018-08-21:
Hi,

Great, but will you add that to tasmota as a 3 phase module ?

0x3333, 2018-08-21:
Sure, @arendst will need to approve a custom driver, but sure.

roblad, 2018-08-21:
ok Iam waiting :-)

ascillato2, 2018-09-04:
Hi,

Any news on this?

0x3333, 2018-09-04:
Not yet, but I'm working on creating a new library non-blocking instead of a custom board. Probably I'm the only one working on this, so, asap I'll finish the library, that will be exclusive to ESP, I'll let you guys know.

roguestreak, 2018-09-05:
Thanks 0x3333, I've just picked one of these up and would like to be able to read multiple of them from one Tasmota'd Sonoff (or wemos), in my case it's single phase so the timing isn't an issue, I just want to be able to get the data from multiple of them without needing multiple ESP8266 devices; would be a good cheap way of adding power monitoring to each circuit in the house, without having to put something like a POW inline.

On a side note, would be interested to see if anyone has tried monitoring Pin 51 when connected to the street-side of an importing/exporting meter (eg. solar PV), to see if it does in fact indicate when you are "feeding back" into the grid?  Even the metering guys at work are unsure of what happens phase-wise when exporting to the grid.

rt400, 2018-09-19:
hi..
i hope that someone have news to use 3 phaze monitor in one WEMOS with TASMOTA..
i try but he recognize only ONE..
[image]
[image]
here i config 2 monitor on different GPIO and still only one work....

0x3333, 2018-10-02:
Time is limited here. I'm testing several options, let you guys know when I progress.

rt400, 2018-10-02:
@0x3333  you are the best man...
thank you..
hope it will allow us to use it

Jason2866, 2018-10-04:
@rt400 
Tasmota can only handle one device of the same type. You cant configure it multiple on different GPIOs

andrethomas, 2018-10-04:
I think the only logical way to do this is to offload the 3 sensors to a pro mini and read sensor values over I2C - Having 3 x PZEM004T connected at the same time in Tasmota is not possible.

I'm guessing since this thread has been hanging for 6 months the above is not likely to happen any time soon. This problem could have been solved with 3 wemos d1 mini's or 3 bare esp8266 boards and just report each phase independently - unless its actually being used to monitor a 3 phase device, having it all on seperate esp8266 modules would not make any difference - besisides, if you want to consolidate one one of the esp8266 modules you can always use knx to push data to that main one.

The latter will take less than 6 months ;)

ascillato, 2018-10-04:
Hi

@0x3333 

Do you have any progress in modifying the Tasmota PZEM driver? or you are fine with the workaround explained by @andrethomas ?

0x3333, 2018-10-04:
Well, using 3 tasmota devices is for sure the fastest option. 

But I'm working on another solution, a custom board with Atmega 2560 reading up to 6 devices, 2 per UART(it has 4 uart, one is used to debug). But Im working in some side projects that are using my spare time. I can't give you a time frame.

Also a option is re-write the pzem library to be async, continuously reading the pzem devices and returning the lastest reading when asked. This would make the tasmota reading much faster, allowing multiple devices. But we should change the way it is today, to allow multiple devices per gpio or multiples gpios.

andrethomas, 2018-10-04:
Yeah, it is not impossible to modify the driver to have 3 inputs but they would need to be 3 different names for the 3 separate RX pins - You do not need TX to be connected as you are only reading what the PZEM is spitting out.

I do not own any of these devices otherwise I would give it a go.

So basically you would ideally end up with something like this:

[image]

andrethomas, 2018-10-04:
Possibly even make it possible to have more than 3 since there are still free GPIO's then people who want to monitor multiple single-phase measurements, from say inside the AC Tripswitch Box of the house...one for lights, one for plugs, one for geysers, one for outbuildings etc... So perhaps also keep this use case in the back of your mind when writing the code - to allow the individual use or 1or 2 x 3 phase use depending on configuration settings, of course.

0x3333, 2018-10-04:
@andrethomas nope. You must have TX. Pzem doesn't spit things, you must request.

andrethomas, 2018-10-04:
Right, common TX then but the RX lines need to be separately handled - cannot be duplicates of the same handler.

roguestreak, 2018-10-05:
Wasn't the idea (if not looking at simultaneous readings) that one RX/TX can be used for multiple PZEMs, you "just" need to configure them first (using a script?) to have an "address", and then the ESP can cycle through these addresses?
I couldn't figure out the configuration bit, or find a script to do it.  I think it could work if we dictate that you must use sequential addresses, then have a console setting to specify start address (eg. 10.0.0.1) and how many PZEMs (eg. 4, so addresses queried are 10.0.0.1, 10.0.0.2, 10.0.0.3, 10.0.0.4)
Good for monitoring all circuits on a single-phase install.

asking23, 2018-10-05:
I have already implemented...3 Phase PZEM-004T using multiple Soft Serial TX RX in NodeMCU Amica. I am not sure that readings are synchronised...but they do work....

[image]

0x3333, 2018-10-05:
@roguestreak yes, but using current drivers is not possible, it is sync and takes too much time waiting. I'm working in another solution, using async reading and synchronized reading.

rt400, 2018-10-07:
@Jason2866  i understant that..
i wonder if anyone can change the pzem004T driver that can handle 3 devices..
with simple arduino its work but i dont know how ...

elmo3lem, 2018-10-07:
can connect directly with out type WiFi password or ssid  ?

0x3333, 2018-10-07:
@rt400. Yes. But as

0x3333, 2018-10-07:
@rt400 yes, but I don't believe it will work, as it is sync and takes to much cpu...

0x3333, 2018-10-15:
Updated the repo. [powermeter](https://github.com/0x3333/powermeter)

Some feedback is great. This week will send to manufacture the boards.

rt400, 2018-10-16:
@0x3333 why not ?
its looks that we can change the variable names ...no ?

0x3333, 2018-10-16:
@rt400 thats not the problem, the problem is that it takes to much time waiting... anyway, give it a try...

rt400, 2018-10-16:
@0x3333 i wish iknow how to do . i dont have any problem to check it.. but a dont know how to recoding the driver for that

rt400, 2018-10-23:
i seems now that in tasmota have only 1 TX and many RX..

0x3333, 2018-10-23:
Hi everyone. I implemented this feature in Espurna Firmware, but probably easy to implement in Tasmota. 

I didn't created a PR yet, but will create soon. Here is the Espurna version. 

https://github.com/0x3333/espurna/blob/multipzem/code/espurna/sensors/PZEM004TSensor.h

0x3333, 2018-10-23:
The sensor reads constantly the devices and cache the readings, when the firmware asks for a reading it returns the last one. 

Probably there is a difference in reading, but no more than 20miliseconds. I do the same magnitudes at once, all voltages, then all currents and so on...

rt400, 2018-11-18:
@0x3333 hi.. i try your code but still no work.. i don't like espurna..so conflity..
there is chance for implement in Tasmota.?

0x3333, 2018-11-18:
Well, I don't use Tasmota anymore... Don't have the time to implement, the energy driver is so complex, with several devices in the same file that the time to understand it doesn't worth.

rt400, 2018-11-18:
@0x3333 ok..
can you explain he how to make espurna firmware bin file for the 3 phaze ?
because i didnt undertsatnd how it is work..

0x3333, 2018-11-19:
Just build it, https://github.com/xoseperez/espurna/wiki/Build-and-update-from-Visual-Studio-Code-using-PlatformIO

andrethomas2, 2018-11-19:
Closing this one for now until one of the dev's have time to work on it.

adityama, 2019-06-17:
> I mean plugging all the 3 PZEM in the same hardware/soft serial, like this:
>
> [image]
>
> The diode can be a BAT45, the pullup resistor can be a 10K to 5V.
>
> But you need to program the PZEM address [...]

I was struggling to find a way to do the same. Finally I found a simple solution to poll 3 pzem004Ts with Tasmota (without changing anything to the code).

I am using PZEM004T v3.0 modules that are using Modbus-RTU.
I connected 3 pzem004 modules to ESP01 UART as per your diagram. In the module configuration I used serialBridge TX, RX (not PZEMxxx).
I set all three modules to different addresses 01, 02, 03.

Then I used SerialSend5 to send read request to module 01. It returned hex data with all the values (in hex) voltage, current and so on. Same data is send over mqtt. I configured my node-red to poll each pzem004 every 6 seconds and processed received hex data to extract required values.
Currently I am using it for 3 phases with three pzem modules. However I believe I could add more modules to it.

I have few pzem004T v1.0 modules lying around. Next thing I will use same method to read them. This needs different reading methods.
The only drawback with this method is that you don't get the Tasmota Web UI to display the readings.

clifforddw, 2019-07-21:
@adityama could you share your node red flow with us, trying to do something very similar.

adityama, 2019-07-27:
> 
> 
> @adityama could you share your node red flow with us, trying to do something very similar.

Yeah sure. Use the attached flows. Just note, I am using PZEM004T v3.0 MODBUS-RTU modules.

let me know if you have any questions regarding the node-red flows.
[3ph-mains-flows.zip](https://github.com/arendst/Sonoff-Tasmota/files/3437963/3ph-mains-flows.zip)

[232 more replies]

closed 2019-09-23 by ascillato2