arendst/Tasmota #2319: 5.12.0i webserver dead?

pascalsaul, 2018-04-01:
Since I've upgraded all the devices to 5.12.0i the webserver doesn't work anymore :(

I sometimes get the basic auth but when I press 'OK' its dead instantly.

More people with the same problem. Even a reset 1 doesn't work. Console and MQTT do work.

cmnd/sonoff/status 0
stat/sonoff/STATUS {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
stat/sonoff/STATUS1 {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.1.12/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:09:09","StartupUTC":"2018-04-01T14:54:12","Sleep":0,"BootCount":1,"SaveCount":10,"SaveAddress":"FA000"}}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"5.12.0i","BuildDateTime":"2018-04-01T16:29:32","Boot":6,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
stat/sonoff/STATUS3 {"StatusLOG":{"SerialLog":2,"WebLog":0,"SysLog":0,"LogHost":"192.168.1.18","LogPort":514,"SSId":["X","X"],"TelePeriod":30,"SetOption":["000A8009","55818000"]}}
stat/sonoff/STATUS4 {"StatusMEM":{"ProgramSize":469,"Free":532,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
stat/sonoff/STATUS5 {"StatusNET":{"Hostname":"sonoff-1723","IPAddress":"192.168.1.78","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"60:01:94:85:46:BB","Webserver":2,"WifiConfig":4}}
stat/sonoff/STATUS6 {"StatusMQT":{"MqttHost":"192.168.1.18","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8546BB","MqttUser":"mqtt","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
stat/sonoff/STATUS7 {"StatusTIM":{"UTC":"Sun Apr 01 15:03:21 2018","Local":"Sun Apr 01 15:03:21 2018","StartDST":"Tue Feb 06 00:00:00 2018","EndDST":"Tue Mar 06 12:00:00 2018","Timezone":99}}
stat/sonoff/STATUS10 {"StatusSNS":{"Time":"2018-04-01T15:03:21"}}
stat/sonoff/STATUS11 {"StatusSTS":{"Time":"2018-04-01T15:03:21","Uptime":"0T00:09:09","POWER":"0","Wifi":{"AP":1,"SSId":"X","RSSI":62,"APMac":"78:C1:A7:3D:B0:B8"}}}

Dickesplumpel, 2018-04-01:
After upgrade from  5.12.0f to 5.12.0i there is no wifi at all. I have one esp on which I first test the new firmware. Console works and shows that it can't connect. WPSconfig fails too. 
Console shows "WPSconfig active for 3 minutes" (for me in german) but it starts again after a few seconds.
Erased flash complete (4Mb) with esptool and ESP8266flasher and flashed again - no success. Flashed firmware direct new and upgrade same result - no wifi. It is an older 8266 Wemos board.

pascalsaul, 2018-04-01:
I've that also on newle flashed devices... Thought all my devices are broken. Looks like there is a serious bug...

Luckyli my upgraded devices are still reacheable.

nhondong, 2018-04-02:
I downgraded to 5.11. This release seems to work fine

Dickesplumpel, 2018-04-02:
Works on ESP12F

Dickesplumpel, 2018-04-02:
I cleaned up my platformio complete, deleted all tasmota versions in .platformio folder, downloaded 5.12.0i again, upgraded platformios libraries,  rebuild c/c++ project index, compiled new, erased flash with esptool and ESP8266flasher (complete 4Mb), flashed with esptool and the old Wemos board runs - don't know why - things have a life of their own ;-)
I will test mqtt with ds18b20 and domoticz till tomorrow to see if its stable.

pascalsaul, 2018-04-04:
@Dickesplumpel I'm always compiling complete new versions like that...

Just did it again with the latest version and same results. Even wget can't save the webpage.

`wuser@xxxx:~$ wget http://192.168.1.55
--2018-04-04 21:38:23--  http://192.168.1.55/
Connecting to 192.168.1.55:80... connected.
HTTP request sent, awaiting response... 401 Unauthorized

Username/Password Authentication Failed.
user@xxxx:~$ wget --http-user=admin --http-password=xxx http://192.168.1.55                          
--2018-04-04 21:38:58--  http://192.168.1.55/
Connecting to 192.168.1.55:80... connected.
HTTP request sent, awaiting response... 401 Unauthorized
Authentication selected: Basic realm="Login Required"
Connecting to 192.168.1.55:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2562 (2.5K) [text/html]
Saving to: âindex.html.1â

index.html.1                 0%[                                       ]       0  --.-KB/s
`
0 bytes are getting downloaded...

Dickesplumpel, 2018-04-05:
Try without username/password leave the system open first and when it works
you can begin closing one door after another and check by every step if
system works. If someone can hack my wlan router it may no problem for him
to hack my sonoffs but i see no reason why anyone should do this so my
system is intern open.

pascalsaul, 2018-04-12:
Well I'm using the user/pass since the beginning and never had a problem... more people are having issues with the latest code changes if I watch all the issues...

almostlunatic, 2018-04-12:
With v. 5.12.0k there's no problem! Good work!

pascalsaul, 2018-04-13:
> With v. 5.12.0k there's no problem! Good work!

I've updated everything to 5.12.0k and no difference...

arendst, 2018-04-13:
Make sure you enable javascript in your webbrowser as the page will only be visible after javascript has run on your client.

alton76, 2018-04-13:
> As far as I can tell I’m pretty sure JavaScript is turn on because like I
> mentioned in my previous post, I’m able to view the web browser of the old
> T1  touch without any problems, only the new ones I’m having issues with.
> Both are running same Tasmota version.
>
>
> In total I have 12 old types, 11 new types. All 11 are unreachable, but
> MQTT seem to be working.

I will look into java just to make sure, also will do an update.

pascalsaul, 2018-04-13:
@arendst Not working with Chrome, IE and even wget fails... 

And I'm pretty sure javascript is enabled otherwise half of the websites on the internet will fail...

arendst, 2018-04-13:
Right. Well as you are one of a few you would have to look for a local solution.

If pages are downloaded incomplete there might be some setting in your wifi environment not allowing local pages above a certain size.

Just to show it could work:
```
root@domus3:~# wget --http-user=admin --http-password=xxxxxx http://192.168.2.193
--2018-04-13 10:44:11--  http://192.168.2.193/
Connecting to 192.168.2.193:80... connected.
HTTP request sent, awaiting response... 401 Unauthorized
Authentication selected: Basic realm="Login Required"
Connecting to 192.168.2.193:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2674 (2.6K) [text/html]
Saving to: ‘index.html.2’

index.html.2                    100%[=======================================================>]   2.61K  --.-KB/s    in 0.03s

2018-04-13 10:44:11 (94.8 KB/s) - ‘index.html.2’ saved [2674/2674]

root@domus3:~#
```

almostlunatic, 2018-04-13:
I had never have a problem with the web server. With 5.12.0i the problem with the access was due to wifi, as soon as wifi was connected and ping was going well the access was good. But wifi was doing very poorly. 5.12.0k is excellent! I just tried it with IE, Edge, FF, Chrome and Opera - it is functioning very well. Chrome with ScriptBlock activated didn't show the AM2302 results but even Toggles worked, in Opera with JS disabled there was no temperature and humidity and Toggles didn't work but menus were working good. As it was said before, try with removed password, doesn't cost anything!

pascalsaul, 2018-04-13:
Its getting quite interesting. I'm using Cisco Aironet' with a WLC so Enterpise grade WiFi at home ;)

All WiFi devices work with webpages except Tasmota. Als ESPeasy webpage does work.

When I connect both my laptop and a random Sonoff to the router WiFi under a different SSID I do get the webpage.

So there is something bugging in a implementation of the webserver or WiFi client but what...

pascalsaul, 2018-04-28:
`stat/th_vloer/STATUS2 {"StatusFWR":{"Version":"5.12.0m","BuildDateTime":"2018-04-28T12:06:52","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
stat/th_vloer/STATUS2 {"StatusFWR":{"Version":"5.12.0m","BuildDateTime":"2018-04-28T12:50:54","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}`

@arendst I first compiled with 2_4_0 today and pointed the browser to it and after basic auth blank page and kept the browser open.
Then upgraded to the latest version with 2_4_1 and the webserver is working again and the browser did an automatic refresh. So there is a bug in 2_4_0 then...

CyBot, 2018-05-10:
Thank you! I had exactly the same problem (webserver only partially responded, or didn't respond at all/timed out), and compiling with 2.4.1 fixed it

int3l, 2018-06-21:
@pascalsaul please mark this Issue as fixed, since you have a solution with the newer core release.

pascalsaul, 2018-06-21:
Fixed with 2.4.1

closed 2018-06-21 by pascalsaul

alton76, 2020-07-27:
Fixed with 2.4.1