arendst/Tasmota #2320: no stable WLAN connection / MQTT Problems

nhondong, 2018-04-01:
Hi, 

I am using Kesen 1, 2 and 3 Chanel Touch. These are the same as the Sonoff T1, T2, T3. 
I flashed them with the v5.12.0 Release (sonoff-DE.bin)

Some devices are next to the Access Point. Some are about 5 to 7 Meters away.

On all devices I got Ping Timeouts. The most timeouts appears on the Devices a few meters away.
Because of these Timeouts (i think), i even got Problems with mqtt signals.
After Reboot it works fine for some minutes, but then it does not send or receive any mqtt command.

I there a way to get a stable connection to these devices?

kind regards, 

Nils

pascalsaul, 2018-04-01:
https://github.com/arendst/Sonoff-Tasmota/issues/2319

Looks like the same...

Dickesplumpel, 2018-04-01:
The 5.12.0f firmware runs rock-stable on 15 devices of different kind, ESP8266-01, ESP12E and F, and of course Sonoff Basics. I don't know if it is right but I have the suspicion that mqtt-sensorvalues are only send when there is a change in a value - I might totally be wrong - confusion is my second name ;-)) - mqtt-switch-commands are send immediately.

nhondong, 2018-04-01:
I am running a mqtt broker and openhab2 on a Pi.
The switches do not send mqtt states to the broker and do not receive commands from the openhab2 after a while.

nhondong, 2018-04-01:
I'll try to use the Version 5.11.
I'll report my results.

nhondong, 2018-04-02:
yesterday I downgraded to 5.11. 
Now after 9 hours, it even works fine.
So, it seems there is a bug in the 5.12 Release