arendst/Tasmota #2321: No Dual R2 option in Module confiq

ThunderMerweS, 2018-04-01:
[image]

- -     Operating System: Win 10
- -     Python Version: 3.65
- -     Sonoff model: Dual R2

    
Hello all,

Bit new to flashing firmware, but here goes.

So after doing the firmware flashing, all was good. However, when it comes to selecting the module, I do not find the Dual R2 Module, just the normal Dual

If I choose the Dual module, I only have 2 GPIO options, and if I change them both to be switches the toggle off/on doesn't do anything.

arendst, 2018-04-02:
Provide initial information before receiving an answer here.

ThunderMerweS, 2018-04-02:
"Provide initial information before receiving an answer here."

Morning, what initial information do you require that I have not stated?

Software version used is the following:

Program Version | 5.9.1
2017-11-13T21:43:41
2_3_0/1.5.3(aec24ac9)

ThunderMerweS, 2018-04-02:
I can however get it to function with using 4 Channel module. It will toggle the 2 switches fine.

arendst, 2018-04-02:
Well it's a beginning providing initial infrmation as stated in the issue when you started it.

Anyway, i'm quitte certain that your version 5.9.1 does not support the R2. Perhaps upgrading might be the next step...

ThunderMerweS, 2018-04-02:
Is there an simpler way to upgrade the version? Which is the latest version?

Or must everything be done from the onset again?

arendst, 2018-04-02:
It all depends on how you rolled it out to your devices. If you didn't compile yourself you can just download the latest release or better change your ota url to the wiki documented download site and ota upgrade your device.

If all is well your settings should be preserved...

ThunderMerweS, 2018-04-02:
Great thanks,

Updated to 5.12 and all is fine now.

This can be closed

closed 2018-04-02 by arendst