arendst/Tasmota #2322: Hue Emulation

mcfarlde, 2018-04-02:
Been using a Sonoff 4 CH Pro and a LinkNode R4 with V5.12 for some time.  I use the Hue emulation to control it from Hassio.  A recent update of Hassio changed how it connects to Hue bridges due to Hue going async.  Since this update, Hassio does not discover the tasmota nodes and the manual config that used to work, does not work anymore.  Any plans to update the hue emulation to follow what Hue now does?  https://github.com/home-assistant/home-assistant/pull/13043
02:56:12 RSL: Received Topic /status, Data Size 1, Data 0
02:56:12 RSL: Group 0, Index 1, Command STATUS, Data 0
02:56:12 RSL: STATUS = {"Status":{"Module":23,"FriendlyName":"Flame","Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
02:56:12 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":"0T00:20:51","Sleep":0,"BootCount":160,"SaveCount":4354,"SaveAddress":"F7000"}}
02:56:12 RSL: STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-02-09T14:54:51","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
02:56:12 RSL: STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":4,"SysLog":4,"LogHost":"domus1","LogPort":514,"SSId1":"58F","SSId2":"58F","TelePeriod":300,"SetOption":"55C00001"}}
02:56:12 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":485,"Free":516,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
02:56:12 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-7010","IPAddress":"10.0.1.137","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.1","Mac":"60:01:94:85:7B:62","Webserver":2,"WifiConfig":1}}
02:56:12 RSL: STATUS7 = {"StatusTIM":{"UTC":"Mon Apr 02 01:56:12 2018","Local":"Mon Apr 02 02:56:12 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
02:56:12 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-04-02T02:56:12"}}
02:56:12 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-04-02T02:56:12","Uptime":"0T00:20:51","Vcc":3.143,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"58F","RSSI":82,"APMac":"D0:17:C2:3D:A0:20"}}}02:56:12 RSL:

arendst, 2018-04-02:
Do not use hue emulation with hass but enable tasmota hass mqtt discovery for a much better experience.

emontnemery, 2018-04-02:
Should the Hue emulation be removed then?
It's quite buggy right now:
- Changing between RGB and white does not work
- Color mode is incorrectly reported
- Hue allows changing single parameters, for example only brightness, this is not working

I have been working on a patch for some of the issues I mentioned, but it seems to be a game of whack a mole..

arendst, 2018-04-02:
@emontnemery your suggested features were never supported by the Tasmota Hue emulation as it is a minor subset of the Hue implementation.

There is just no space in the chip to support Tasmota features and full Hue.

In case of Hass the Tasmota Mqtt discovery provides much more functionality than Hue did.

mcfarlde, 2018-04-02:
I will switch to MQTT.  I’m curious about MQTT discovery you mention.

closed 2018-04-02 by mcfarlde