arendst/Tasmota #2324: Sonof Tasmota  firmware 5.12.0i

JPSINHA, 2018-04-02:
The new firmware 5.12.0i is working fine with added Timer. However, it requires NTPServer for time updating. Cat it possible
1.  change default time from 1970 to any...
2. Update from system time ie. computer or mobile
3. add RTC with Sonoff 4Ch pro for time 
If any one possible what hardware and softoware require

arendst, 2018-04-02:
Or just get NTP functional....

Dickesplumpel, 2018-04-02:
When I put "timers" in the console it shows 4 Timers and 12 Timer. Are these timers conncted to "Update Zeitplan" for domoticz? This would be bloody amazing!

arendst, 2018-04-02:
Why on earth should they be related to Domoticz timers? You do not want to have different timers doing the same thing.

Domoticz timers have been working all the time as they just send Power On/Off at their expiration.

Dickesplumpel, 2018-04-02:
You're too fast - I killed that short later ;-)) 
That may be right, I'm using domoticz only since two weeks and don't know anything about its stability and still learning - i am fascinated of its "easy to useness".I've started with IOT a year ago. The last year I have used home-assistant and shot it several times when changing things in the automation.

Zoritto, 2018-04-03:
I downgraded to 5.11.1 This release seems to work fine

Dickesplumpel, 2018-04-06:
In my Network sometimes and with some routers there seems to be a problem of resolving hostnames so i set "NtpServer1 134.130.4.17"  (RWTH Aachen) in the console which works fine.