arendst/Tasmota #2325: No IP address

MimbaMonkeyHouse, 2018-04-02:
Hi all,

I've had major surge issues at home. The only devices affected seem to be the Sonoff. All POWs are down, so are most TH16s. The Basics seem to be fine. Troubleshooting has allowed me to get some devices back online by re-entering the correct WLAN info. Others are still MIA. Via the router, I can see they attempt to connect and can see their MAC addresses, but they seem unable to obtain an IP address.
Has this happened to others? What solution did people come up with?

MimbaMonkeyHouse, 2018-04-03:
Yea, no. I've cycled the router on/off multiple times. I've also tried with an alternative WLAN SSID and router and still no IP. I'm trying to find an alternative solution before taking all the devices out and reflashing them. So far I tried multiple button presses and 4 second press to no avail. 
It's a real pain to take all 15 devices away from their location. Most of them were flashed with Sonota so they will need to be soldered on...

chr00ted, 2018-05-23:
I see the same issue as well.

closed 2018-07-10 by ascillato2