arendst/Tasmota #2326: Support Kelvin unit for color temperature

renne, 2018-04-02:
I suggest to support the unit Kelvin for colour temperature by using SetOption to specify the colour temperature of the red, green, blue, cold white and warm white strip. If the desired color temperature is between cold white and warm white use cross-fade. If the colour temperature remains under or exceeds the range off the white stripes I suggest to use [Tanner Hellands algorithm for Kelvin-to-RGB conversion](http://www.tannerhelland.com/4435/convert-temperature-rgb-algorithm-code/).

arendst, 2018-04-02:
Well give it a try and let me know your experience using both a ws2812 led string and some pwm controlled  led.. Verify if you manage to get the same color temperature using the same kelvin value...

renne, 2018-06-06:
On my ToDo list ...

closed 2018-08-05 by stale[bot]