arendst/Tasmota #2327: 5.12.0d till 5.12.0i unreliable on NodeMcu. Working fine on every other device!

Jason2866, 2018-04-02:
I have 2 NodeMcu with this behaviour. If i go back to version 5.12.0b the work fine.
The devices are laggy and are doing reconnects all the time. With the same compiled Version
5.12.0d or 5.12.0h on every other device (Sonoff S20, Sonoff Basic, WemosMini)
it works perfect. Responsable and not even one reconnect.
What is the difference between a NodeMCU and the rest?
And what has been changed from the earlier version 5.12.0b to the newer ones that causes this?
I have built all versions on Atom.io with absolut the same setup.

10:19:15 MQT: tele/sonoff-87786C/SENSOR = {"Time":"2018-04-02T10:19:15","Switch2":"OFF","AM2301":{"Temperature":21.8,"Humidity":40.3},"BMP280":{"Temperature":22.1,"Pressure":973.0,"SeaPressure":1016.6},"TempUnit":"C"}
10:24:16 MQT: tele/sonoff-87786C/STATE = {"Time":"2018-04-02T10:24:16","Uptime":"5T22:23:41","Vcc":2.784,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":88,"APMac":"00:A0:57:1E:69:50"}}
10:24:16 MQT: tele/sonoff-87786C/SENSOR = {"Time":"2018-04-02T10:24:16","Switch2":"OFF","AM2301":{"Temperature":21.9,"Humidity":40.4},"BMP280":{"Temperature":22.3,"Pressure":973.0,"SeaPressure":1016.6},"TempUnit":"C"}
10:29:17 MQT: tele/sonoff-87786C/STATE = {"Time":"2018-04-02T10:29:17","Uptime":"5T22:28:42","Vcc":2.770,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":90,"APMac":"00:A0:57:1E:69:50"}}
10:29:17 MQT: tele/sonoff-87786C/SENSOR = {"Time":"2018-04-02T10:29:17","Switch2":"OFF","AM2301":{"Temperature":22.0,"Humidity":40.5},"BMP280":{"Temperature":22.4,"Pressure":972.9,"SeaPressure":1016.5},"TempUnit":"C"}
10:34:18 MQT: tele/sonoff-87786C/STATE = {"Time":"2018-04-02T10:34:18","Uptime":"5T22:33:43","Vcc":2.784,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":90,"APMac":"00:A0:57:1E:69:50"}}
10:34:18 MQT: tele/sonoff-87786C/SENSOR = {"Time":"2018-04-02T10:34:18","Switch2":"OFF","AM2301":{"Temperature":22.1,"Humidity":40.6},"BMP280":{"Temperature":22.6,"Pressure":972.9,"SeaPressure":1016.5},"TempUnit":"C"}
10:39:19 MQT: tele/sonoff-87786C/STATE = {"Time":"2018-04-02T10:39:19","Uptime":"5T22:38:44","Vcc":2.784,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":92,"APMac":"00:A0:57:1E:69:50"}}
10:39:19 MQT: tele/sonoff-87786C/SENSOR = {"Time":"2018-04-02T10:39:19","Switch2":"OFF","AM2301":{"Temperature":22.2,"Humidity":40.9},"BMP280":{"Temperature":22.7,"Pressure":972.9,"SeaPressure":1016.4},"TempUnit":"C"}
10:44:20 MQT: tele/sonoff-87786C/STATE = {"Time":"2018-04-02T10:44:20","Uptime":"5T22:43:45","Vcc":2.784,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":88,"APMac":"00:A0:57:1E:69:50"}}
10:44:21 MQT: tele/sonoff-87786C/SENSOR = {"Time":"2018-04-02T10:44:20","Switch2":"OFF","AM2301":{"Temperature":22.3,"Humidity":41.0},"BMP280":{"Temperature":23.0,"Pressure":972.8,"SeaPressure":1016.4},"TempUnit":"C"}
10:49:21 MQT: tele/sonoff-87786C/STATE = {"Time":"2018-04-02T10:49:21","Uptime":"5T22:48:46","Vcc":2.784,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":88,"APMac":"00:A0:57:1E:69:50"}}
10:49:22 MQT: tele/sonoff-87786C/SENSOR = {"Time":"2018-04-02T10:49:21","Switch2":"OFF","AM2301":{"Temperature":22.5,"Humidity":41.1},"BMP280":{"Temperature":23.0,"Pressure":972.7,"SeaPressure":1016.3},"TempUnit":"C"}
10:54:23 MQT: tele/sonoff-87786C/STATE = {"Time":"2018-04-02T10:54:23","Uptime":"5T22:53:48","Vcc":2.770,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":88,"APMac":"00:A0:57:1E:69:50"}}
10:54:23 MQT: tele/sonoff-87786C/SENSOR = {"Time":"2018-04-02T10:54:23","Switch2":"OFF","AM2301":{"Temperature":22.7,"Humidity":41.3},"BMP280":{"Temperature":23.1,"Pressure":972.7,"SeaPressure":1016.3},"TempUnit":"C"}
10:59:24 MQT: tele/sonoff-87786C/STATE = {"Time":"2018-04-02T10:59:24","Uptime":"5T22:58:49","Vcc":2.784,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":86,"APMac":"00:A0:57:1E:69:50"}}
10:59:24 MQT: tele/sonoff-87786C/SENSOR = {"Time":"2018-04-02T10:59:24","Switch2":"OFF","AM2301":{"Temperature":22.8,"Humidity":41.3},"BMP280":{"Temperature":23.2,"Pressure":972.8,"SeaPressure":1016.3},"TempUnit":"C"}
11:02:00 MQT: tele/sonoff-87786C/UPTIME = {"Time":"2018-04-02T11:02:00","Uptime":"5T23:01:25"}
11:04:25 MQT: tele/sonoff-87786C/STATE = {"Time":"2018-04-02T11:04:25","Uptime":"5T23:03:50","Vcc":2.784,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":88,"APMac":"00:A0:57:1E:69:50"}}
11:04:25 MQT: tele/sonoff-87786C/SENSOR = {"Time":"2018-04-02T11:04:25","Switch2":"OFF","AM2301":{"Temperature":22.9,"Humidity":41.4},"BMP280":{"Temperature":23.4,"Pressure":972.7,"SeaPressure":1016.3},"TempUnit":"C"}
11:05:26 CMD: Status 0
11:05:26 MQT: stat/sonoff-87786C/STATUS = {"Status":{"Module":18,"FriendlyName":[],"Topic":"sonoff-87786C","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
11:05:26 MQT: stat/sonoff-87786C/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"5T23:04:51","StartupUTC":"2018-03-27T10:00:35","Sleep":1,"BootCount":105,"SaveCount":1677,"SaveAddress":"FA000"}}
11:05:26 MQT: stat/sonoff-87786C/STATUS2 = {"StatusFWR":{"Version":"5.12.0i","BuildDateTime":"2018-03-27T11:41:01","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
11:05:26 MQT: stat/sonoff-87786C/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["Jason_Home_WLAN",""],"TelePeriod":300,"SetOption":["00000009","55818000"]}}
11:05:26 MQT: stat/sonoff-87786C/STATUS4 = {"StatusMEM":{"ProgramSize":480,"Free":520,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
11:05:26 MQT: stat/sonoff-87786C/STATUS5 = {"StatusNET":{"Hostname":"sonoff-87786C-6252","IPAddress":"192.168.2.139","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"EC:FA:BC:87:78:6C","Webserver":2,"WifiConfig":3}}
11:05:26 MQT: stat/sonoff-87786C/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.114","MqttPort":1883,"MqttClientMask":"sonoff-87786C","MqttClient":"sonoff-87786C","MqttUser":"openhabian","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:05:26 MQT: stat/sonoff-87786C/STATUS7 = {"StatusTIM":{"UTC":"Mon Apr 02 09:05:26 2018","Local":"Mon Apr 02 11:05:26 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99}}
11:05:26 MQT: stat/sonoff-87786C/STATUS10 = {"StatusSNS":{"Time":"2018-04-02T11:05:26","Switch2":"OFF","AM2301":{"Temperature":23.0,"Humidity":41.3},"BMP280":{"Temperature":23.4,"Pressure":972.7,"SeaPressure":1016.3},"TempUnit":"C"}}
11:05:26 MQT: stat/sonoff-87786C/STATUS11 = {"StatusSTS":{"Time":"2018-04-02T11:05:26","Uptime":"5T23:04:51","Vcc":2.768,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":88,"APMac":"00:A0:57:1E:69:50"}}}

arendst, 2018-04-02:
Ask the NodeMcu seller.

Jason2866, 2018-04-02:
Thx for your comment. I will replace the NodeMCUs (with know working WemosMinis)...

closed 2018-04-02 by Jason2866