arendst/Tasmota #2328: Sonoff - NoT showing UPDATED temperature with mqtt command

L00000813, 2018-04-02:
Sonoff TH16

I have my Sonoff setup and configured.

I am using the latest software version (downloaded from Github today).

I subscribe to the sonoff (working)
I run a command to get the status of the sensor (working)

Issue:
I dip the sensor in water and continuously run status command but the same temperature is returned.
The temperature reading is not changing - IN MQTT query.

However:
If I have the sonoff (device) web interface open - the temperature does change, both on the web interface and when I run status commands using mqtt. 

In a nutshell: 
If I have a web interface open for the sonoff, everything updates instantly as the temperature reading changes, If the web interface is NOTopen - the temperature reading in MQTT stays constant, even if the temperature has changed.

I need to be able to run status commands and get the current live temperature?

Thanks.

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

arendst, 2018-04-02:
Provide output of command ``status 0``!

L00000813, 2018-04-02:
Sorry -I did not provide it as I did not think it was relevant here.

Remember: If I have the live sonoff interface connection to the sonoff device, the temperature does update when I run mqtt status commands. 

But if I don't have the web interface open (connection), the temperature reading does not update through mqtt status commands.

14:03:11 CMD: status 0
14:03:11 MQT: stat/sensor/STATUS = {"Status":{"Module":4,"FriendlyName":["Sonoff"],"Topic":"sensor","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
14:03:11 MQT: stat/sensor/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sensor.ino.bin","RestartReason":"Software/System restart","Uptime":"0T02:59:49","StartupUTC":"2018-04-02T10:03:22","Sleep":0,"BootCount":4,"SaveCount":18,"SaveAddress":"FA000"}}
14:03:11 MQT: stat/sensor/STATUS2 = {"StatusFWR":{"Version":"5.12.0i","BuildDateTime":"2018-04-02T10:58:13","Boot":6,"Core":"2_4_0_RC2","SDK":"2.1.0(7106d38)"}}
14:03:11 MQT: stat/sensor/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["Wi-Fi-Raspberry","vodafone-7AB7"],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
14:03:11 MQT: stat/sensor/STATUS4 = {"StatusMEM":{"ProgramSize":523,"Free":480,"Heap":15,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
14:03:11 MQT: stat/sensor/STATUS5 = {"StatusNET":{"Hostname":"sensor-3288","IPAddress":"192.168.1.14","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"EC:FA:BC:87:0C:D8","Webserver":2,"WifiConfig":3}}
14:03:11 MQT: stat/sensor/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.15","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_870CD8","MqttUser":"","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
14:03:11 MQT: stat/sensor/STATUS7 = {"StatusTIM":{"UTC":"Mon Apr 02 13:03:11 2018","Local":"Mon Apr 02 14:03:11 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
14:03:11 MQT: stat/sensor/STATUS10 = {"StatusSNS":{"Time":"2018-04-02T14:03:11","DS18B20":{"Temperature":17.9},"TempUnit":"C"}}
14:03:11 MQT: stat/sensor/STATUS11 = {"StatusSTS":{"Time":"2018-04-02T14:03:11","Uptime":"0T02:59:49","Vcc":3.462,"POWER":"OFF","Wifi":{"AP":2,"SSId":"vodafone-7AB7","RSSI":72,"APMac":"78:6A:89:D2:7A:C8"}}}

arendst, 2018-04-02:
As your TelePeriod is set for 300 seconds you only receive an update every 5 minutes. If you want more updates change teleperiod to a lower value. Bottom is 10 seconds.

You might want to consider to update your ESP/Arduino library too as the 2.4.0-RC2 is way overdue and not supported by Tasmota.

L00000813, 2018-04-02:
I am not using tele.. I just want to get details when i request them ??

My subscribe command is:     mosquitto_sub ....        -t stat/sensor/# -v

My pub command is:              mosquitto_pub.....        -t cmnd/sensor/status -m 10

arendst, 2018-04-02:
You won't get details if you want them for the simple reason that it takes time for the sensor to gather a stable temperature. 

Teleperiod initiates a new read before it provides the info to the user. Some sensors types need more time than others.

You won't get real time temperatures with Tasmota.

L00000813, 2018-04-02:
Ok, But why does it show real-time temperature in the sonoff web interface? The web page temperature changes if temperature changes.

And - if I have the web interface open, the mqtt does give me back the live temperature?

arendst, 2018-04-12:
Will be fixed in next pre-release.

closed 2018-04-12 by arendst