arendst/Tasmota #2330: Missing pictures in "S20 Smart Socket" Wiki Page

pReya, 2018-04-02:
Unfortunately most of the pictures from the [Wiki page about the "S20 Smart Socket"](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-S20-Smart-Socket) are not working anymore. Seems they were deleted at the image hoster. Maybe they can be reuploaded somewhere else? Why not put them in the repo? I'd love to help, but I don't have the original pictures.

curzon01, 2018-04-02:
try asking author [BrutalBirdie](https://github.com/BrutalBirdie) who has made the first version with pic-upload.de references, see [page history](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-S20-Smart-Socket/_history)

BrutalBirdie, 2018-04-04:
If I can find my S20 Switches I will take new pictures asap.

BrutalBirdie, 2018-04-04:
@pReya we could use the pictures from the official Sonoff S20 CE-EMC Report until someone someone (me) can make pictures again.
https://www.itead.cc/wiki/images/3/38/Sonoff_S20_CE-EMC_Report.pdf Page 46

pReya, 2018-04-04:
I'm flashing my S20 today or tomorrow. I'll take pictures along the way, so don't worry about finding yours. I just thought maybe you have the pictures still on your hard drive somewhere.

pReya, 2018-04-30:
Sorry, I never got around to take pictures of mine. Thanks for uploading yours again.

closed 2018-04-30 by pReya