arendst/Tasmota #2333: Make WiFI/WLAN static IP address/gateway/DNS configurable

knopserl, 2018-04-02:
Not sure if this was already requested (I have at least searched for it).
Would it be possible to extend the WLAN config dialog to include a static IP address/gateway/DNS to store it as part of the configuration?
I dont like DHCP and flashing for a new IP address is also not convinient.

BTW: The is a branch from Stefan Bode which includes a port PCF8574 port extender, which I'm requireing like Stefan and others. Would be nice if that could be included in one of the next original Tasmota project version.

reloxx13, 2018-04-02:
how about use commands?...

knopserl, 2018-04-02:
you mean the console in the Web-UI? Can I change it with the conole commands and does it persist the WLAN settings (IP/GW/DNS)?
It would be anyhow nice to have this three parameters in the Web-UI (at least for the primary AP1 for the alternative would be even more convinient)

Dickesplumpel, 2018-04-03:
Yes it persist in the WLAN settings.

Frogmore42, 2018-04-04:
Has been discussed recently.
https://github.com/arendst/Sonoff-Tasmota/issues/2219

The Web UI is there for simple things that most people need/would find useful/not confusing.  Static IP is not something most people should be using.  If you are one of the few who NEED it, it is supported directly from the console (web and serial).

What do you have against DHCP?

Aileron79, 2018-04-27:
There is absolutely nothing confusing about static IP addresses. In a proper home automation environment it makes sense to assign static ip addresses to various switches to make other hardware easily communicate with that switch. I agree that this is a required setting for any GUI.

Frogmore42, 2018-04-27:
If you feel it is required, feel free to fork the code and add it.  While there are some people who use static IP addresses, most do not, nor is there an actual need to do so in most cases.  Perhaps you have a case where you feel it is indispensable.  That is the beauty of open source, you are free to make it do what you feel is important.  I don't control what happens to Tasmota, but not even Theo (who created it and maintains it) can stop you from doing what you feel is required.

Now, if your goal is to convince Theo that static IP has to be in the GUI and you believe it is his requirement to fix, I am not sure you have done that so far (but don't let me stop you from trying :-)

As far as your last statement goes 

 `In` a proper home automation environment it makes sense to assign static ip addresses to various switches to make other hardware easily communicate with that switch

dikodahan, 2018-04-27:
I 100% agree!

Frogmore42, 2018-04-27:
Posted before I finished.  I use MQTT, none of my devices need to know the IP address of anything other than the broker.  Are you saying my environment is not "proper"?

Now, my router automatically assigns the same IP address to a particular device each time it requests one.  It also has the capability of assigning a real static IP address to specific MAC addresses.  I use both of these capabilities.  Generally, I find that the mostly static IP addresses are more than adequate.  Now, if you are using HTTP or something that is directly node to node, they do need to know how to talk to each other.  But, static IP addresses are so late 80s, join the new millennium already in progress :-)

closed 2018-06-26 by stale[bot]