arendst/Tasmota #2334: No commits authors credits ?

soif, 2018-04-02:
@arendst

BTW : Why every commits submitted from my (or from any other contributors) Pull-Requests are systematically included **without** the original author's credit (always changed to you) ?

Is this really intended? 
Do you need help with Git merging / cherry-picking ? 

Keep Up the good work. Best regards 🍺

arendst, 2018-04-02:
Itls definitly intended for the simple reason that until now I only had one pr tha could go in production without having to change a thing.

Most pr are made by users who want to solve their issue without full testing and keeping in mind the consequences of their implementation.

When their pr is a major change their credits have been included.

soif, 2018-04-06:
I can't speak for others, but here is a *very basic example* in  my #1983 PR : 
---> [My changes ](https://github.com/arendst/Sonoff-Tasmota/pull/1983/files) **VS** your following [commit](https://github.com/arendst/Sonoff-Tasmota/commit/9b5ae54350868ab4023a5fc5345931c88b15ffe6).

What is exactly the difference "_tha could go in production without having to change a thing._" ?

Is this really a PR " _made by users who want to solve their issue without full testing and keeping in mind the consequences of their implementation._" ?

Just **trying to Understand** what was wrong.....

closed 2018-08-06 by stale[bot]

amayii0, 2019-02-06:
@soif I found this after my picture was used without credit in chapter "GPIO Locations" : https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Basic
Taken from "Step 1: Adding a Serial Header to the Sonoff" : https://www.instructables.com/id/Use-Homie-Firmware-to-Drive-Sonoff-Switch-Module-E/

Sharing is caring, but... :-)