arendst/Tasmota #2336: Sonoff sending "ON" command upon network reconnection

Caswell1000, 2018-04-02:
Thanks for the great software, I flashed tasmota 5.11.1  on the sonoff basic. Everything works great, the issue I have is that when the sonoff reconnects to the router and cloudmqtt, it sends a command that switches the sonoff "ON", I tried switching off all retained statuses. How can I disable this as I use the sonoff to open and close my garage. The console dump is shown below, at 18:04, there is LWT (retained) , followed by cmnd/sonofagarage/POWER,  I suspect they trigger the garage to open, it happens after a router reboot, or when the sonoff regains connection. How can I get rid of that, I have removed birth and will messages on home assistant. I use home assistant as my dashboard and cloudmqtt as my broker with mqtt buddy as an app in android and iPhone, the command in both phones is cmnd/sonofagarage/power with Toggle as payload. My homeassistant retain flags are false and the QOS  is 0, mqtt buddy has no place to modify QOS . Everything works well except the sporadic operations. 

```
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP1 FamilyRouter in mode 11N as sonofagarage-7402...
00:00:11 WIF: Connected
00:00:11 DNS: Initialized
00:00:11 HTP: Web server active on sonofagarage-7402.local with IP address 10.0.0.4
18:03:48 MQT: Attempting connection...
18:03:48 MQT: Connect failed to m10.cloudmqtt.com:xxxxx(I hid this) , rc -2. Retry in 10 sec
18:03:59 MQT: Attempting connection...
18:04:00 MQT: Connected
18:04:00 MQT: tele/sonofagarage/LWT = Online (retained)
18:04:00 MQT: cmnd/sonofagarage/POWER = 
18:04:00 MQT: tele/sonofagarage/INFO1 = {"Module":"Sonoff Basic","Version":"5.11.1","FallbackTopic":"MrC","GroupTopic":"sonoffs"}
18:04:00 MQT: tele/sonofagarage/INFO2 = {"WebServerMode":"Admin","Hostname":"sonofagarage-7402","IPAddress":"10.0.0.4"}
18:04:00 MQT: tele/sonofagarage/INFO3 = {"RestartReason":"Software/System restart"}
18:04:01 MQT: stat/sonofagarage/RESULT = {"POWER":"ON"}
18:04:01 MQT: stat/sonofagarage/POWER = ON
18:04:01 MQT: stat/sonofagarage/RESULT = {"POWER":"ON"}
18:04:01 MQT: stat/sonofagarage/POWER = ON
18:04:04 MQT: stat/sonofagarage/RESULT = {"POWER":"OFF"}
18:04:04 MQT: stat/sonofagarage/POWER = OFF
18:13:36 APP: Serial logging disabled
```

roguestreak, 2018-04-03:
What have you got "PowerOnState" and "PowerRetain" set to?  Probably not it, but worth checking.

Caswell1000, 2018-04-03:
I changed poweronstate to 1, powerretain to 0, I had the same issue after rebooting the router, all three sonoffs boot to "on" even after changing them

Frogmore42, 2018-04-03:
Poweronstate 1
Says turn on the relay on power on

You probably want poweronstate 0
That says turn off the relay on power on.

Caswell1000, 2018-04-03:
I changed all of them, the first two are on pulsetime xx so they toggle  and the last on pulsetime 0 so it's on and off, results are as follows starting with the two that toggle

```
00:00:00 Project sonoff Sonoff (Topic sonofagarage, Fallback MrC, GroupTopic sonoffs) Version 5.11.1
00:00:00 WIF: Connecting to AP2 indebuurt2 in mode 11N as sonofagarage-7402...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP1 FamilyRouter in mode 11N as sonofagarage-7402...
00:00:11 WIF: Connected
00:00:11 DNS: Initialized
00:00:11 HTP: Web server active on sonofagarage-7402.local with IP address 10.0.0.4
06:29:54 MQT: Attempting connection...
06:29:54 MQT: Connected
06:29:54 MQT: tele/sonofagarage/LWT = Online (retained)
06:29:54 MQT: cmnd/sonofagarage/POWER = 
06:29:54 MQT: tele/sonofagarage/INFO1 = {"Module":"Sonoff Basic","Version":"5.11.1","FallbackTopic":"MrC","GroupTopic":"sonoffs"}
06:29:54 MQT: tele/sonofagarage/INFO2 = {"WebServerMode":"Admin","Hostname":"sonofagarage-7402","IPAddress":"10.0.0.4"}
06:29:54 MQT: tele/sonofagarage/INFO3 = {"RestartReason":"Software/System restart"}
06:29:55 MQT: stat/sonofagarage/RESULT = {"POWER":"ON"}
06:29:55 MQT: stat/sonofagarage/POWER = ON
06:29:55 MQT: stat/sonofagarage/RESULT = {"POWER":"ON"}
06:29:55 MQT: stat/sonofagarage/POWER = ON
06:29:58 MQT: stat/sonofagarage/RESULT = {"POWER":"OFF"}
06:29:58 MQT: stat/sonofagarage/POWER = OFF
```
Another toggler

```

00:00:00 Project sonoff Sonoff (Topic Testchip, Fallback Testchip, GroupTopic sonoffs) Version 5.11.1
00:00:00 WIF: Connecting to AP2 indebuurt2 in mode 11N as Testchip-0040...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP1 FamilyRouter in mode 11N as Testchip-0040...
00:00:11 WIF: Connected
00:00:11 DNS: Initialized
00:00:11 HTP: Web server active on Testchip-0040.local with IP address 10.0.0.5
06:29:53 MQT: Attempting connection...
06:29:53 MQT: Connected
06:29:53 MQT: /tele/Testchip/LWT = Online (retained)
06:29:53 MQT: /cmnd/Testchip/POWER = 
06:29:53 MQT: /tele/Testchip/INFO1 = {"Module":"Sonoff Basic","Version":"5.11.1","FallbackTopic":"Testchip","GroupTopic":"sonoffs"}
06:29:53 MQT: /tele/Testchip/INFO2 = {"WebServerMode":"Admin","Hostname":"Testchip-0040","IPAddress":"10.0.0.5"}
06:29:53 MQT: /tele/Testchip/INFO3 = {"RestartReason":"Software/System restart"}
06:29:54 MQT: stat/Testchip/RESULT = {"POWER":"ON"}
06:29:54 MQT: stat/Testchip/POWER = ON
06:29:54 MQT: stat/Testchip/RESULT = {"POWER":"OFF"}
06:29:54 MQT: stat/Testchip/POWER = OFF
06:29:54 MQT: /stat/Testchip/RESULT = {"POWER":"OFF"}
06:29:54 MQT: /stat/Testchip/POWER = OFF

```

The one below is on pulsetime 0

```
00:00:15 WIF: Connected
00:00:15 DNS: Initialized
00:00:15 HTP: Web server active on tsakibed-1230.local with IP address 10.0.0.6
06:29:51 MQT: Attempting connection...
06:29:52 MQT: Connected
06:29:52 MQT: tele/tsakibed/LWT = Online (retained)
06:29:52 MQT: cmnd/tsakibed/POWER = 
06:29:52 MQT: tele/tsakibed/INFO1 = {"Module":"Sonoff Basic","Version":"5.11.1","FallbackTopic":"tsakibed","GroupTopic":"sonoffs"}
06:29:52 MQT: tele/tsakibed/INFO2 = {"WebServerMode":"Admin","Hostname":"tsakibed-1230","IPAddress":"10.0.0.6"}
06:29:52 MQT: tele/tsakibed/INFO3 = {"RestartReason":"Software/System restart"}
06:29:53 MQT: stat/tsakibed/RESULT = {"POWER":"ON"}
06:29:53 MQT: stat/tsakibed/POWER = ON
06:29:53 MQT: stat/tsakibed/RESULT = {"POWER":"OFF"}
06:29:53 MQT: stat/tsakibed/POWER = OFF
06:29:53 MQT: stat/tsakibed/RESULT = {"POWER":"ON"}
06:29:53 MQT: stat/tsakibed/POWER = ON
06:29:53 MQT: stat/tsakibed/RESULT = {"POWER":"OFF"}
06:29:53 MQT: stat/tsakibed/POWER = OFF
06:29:53 MQT: stat/tsakibed/RESULT = {"POWER":"OFF"}
06:29:53 MQT: stat/tsakibed/POWER = OFF

```

They all go through stat/xxx/power on,  so even with poweronstate at  0 the still go through on upon reboot, how do I prevent any reboot commands?

Caswell1000, 2018-04-03:
```
06:29:53 MQT: /tele/Testchip/LWT = Online (retained)
```

COULD THIS RETAINED LWT BE AN ISSUE, HOW DO I DISABLE IT?

Frogmore42, 2018-04-03:
That is the Last Will and Testament of the device.  It gets sent when the mqtt server has not heard from the device in awhile.  It should be retained to function and is not the problem you seek.

You are not going to be able to prevent reboots.  They happen, if for no other reason than power failures.  Something is misconfigured.  I have quite a few Sonoffs and none of them turn on when they reboot or reconnect.

You probably need to completely reset (erase all flash) on the device and then reflash and start with a new name and don't configure it in your HA.  Then try rebooting and reconnecting and see if you still have the problem.  Once you have verified it works correctly, you can add it to your HA.  If it now starts misbehaving, you know what part is causing the problem.

You have a lot of parts to your system, so it might not be easy getting it working.

Caswell1000, 2018-04-03:
Thanks, I am also suspecting the ha connection as I was using the sonofagarage with no hassles, I will reflash, but I am not sure if I know how to erase the flash, I will have to look for that somewhere.  Before flashing I will start by changing the topics and removing the sonoffs in my configuration.yaml. Will feedback later.

Caswell1000, 2018-04-03:
```
00:00:00 Project sonoff Sonoff (Topic sonofagarage1, Fallback sonofagarage1, GroupTopic sonoffs) Version 5.11.1
00:00:00 WIF: Connecting to AP2 indebuurt2 in mode 11N as sonofagarage1-7402...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP1 FamilyRouter in mode 11N as sonofagarage1-7402...
00:00:12 WIF: Connected
00:00:12 DNS: Initialized
00:00:12 HTP: Web server active on sonofagarage1-7402.local with IP address 10.0.0.4
20:22:05 MQT: Attempting connection...
20:22:06 MQT: Connected
20:22:06 MQT: tele/sonofagarage1/LWT = Online (retained)
20:22:06 MQT: cmnd/sonofagarage1/POWER = 
20:22:06 MQT: tele/sonofagarage1/INFO1 = {"Module":"Sonoff Basic","Version":"5.11.1","FallbackTopic":"sonofagarage1","GroupTopic":"sonoffs"}
20:22:06 MQT: tele/sonofagarage1/INFO2 = {"WebServerMode":"Admin","Hostname":"sonofagarage1-7402","IPAddress":"10.0.0.4"}
20:22:06 MQT: tele/sonofagarage1/INFO3 = {"RestartReason":"Software/System restart"}
20:22:07 MQT: stat/sonofagarage1/RESULT = {"POWER":"OFF"}
20:22:07 MQT: stat/sonofagarage1/POWER = OFF

```
Changed it from sonofagarage to sonofagarage 1, no mqtt app and no HA, it no longer cycles via "on",  Ma HA install is problematic, so I will reflash it with Hassbian because my manual install gave me headaches anyway, I will let you know @Frogmore42, thanks for your assistance.

Caswell1000, 2018-04-03:
The  app works fine and a reboot after the app linking also results in good results, I am now 99percent sure that it was the ha setup, I'm pretty tired now, so I will flash hassbian some other time, thanks for assisting, at least the problem has been isolated, so I will close the issue.

closed 2018-04-03 by Caswell1000