arendst/Tasmota #2337: Home Assistant, Sonoff and local switch (gpio14 and gnd) issue

Dev0003, 2018-04-02:
Hello Everybody
In the recent days i have been struggling to get my Sonoff Basic to work locally with the Gpio14 switch and the on board push-button. Both the GPIO14 switch and on board push-button only work momentarily (a second) when activated and back to OFF state. I'm not sure why? I am using a pull down 10k resistor and 33pf capacitor, as well as GPIO14 sensor is set to 9.

However if i control my sonoff's from Home Assistant (Hassio) , the relay switches perfectly on command with state change.

This is the sample code im using for sonoffs:
  platform: mqtt
  name: "Garage Light"
  state_topic: "stat/sonoff8/POWER"
  command_topic: "cmnd/sonoff8/POWER"
  optimistic: false
  payload_on: "ON"
  payload_off: "OFF"
  qos: 1
  retain: true

Please help!

Frogmore42, 2018-04-02:
You didn't include any of the requested information, which makes it harder to say for sure what is wrong.  But here are some things to check.  Normally a GPIO has a pull UP (resistor between 3V3 and the GPIO) not a pull DOWN (resistor between GPIO and GND).  Perhaps you just used the wrong word, because it probably wouldn't work at all with a pull down.  The other thing is that you have retain: true (not clear where you set that, but that might be what is causing the issue).  The requested log showing the error occur would help know for sure.

Leseratte10, 2018-04-03:
A pull down resistor would work just well, like a pull up resistor. If the pull resistor was the problem, the Sonoff would switch randomly (unrelated to a button press) or not at all. 
I remeber reading about a problem with HomeAssistant overwriting the current state of a Sonoff somewhere in the past, but I don't remember if it was here in the issue tracker or elsewhere. 

I would check both the Sonoff log and the Home Assistant log to find out who triggers the state change.

Dev0003, 2018-04-03:
Hi All

Thanks for the feedback, I'm sure the resistor is not the issue.
I tried removing the host IP and topic from the sonoff firmware/app , as if the sonoff was not part of home assistants configuration. The sonoff's push-button and external switch works perfectly and retains it state.
The problems is when it is part of home assistants configuration, Below is part of a the log file from the Sonoff firmware. You can see when i toggle the push-button or external switch to a ON state, it is overwritten to OFF by Mqt..I think...?

06:16:33 MQT: Attempting connection...
06:16:33 MQT: Connected
06:16:33 MQT: tele/sonoff8/LWT = Online (retained)
06:16:33 MQT: cmnd/sonoff8/POWER = 
06:16:33 MQT: stat/sonoff8/RESULT = {"POWER":"OFF"}
06:16:33 MQT: stat/sonoff8/POWER = OFF (retained)
06:16:34 MQT: cmnd/sonoff8/POWER1 = ON
06:16:34 MQT: stat/sonoff8/RESULT = {"POWER":"ON"}
06:16:34 MQT: stat/sonoff8/POWER = ON (retained)
06:16:34 MQT: Attempting connection...
06:16:34 MQT: Connected
06:16:34 MQT: tele/sonoff8/LWT = Online (retained)
06:16:34 MQT: cmnd/sonoff8/POWER = 
06:16:34 MQT: stat/sonoff8/RESULT = {"POWER":"OFF"}
06:16:34 MQT: stat/sonoff8/POWER = OFF (retained)
06:16:34 MQT: cmnd/sonoff8/POWER1 = ON
06:16:34 MQT: stat/sonoff8/RESULT = {"POWER":"ON"}
06:16:34 MQT: stat/sonoff8/POWER = ON (retained)
06:16:35 MQT: Attempting connection...
06:16:35 MQT: Connected

Frogmore42, 2018-04-03:
This is probably the problem
06:16:34 MQT: stat/sonoff8/POWER = ON (retained)
The state is being retained by something.  You HA software or mqtt is configured incorrectly it seems.

Dev0003, 2018-04-03:
I'm also convinvced that my configuration on HA is incorrect, don't know
where to look.....

However I do remember , switching on "PowerRetain" to 1 within the Sonoff
console , because HA on toggle command was automatically returning to OFF.
I could only get this right by using this command within Sonoff.

Thanks.

Dev0003, 2018-04-03:
Solved, Sonoff Client ID was same as MQTT broker client ID...:) So i changed it and everything works.....yay

closed 2018-04-03 by Dev0003