arendst/Tasmota #2338: AP mode network security key

jepean, 2018-04-02:
Hi,

I flashed Tasmota using ESPtool and hit the button to go to AP mode. But when I try to connect to the AP, I am prompted for a network security key? I have no idea what this key is. Where do I find it?

reloxx13, 2018-04-03:
Your wifi password...

jepean, 2018-04-03:
Seriously? Am I that stupid? So sorry!

reloxx13, 2018-04-03:
shit happens :D

reloxx13, 2018-04-03:
but i cannot see "secruity key" in the english translation file, are you sure you flashed tasmota on it? sounds like it is the orginal firmware.

EN Translation:
https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/language/en-GB.h

jepean, 2018-04-04:
I have no idea what happened, but it must have been some kind of glitch in my computer because when I tried the <Four button presses -> AP mode> today it worked fine. No password promt as I expected, and I was able to configure the Sonoff's wifi settings through the browser.

Thank you for your replies!

closed 2018-04-04 by jepean