arendst/Tasmota #2340: Support - iTead Sonoff Pow R2

mirino75, 2018-04-03:
Hi,

iTead Sonoff Pow **R2** - When will the device be supported? Not yet in the supported list ...

thx
mirino

Merwenus, 2018-04-19:
anyone made it work?

arendst, 2018-04-19:
Still waiting for mine.

As I said before pls try selecting module S31 as it's probably the same pin layout and it definitely uses the same energy monitoring chip (CSE7766) as the S31.

arendst, 2018-04-20:
UPDATE:

Received mine today and as expected it works "out of the box" when configured as module S31.

Important note: The power connection of the Pow R2 is different from to previous Pow so be carful if you replace an old Pow with a R2 version:
```
- Old version: Lout-E-E-N-N-Lin
- New R2 version: Lout-E-E-Lin-N-N
```
So if you connect it wrong it will blow the fuse as Lin will connected to N ;-)

To make use as easy as possible I'll add the Sonoff Pow R2 as new module anyway.

closed 2018-04-20 by arendst