arendst/Tasmota #2341: Sonoff DEV-not working sensors, GPIOs configuration

Hugo477, 2018-04-03:
A few comments in one:
1. 
There are GPIOs 1,3,4,5,12,13,14 available on Sonoff DEV board.
In configuration UI are 1,3,4,5,12,14 only. Did I missed something, or it is the correct state?

2. 
On Sonoff DEV with 5.12.0g version flashed I've had connected DS 18x20 to GPIO14 only, it works well.
Now I connected AM2301 to GPIO4 and Switch1 (actually it is a rain sensor) to GPIO5.
DS18x20 and Switch1 work properly on the other side AM2301 is not displayed, there is no readout at all looks like it is not connected or recognised.
Anyhow, is it a bug or did I sthg wrong? How to make it working properly? Please...

GPIO command says:
MAC address is redacted.
`13:18:36 CMD: gpio
13:18:36 MQT: stat/DEV/RESULT = {"GPIO1":"0 (None)","GPIO3":"0 (None)","GPIO4":"2 (AM2301)","GPIO5":"9 (Switch1)","GPIO12":"0 (None)","GPIO14":"4 (DS18x20)"}
13:18:43 MQT: tele/DEV/STATE = {"Time":"2018-04-03T13:18:43","Uptime":"0T00:00:23","Vcc":3.464,"Wifi":{"AP":1,"SSId":"HugoNET","RSSI":70,"APMac":"08:96:xx:xx:xx:xx"}}
13:18:43 MQT: tele/DEV/SENSOR = {"Time":"2018-04-03T13:18:43","Switch1":"ON","DS18B20":{"Temperature":19.8},"TempUnit":"C"}`

after 15 mins:

`13:32:32 MQT: tele/DEV/STATE = {"Time":"2018-04-03T13:32:32","Uptime":"0T00:14:12","Vcc":3.463,"Wifi":{"AP":1,"SSId":"HugoNET","RSSI":58,"APMac":"08:96:xx:xx:xx:xx"}}
13:32:32 MQT: tele/DEV/SENSOR = {"Time":"2018-04-03T13:32:32","Switch1":"ON","DS18B20":{"Temperature":19.7},"TempUnit":"C"}`

Status 0 command says:
MAC address is redacted.
`13:24:12 CMD: status 0
13:24:12 MQT: stat/DEV/STATUS = {"Status":{"Module":19,"FriendlyName":"DEV","Topic":"DEV","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
13:24:12 MQT: stat/DEV/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:05:52","StartupUTC":"2018-04-03T12:18:20","Sleep":0,"BootCount":29,"SaveCount":65,"SaveAddress":"FB000"}}
13:24:12 MQT: stat/DEV/STATUS2 = {"StatusFWR":{"Version":"5.12.0g","BuildDateTime":"2018-03-20T18:43:45","Boot":4,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
13:24:12 MQT: stat/DEV/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId1":"HugoNET","SSId2":"","TelePeriod":10,"SetOption":"00000009"}}
13:24:12 MQT: stat/DEV/STATUS4 = {"StatusMEM":{"ProgramSize":498,"Free":504,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
13:24:12 MQT: stat/DEV/STATUS5 = {"StatusNET":{"Hostname":"DEV-0219","IPAddress":"192.168.1.59","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:yy:yy:yy:yy","Webserver":2,"WifiConfig":3}}
13:24:12 MQT: stat/DEV/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.107","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9240DB","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
13:24:12 MQT: stat/DEV/STATUS7 = {"StatusTIM":{"UTC":"Tue Apr 03 12:24:12 2018","Local":"Tue Apr 03 13:24:12 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
13:24:12 MQT: stat/DEV/STATUS10 = {"StatusSNS":{"Time":"2018-04-03T13:24:12","Switch1":"ON","DS18B20":{"Temperature":19.8},"TempUnit":"C"}}
13:24:12 MQT: stat/DEV/STATUS11 = {"StatusSTS":{"Time":"2018-04-03T13:24:12","Uptime":"0T00:05:52","Vcc":3.463,"Wifi":{"AP":1,"SSId":"HugoNET","RSSI":66,"APMac":"08:96:xx:xx:xx:xx"}}}
13:24:17 MQT: tele/DEV/STATE = {"Time":"2018-04-03T13:24:17","Uptime":"0T00:05:57","Vcc":3.462,"Wifi":{"AP":1,"SSId":"HugoNET","RSSI":70,"APMac":"08:96:xx:xx:xx:xx"}}`

I did not try rest of my sonoffs yet (would like to test during this week), so it might be solely Sonoff DEV related issue.

Hugo477, 2018-04-03:
You are right GPIO13 is connected to the blue LED nonetheless it  has a pin in one connector as well.

With sensors I am completely confused now.
I tried to connect AM2301 with a pull-up rezistor with no expected result. It still behaves like there is nothing connected. 
Strange is that there is nothing displayed even zeroes (AM2301 Temperature  and AM2301 Humidity on the main screen under or above DS 18x20).

arendst, 2018-04-03:
Enable more debugging info with ``weblog 3`` or ``seriallog 3``. You should at least see this:
```
17:34:35 DHT: Received 01, B9, 00, FB, B5 =? B5
17:34:35 DHT: Received 01, B1, 00, FA, AC =? AC
17:34:35 MQT: tele/wemos4/SENSOR = {"Time":"2018-04-03T17:34:35","AM2301-12":{"Temperature":25.1,"Humidity":44.1},"SI7021-14":{"Temperature":25.0,"Humidity":43.3},"TempUnit":"C"}
```

Hugo477, 2018-04-03:
DS 18x20 to GPIO14, AM2301 to GPIO4 and rain sensor to GPIO5.

```
17:25:27 CMD: weblog 3
17:25:27 MQT: stat/DEV/RESULT = {"WebLog":3}
17:25:27 CFG: Saved to flash at F7, Count 133, Bytes 1648
17:25:32 MQT: tele/DEV/STATE = {"Time":"2018-04-03T17:25:32","Uptime":"0T00:02:55","Vcc":3.462,"Wifi":{"AP":1,"SSId":"HugoNET","RSSI":100,"APMac":"F4:F2:xx:xx:xx:xx"}}
17:25:32 DHT: Timeout waiting for start signal high pulse
17:25:32 MQT: tele/DEV/SENSOR = {"Time":"2018-04-03T17:25:32","Switch1":"ON","DS18B20":{"Temperature":18.5},"TempUnit":"C"}
```
`17:29:09 CMD: seriallog 3
17:29:09 RSL: Group 0, Index 1, Command SERIALLOG, Data 3
17:29:09 MQT: stat/DEV/RESULT = {"SerialLog":"3 (Active 3)"}
17:29:10 CFG: Saved to flash at F6, Count 134, Bytes 1648
17:29:14 MQT: tele/DEV/STATE = {"Time":"2018-04-03T17:29:14","Uptime":"0T00:06:37","Vcc":3.462,"Wifi":{"AP":1,"SSId":"HugoNET","RSSI":100,"APMac":"F4:F2:xx:xx:xx:xx"}}
17:29:14 DHT: Timeout waiting for start signal high pulse
17:29:14 MQT: tele/DEV/SENSOR = {"Time":"2018-04-03T17:29:14","Switch1":"ON","DS18B20":{"Temperature":0
18.5},"TempUnit":"C"}`

If I remove all sensors and connected AM2301 to GPIO14:

`17:34:37 DHT: Timeout waiting for start signal high pulse
17:34:47 MQT: tele/DEV/STATE = {"Time":"2018-04-03T17:34:47","Uptime":"0T00:01:03","Vcc":3.462,"Wifi":{"AP":1,"SSId":"HugoNET","RSSI":100,"APMac":"F4:F2:6D:9B:94:A2"}}`

Hugo477, 2018-04-03:
I found at home spare "virgin" TH16.
Tested with ewelink and AM2301 - worked well.
Flashed with a fresh compilation of 5.12.0i, where I changed `user_config.h` - Wifi ssid, password, static IP and router IP, in 'platformio.ini` uncommented `env_default = sonoff' than build, save, flash, no errors at all.
And .... exactly the same as DEV.
Might it be that I have bugs in my compilation?

arendst, 2018-04-04:
Do you provide  enough power? When connecting the am2301 you'll need to power cycle also.

Hugo477, 2018-04-04:
It was my first concern. Sonoff DEV is powered by USB adapter providing up to 2 Amps.

Hugo477, 2018-04-04:
Is there any chance to have it working with all sensors if I  erase flash by `esptool.py --port COM5 erase_region 0x0F4000 0x008000` and reflash 5.12.0i again?

Hugo477, 2018-04-05:
I've tried your solution and it works :)) but why??
Great thank you anyhow..

Dickesplumpel, 2018-04-05:
Trying to answer questions like why can bring you in great trouble, thats what life can show you when you look back when you are older! ;-))
It works thats all that counts for me. I'm playing around with these things often I fail and sometimes I make a discovery - thats all.

closed 2018-04-05 by Hugo477