arendst/Tasmota #2342: mqtt stops working when adding a second sensor

aykot, 2018-04-03:
Make sure these boxes are checked before submitting your issue - Thank you!

- [x] What hardware you are using
- [x] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

ESP8266 with SI7021 works perfectly and sends to mqtt humidity and temperature values.
when I connect BH1750 Ambient Light Intensity sensor to tx/rx, tasmota web interface logs the three values (lux, temp and humidity) but no reports are sent to mqtt. 
If I physically disconnect BH1750 and reset, the mqtt starts working again.

BH1750 connections:
VCC -> 3v3
GND -> GND
SCL -> RX
SDA -> TX
ADDR ->  not connected
[image]
status 0 (unplugged BH1750):

> 15:38:14 RSL: stat/sonoff-sensores/STATUS = {"Status":{"Module":18,"FriendlyName":[],"Topic":"sonoff-sensores","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
15:38:14 RSL: stat/sonoff-sensores/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"17624T06:38:13","StartupUTC":"1970-01-01T08:00:01","Sleep":0,"BootCount":59,"SaveCount":570,"SaveAddress":"3F6000"}}
15:38:14 RSL: stat/sonoff-sensores/STATUS2 = {"StatusFWR":{"Version":"5.12.0i","BuildDateTime":"2018-03-23T20:02:17","Boot":5,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
15:38:14 RSL: stat/sonoff-sensores/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["AKT_ACCESS_POINT_5G","WLAN_AKTX"],"TelePeriod":100,"SetOption":["00000009","55818000"]}}
15:38:14 RSL: stat/sonoff-sensores/STATUS4 = {"StatusMEM":{"ProgramSize":529,"Free":2540,"Heap":23,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":2}}
15:38:14 RSL: stat/sonoff-sensores/STATUS5 = {"StatusNET":{"Hostname":"sonoff-sensores-3917","IPAddress":"192.168.1.50","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"80.58.61.250","Mac":"5C:CF:7F:76:EF:4D","Webserver":2,"WifiConfig":3}}
15:38:14 RSL: stat/sonoff-sensores/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.100","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_76EF4D","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:38:14 RSL: stat/sonoff-sensores/STATUS7 = {"StatusTIM":{"UTC":"Tue Apr 03 14:38:14 2018","Local":"Tue Apr 03 15:38:14 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
15:38:14 RSL: stat/sonoff-sensores/STATUS10 = {"StatusSNS":{"Time":"2018-04-03T15:38:14","SI7021":{"Temperature":23.0,"Humidity":45.1},"TempUnit":"C"}}
15:38:14 RSL: stat/sonoff-sensores/STATUS11 = {"StatusSTS":{"Time":"2018-04-03T15:38:14","Uptime":"17624T06:38:13","Vcc":2.993,"Wifi":{"AP":2,"SSId":"WLAN_AKTX","RSSI":80,"APMac":"B2:46:FC:6A:53:00"}}}

status 0 (plugged BH1750):

> 15:40:33 RSL: stat/sonoff-sensores/STATUS = {"Status":{"Module":18,"FriendlyName":[],"Topic":"sonoff-sensores","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
15:40:33 RSL: stat/sonoff-sensores/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:22","StartupUTC":"2018-04-03T14:40:11","Sleep":0,"BootCount":60,"SaveCount":571,"SaveAddress":"3F5000"}}
15:40:33 RSL: stat/sonoff-sensores/STATUS2 = {"StatusFWR":{"Version":"5.12.0i","BuildDateTime":"2018-03-23T20:02:17","Boot":5,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
15:40:33 RSL: stat/sonoff-sensores/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["AKT_ACCESS_POINT_5G","WLAN_AKTX"],"TelePeriod":100,"SetOption":["00000009","55818000"]}}
15:40:33 RSL: stat/sonoff-sensores/STATUS4 = {"StatusMEM":{"ProgramSize":529,"Free":2540,"Heap":23,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":2}}
15:40:33 RSL: stat/sonoff-sensores/STATUS5 = {"StatusNET":{"Hostname":"sonoff-sensores-3917","IPAddress":"192.168.1.50","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"80.58.61.250","Mac":"5C:CF:7F:76:EF:4D","Webserver":2,"WifiConfig":3}}
15:40:33 RSL: stat/sonoff-sensores/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.100","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_76EF4D","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:40:33 RSL: stat/sonoff-sensores/STATUS7 = {"StatusTIM":{"UTC":"Tue Apr 03 14:40:33 2018","Local":"Tue Apr 03 15:40:33 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
15:40:33 RSL: stat/sonoff-sensores/STATUS10 = {"StatusSNS":{"Time":"2018-04-03T15:40:33","SI7021":{"Temperature":23.0,"Humidity":45.2},"BH1750":{"Illuminance":110},"TempUnit":"C"}}
15:40:33 RSL: stat/sonoff-sensores/STATUS11 = {"StatusSTS":{"Time":"2018-04-03T15:40:33","Uptime":"0T00:00:22","Vcc":2.993,"Wifi":{"AP":2,"SSId":"WLAN_AKTX","RSSI":76,"APMac":"B2:46:FC:6A:53:00"}}}

Frogmore42, 2018-04-03:
This line seems to indicate things are working somewhat
15:40:33 RSL: stat/sonoff-sensores/STATUS10 = {"StatusSNS":{"Time":"2018-04-03T15:40:33","SI7021":{"Temperature":23.0,"Humidity":45.2},"BH1750":{"Illuminance":110},"TempUnit":"C"}}

I would not use the RX TX lines for sensors unless it was all that was left.  You have to turn off serial logging to get them to work properly.

aykot, 2018-04-03:
@Frogmore42 thanks for the input. Changed the pins from TX/RX to D4/D2 and the problem persists.
[image]

arendst, 2018-04-03:
The fact it states 16:31:05 RSL means the MQTT connection is gone. it should say 16:31:05 MQT.

You're also using a non-tasmota supported memory model. Only 1M(no spiffs) is supported.

aykot, 2018-04-03:
@arendst thanks for the response, but according to the wiki the NodeMCU esp8266 is indeed supported. I don't really understand what do you mean by Only 1M(no spiffs) when it works without problem with the other two sensors.

Frogmore42, 2018-04-04:
this is the part (memory model) that is not supported:
"ProgramFlashSize":4096

There are different ways to set that depending on how you are building the image.

NodeMCU is supported and generally works.  But, do you have this on a breadboard?

I discovered a couple of days ago that some of my NodeMCU devices behave very strangely when I have them on a breadboard.  I had assumed that they were just defective, but I was curious so I removed it from the breadboard and used jumper wires to connect the 4 pins I actually needed on the breadboard.  With this small change, it started working just fine.  Previously, it would get hardware watchdogs and reboot and not be able to connect and get the watchdog again.  If I reloaded the FW it would work fine for a little bit.  My guess is that all the metal in the breadboard was causing problems with the 2.4GHz Wi-Fi signal.  Now, I have several other Wemos D1 Minis and one or two other NodeMCUs that work fine on a breadboard.  But, having the Wi-Fi antenna right next to the metal jumpers in a breadboard is a really bad idea according to many RF experts.

isoriaa, 2018-04-11:
I have the same problem, did you fix it.

aykot, 2018-04-12:
@isoriaa sorry, but my module was not supported as @arendst pointed, so I gave up and used a second module

arendst, 2018-04-12:
It's not your hardware that is not supported. It's your compiled software memory model that is not supported. 

Learn how to set your IDE (or use provided platformIO ini file) for 1M (No SPIFFS) during initial serial upload.

aykot, 2018-04-12:
@arendst sorry to disturb you again, but I´ve reflashed with 1 MB no SPIFSS, rewired everything again to only one board and the problem is exactly the same. I attach you the status 0 response.

> 21:56:55 RSL: stat/sonoff-sensores/STATUS = {"Status":{"Module":18,"FriendlyName":[],"Topic":"sonoff-sensores","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
21:56:55 RSL: stat/sonoff-sensores/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:05:02","StartupUTC":"2018-04-12T20:51:53","Sleep":0,"BootCount":3,"SaveCount":10,"SaveAddress":"F6000"}}
21:56:55 RSL: stat/sonoff-sensores/STATUS2 = {"StatusFWR":{"Version":"5.12.0i","BuildDateTime":"2018-04-12T22:42:26","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
21:56:55 RSL: stat/sonoff-sensores/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["AKT_ACCESS_POINT_5G","WLAN_AKTX"],"TelePeriod":100,"SetOption":["00000009","55818000"]}}
21:56:55 RSL: stat/sonoff-sensores/STATUS4 = {"StatusMEM":{"ProgramSize":529,"Free":472,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
21:56:55 RSL: stat/sonoff-sensores/STATUS5 = {"StatusNET":{"Hostname":"sonoff-sensores-3917","IPAddress":"192.168.1.52","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"80.58.61.250","Mac":"5C:CF:7F:76:EF:4D","Webserver":2,"WifiConfig":3}}
21:56:55 RSL: stat/sonoff-sensores/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.100","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_76EF4D","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:56:55 RSL: stat/sonoff-sensores/STATUS7 = {"StatusTIM":{"UTC":"Thu Apr 12 20:56:55 2018","Local":"Thu Apr 12 21:56:55 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
21:56:55 RSL: stat/sonoff-sensores/STATUS10 = {"StatusSNS":{"Time":"2018-04-12T21:56:55","SI7021":{"Temperature":20.5,"Humidity":52.5},"BH1750":{"Illuminance":5},"TempUnit":"C"}}
21:56:55 RSL: stat/sonoff-sensores/STATUS11 = {"StatusSTS":{"Time":"2018-04-12T21:56:55","Uptime":"0T00:05:02","Vcc":2.997,"Wifi":{"AP":2,"SSId":"WLAN_AKTX","RSSI":76,"APMac":"B2:46:FC:6A:53:00"}}}

arendst, 2018-04-12:
Check your power supply. Do not use esp/arduino 2.4.1.

RSL means no mqtt connection. Get a solid mqtt connection first. Then log what messages you receive with logging level 3 when mqtt drops connection.

This normally works fine so prepare for a local solution.

aykot, 2018-04-12:
@arendst Switched to a new power supply (5V 1A), switched to logging level 3 and hot plugged the last sensor when an stable mqtt connection was made. (marked in code with brackets)

> 00:00:00 CFG: Loaded from flash at F9, Count 15
00:00:00 APP: Boot Count 8
00:00:00 Project sonoff Sonoff-sensores (Topic sonoff-sensores, Fallback DVES_76EF4D, GroupTopic sonoffs) Version 5.12.0i-2_4_1
00:00:00 CFG: Saved to flash at F8, Count 16, Bytes 1712
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP2 WLAN_AKTX in mode 11N as sonoff-sensores-3917...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:04 WIF: Attempting connection...
00:00:04 APP: (UTC) Thu Apr 12 21:54:23 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
22:54:24 WIF: Connected
22:54:24 DNS: Initialized
22:54:24 HTP: Web server active on sonoff-sensores-3917.local with IP address 192.168.1.52
22:54:24 HTP: Console
22:54:25 MQT: Attempting connection...
22:54:25 MQT: Connected
22:54:25 MQT: tele/sonoff-sensores/LWT = Online (retained)
22:54:25 MQT: cmnd/sonoff-sensores/POWER = 
22:54:25 MQT: Subscribe to cmnd/sonoff-sensores/#
22:54:25 MQT: Subscribe to cmnd/sonoffs/#
22:54:25 MQT: Subscribe to cmnd/DVES_76EF4D/#
22:54:25 MQT: tele/sonoff-sensores/INFO1 = {"Module":"Generic","Version":"5.12.0i","FallbackTopic":"DVES_76EF4D","GroupTopic":"sonoffs"}
22:54:25 MQT: tele/sonoff-sensores/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-sensores-3917","IPAddress":"192.168.1.52"}
22:54:25 MQT: tele/sonoff-sensores/INFO3 = {"RestartReason":"External System"}
22:54:32 SHT: Sensor did not ACK command
22:54:33 RSL: tele/sonoff-sensores/STATE = {"Time":"2018-04-12T22:54:33","Uptime":"0T00:00:14","Vcc":2.997,"Wifi":{"AP":2,"SSId":"WLAN_AKTX","RSSI":82,"APMac":"B2:46:FC:6A:53:00"}}
22:54:33 DHT: Received 01, D1, 00, E6, B8 =? B8
22:54:33 MQT: tele/sonoff-sensores/SENSOR = {"Time":"2018-04-12T22:54:33","SI7021":{"Temperature":23.0,"Humidity":46.5},"TempUnit":"C"}
22:54:42 SHT: Sensor did not ACK command
22:54:43 RSL: tele/sonoff-sensores/STATE = {"Time":"2018-04-12T22:54:43","Uptime":"0T00:00:24","Vcc":2.996,"Wifi":{"AP":2,"SSId":"WLAN_AKTX","RSSI":74,"APMac":"B2:46:FC:6A:53:00"}}
22:54:43 DHT: Received 01, D0, 00, E6, B7 =? B7
22:54:43 MQT: tele/sonoff-sensores/SENSOR = {"Time":"2018-04-12T22:54:43","SI7021":{"Temperature":23.0,"Humidity":46.4},"TempUnit":"C"}
22:54:52 SHT: Sensor did not ACK command
22:54:53 RSL: tele/sonoff-sensores/STATE = {"Time":"2018-04-12T22:54:53","Uptime":"0T00:00:34","Vcc":2.997,"Wifi":{"AP":2,"SSId":"WLAN_AKTX","RSSI":80,"APMac":"B2:46:FC:6A:53:00"}}
22:54:53 DHT: Received 01, D0, 00, E6, B7 =? B7
22:54:53 MQT: tele/sonoff-sensores/SENSOR = {"Time":"2018-04-12T22:54:53","SI7021":{"Temperature":23.0,"Humidity":46.4},"TempUnit":"C"}
22:55:02 SHT: Sensor did not ACK command
22:55:03 RSL: tele/sonoff-sensores/STATE = {"Time":"2018-04-12T22:55:03","Uptime":"0T00:00:44","Vcc":2.997,"Wifi":{"AP":2,"SSId":"WLAN_AKTX","RSSI":78,"APMac":"B2:46:FC:6A:53:00"}}
22:55:03 DHT: Received 01, D0, 00, E6, B7 =? B7
22:55:03 MQT: tele/sonoff-sensores/SENSOR = {"Time":"2018-04-12T22:55:03","SI7021":{"Temperature":23.0,"Humidity":46.4},"TempUnit":"C"}
22:55:12 SHT: Sensor did not ACK command
22:55:13 RSL: tele/sonoff-sensores/STATE = {"Time":"2018-04-12T22:55:13","Uptime":"0T00:00:54","Vcc":2.997,"Wifi":{"AP":2,"SSId":"WLAN_AKTX","RSSI":68,"APMac":"B2:46:FC:6A:53:00"}}
22:55:13 DHT: Received 01, D0, 00, E6, B7 =? B7
22:55:13 MQT: tele/sonoff-sensores/SENSOR = {"Time":"2018-04-12T22:55:13","SI7021":{"Temperature":23.0,"Humidity":46.4},"TempUnit":"C"}
22:55:22 SHT: Sensor did not ACK command
22:55:23 RSL: tele/sonoff-sensores/STATE = {"Time":"2018-04-12T22:55:23","Uptime":"0T00:01:04","Vcc":2.274,"Wifi":{"AP":2,"SSId":"WLAN_AKTX","RSSI":78,"APMac":"B2:46:FC:6A:53:00"}}
22:55:23 DHT: Timeout waiting for start signal high pulse
22:55:23 MQT: tele/sonoff-sensores/SENSOR = {"Time":"2018-04-12T22:55:23","SI7021":{"Temperature":23.0,"Humidity":46.4},"TempUnit":"C"}
22:55:32 SHT: Sensor did not ACK command
22:55:33 RSL: tele/sonoff-sensores/STATE = {"Time":"2018-04-12T22:55:33","Uptime":"0T00:01:14","Vcc":2.466,"Wifi":{"AP":2,"SSId":"WLAN_AKTX","RSSI":80,"APMac":"B2:46:FC:6A:53:00"}}
22:55:33 DHT: Timeout waiting for start signal high pulse
22:55:33 MQT: tele/sonoff-sensores/SENSOR = {"Time":"2018-04-12T22:55:33","SI7021":{"Temperature":23.0,"Humidity":46.4},"TempUnit":"C"}
22:55:42 SHT: Sensor did not ACK command
22:55:43 RSL: tele/sonoff-sensores/STATE = {"Time":"2018-04-12T22:55:43","Uptime":"0T00:01:24","Vcc":2.658,"Wifi":{"AP":2,"SSId":"WLAN_AKTX","RSSI":82,"APMac":"B2:46:FC:6A:53:00"}}
22:55:43 DHT: Received 02, 87, 00, E9, 72 =? 72
22:55:43 MQT: tele/sonoff-sensores/SENSOR = {"Time":"2018-04-12T22:55:43","SI7021":{"Temperature":23.3,"Humidity":64.7},"TempUnit":"C"}
22:55:52 SHT: Sensor did not ACK command
22:55:53 RSL: tele/sonoff-sensores/STATE = {"Time":"2018-04-12T22:55:53","Uptime":"0T00:01:34","Vcc":2.670,"Wifi":{"AP":2,"SSId":"WLAN_AKTX","RSSI":84,"APMac":"B2:46:FC:6A:53:00"}}
22:55:53 DHT: Received 02, 16, 00, E7, FF =? FF
22:55:53 MQT: tele/sonoff-sensores/SENSOR = {"Time":"2018-04-12T22:55:53","SI7021":{"Temperature":23.1,"Humidity":53.4},"TempUnit":"C"}
22:56:02 SHT: Sensor did not ACK command
22:56:03 RSL: tele/sonoff-sensores/STATE = {"Time":"2018-04-12T22:56:03","Uptime":"0T00:01:44","Vcc":2.997,"Wifi":{"AP":2,"SSId":"WLAN_AKTX","RSSI":84,"APMac":"B2:46:FC:6A:53:00"}}
[... cut ...]
22:56:33 RSL: tele/sonoff-sensores/SENSOR = {"Time":"2018-04-12T22:56:33","SI7021":{"Temperature":23.0,"Humidity":46.8},"BH1750":{"Illuminance":29},"TempUnit":"C"}
22:56:42 SHT: Sensor did not ACK command
22:56:43 RSL: tele/sonoff-sensores/STATE = {"Time":"2018-04-12T22:56:43","Uptime":"0T00:02:24","Vcc":2.994,"Wifi":{"AP":2,"SSId":"WLAN_AKTX","RSSI":86,"APMac":"B2:46:FC:6A:53:00"}}
22:56:43 DHT: Received 01, D6, 00, E6, BD =? BD
22:56:43 RSL: tele/sonoff-sensores/SENSOR = {"Time":"2018-04-12T22:56:43","SI7021":{"Temperature":23.0,"Humidity":47.0},"BH1750":{"Illuminance":21},"TempUnit":"C"}

thomas-lentz, 2019-09-09:
Well, I experience the same problem and I narrowed it down to a message size problem.
I shortened the message so often, that the "RSL:" changed back to "MQT:". I guess the error came with a major change of the support.ino or in the MQTT driver. 
Maybe somebody could have a look into it. I'm using MQTT via TCP (not TLS) and the maximum working message length is 128 character as you can see: 

```
00:00:10 RSL: 1/2/INFO1 = {"Module":"RCSG_JETTY","Version":"6.6.0.10 stb-1.11(sonoff)","FallbackTopic":"cmnd/rcsg-jetty_fb/","GroupTopic":"12"}
00:00:07 RSL: 1/2/INFO1 = {"Module":"My-Module1","Version":"6.6.0.10 stb-1.11(sonoff)","FallbackTopic":"cmnd/x12345678_fb/","GroupTopic":"z"}
00:00:06 RSL: 1/2/INFO1 = {"Module":"My-Module1","Version":"6.6.0.10 stb-1.11(sonoff)","FallbackTopic":"cmnd/x1234567_fb/","GroupTopic":"z"}
00:00:07 RSL: 1/2/INFO1 = {"Module":"My-Module1","Version":"6.6.0.10 stb-1.11(sonoff)","FallbackTopic":"cmnd/x123456_fb/","GroupTopic":"z"}
00:00:07 MQT: 1/2/INFO1 = {"Module":"My-Module1","Version":"6.6.0.10 stb-1.11(sonoff)","FallbackTopic":"cmnd/x12345_fb/","GroupTopic":"z"}
                                                                                                             1         1         1       1 1         1
         1         2         3         4         5         6         7         8         9         1         0         1         2       2 3         4
123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890
```

thomas-lentz, 2019-09-09:
Okay, this took hours of sleep, but now I solved it.
The problem is somewhere else - it is an old "PubSubClient-EspEasy".
It has to be the "**PubSubClient-EspEasy-2.7.12**" or newer.
In my case there were the correct and the older library in the path - guess which one was taken (yes, I know there have been warnings from the compiler ...).

A 128 characters are by far not enough!
These are the problematic lines:
```
// MQTT_MAX_PACKET_SIZE : Maximum packet size
#ifndef MQTT_MAX_PACKET_SIZE
#define MQTT_MAX_PACKET_SIZE 128
#endif
```

So if you have the "RSL:" instead of "MQT:" check the version of your "PubSubClient-EspEasy".