arendst/Tasmota #2343: Tasmota Crashing and rebooting

gadgetmanecu, 2018-04-03:
Hi I have a number of Sonoff Basic modules version 1.1

I have 5.12.0 running on the two modules and they are both exhibiting the same behavior of the web page loading but then becoming unresponsive almost instantly and after about 1-2 minutes the system reboots again.

I have attached the status 0 dump that was very hard to get because the sonoff webpage was very unresponsive.  I have to have 15-20 attempts to set any command in the console as the window of it accepting input before the web server becomes unresponsive and the sonoff web page disappears is very small. 

These are two new sonoffs received from itlead. I have an external push button switch attached as switch 1 for push on and push off.

These were loaded using Arduino IDE 1.8.5, which I have also used to load a Sonoff SV a month or so ago that was exhibiting similar symptoms until I locked the unit into WifiConfig mode 5 and since then it has been extremely stable.

Going out of my mind a little with this issue, Can someone work out what the issue is here,

[sonoff issue status 0.txt](https://github.com/arendst/Sonoff-Tasmota/files/1872069/sonoff.issue.status.0.txt)

phreakmonkey, 2018-04-04:
FYI - I had two Sonoff Basics that were running much older Sonoff-Tasmota firmware that went into a crash-reboot loop after updating them to 5.12.0i.   Holding down button0 for 40+ seconds to force a config erase (and restore the settings from user_config.h) fixed them.  Strangly, a third Sonoff-Basic running upgraded to 5.12.0i didn't exhibit this behavior and worked fine.

Try the manual config-reset, I'm interested to know if it works for you also.

arendst, 2018-04-04:
Try development branch version 5.12.0i and report back. If possible with ``status 0`` report you should be able to get just after the restart.