arendst/Tasmota #2344: Can't flash Tasmota on Sonoff Basic

bachya, 2018-04-03:
Hi all – apologies for yet another "I can't flash" issue thread, but I've tried everything I can think of (and everything I've read in various other issues) and I just can't get it to work. I'm 99.9% certain this isn't a Tasmota issue, but given the expertise of everyone here, I'm hoping I can post everything I have and get your advice on how to proceed. I'm used to smashing through issues, so this is irking me. 😎 

Thanks in advance for stomaching all this; again, I offer it for completeness and in the hope that someone smarter than I can offer advice.

# Hardware

## Sonoff

I'm using a [Sonoff Basic](https://www.aliexpress.com/item/ITEAD-Sonoff-Wireless-Wifi-Switch-For-Smart-Home-Automation-Relay-Module-10A-90-250V-220V-Support/32829067852.html). For reasons I'll share shortly, I assume this Sonoff Basic is using a firmware higher than 1.6.

[image]

## FTDI Converter

I'm using a [FT232RL FT232 FTDI USB to TTL 3.3V 5.5V Serial Adapter Module Download Cable for Mini Port](https://www.aliexpress.com/item/FT232RL-FT232-USB-TO-TTL-5V-3-3V-Download-Cable-To-Serial-Adapter-Module-For-USB/32634246159.html) (a mouthful!). I've confirmed that connecting a Mini USB port provides 3.3v across VCC and GND.

[image]

## Computer

I'm using a MacBook Pro (13-inch, 2017, Four Thunderbolt 3 Ports) running macOS High Sierra 10.13.4. I've installed [the VCP drivers for FTDI devices](http://www.ftdichip.com/Drivers/VCP.htm) (per this [comment](https://github.com/arendst/Sonoff-Tasmota/issues/957#issuecomment-338779258)) and can confirm (via `Apple > System Report > USB`) that the MacBook recognizes the FTDI adapter when connected.

## USB Hub

Because this generation of MacBook is USB-C-only, I connect the Mini USB cable via a USB-C hub. Can't imagine this matters, but including it for completeness.

[image]

# Hardware Preparation Attempts

## 1. Via SonOTA

I first attempted to flash Tasmota via [SonOTA](https://github.com/mirko/SonOTA) (given that I've had success with it before). I got about 90% of the way through the process, but was halted at the last step (i.e., the "FinalStage" network). A comment on that project's issues page led me to believe that if I was having troubles, it might mean that I was using a Sonoff firmware higher than 1.6 and would need a manual flash.

I share this with you not to debug SonOTA, but to let you know that "something" was flashed onto the Sonoff during this process. I can tell because the process of holding down the Sonoff button and seeing the green LED flash consistently no longer occurs. Again, providing for completeness.

## 2. Via Serial Connection

Reading around, I deduced that I didn't specifically _need_ to solder header pins onto the Sonoff; rather, I could use [male-to-female Dupont cables](https://www.amazon.com/RioRand-Dupont-Male-Male-Female-Female-Female-Male/dp/B00J5NSOVA):

* Connect the female ends to the correct pins on the FTDI converter
* Manually hold the male pins in place through the correct holes on the Sonoff

I attempted flashing via three different tools:

* The Arduino IDE
* PlatformIO
* `esptool.py`

For each tool, I attempted a few different procedures, each with the same result (detailed in the next section). For completeness, these procedures are listed here.

### Procedure 1

1. Connect a cable between FTDI GND and Sonoff GND.
2. Connect a cable between FTDI RX and Sonoff TX.
3. Connect a cable between FTDI TX and Sonoff RX.
4. Connect a cable to FTDI VCC and leave the male lead alone.
4. Connect the Micro USB cable between the FTDI converter and the USB-C hub.
5. Hold down the button on the Sonoff (bringing GPIO0 to GND).
6. Insert the VCC male lead to 3v3.
7. Let go of the button after 1-2 seconds.
8. Attempt to flash.

### Procedure 2

1. Connect a cable between FTDI GND and Sonoff GND.
2. Connect a cable between FTDI RX and Sonoff TX.
3. Connect a cable between FTDI TX and Sonoff RX.
4. Connect a cable between FTDI VCC and Sonoff 3v3.
5. Hold down the button on the Sonoff (bringing GPIO0 to GND).
6. Connect the Micro USB cable between the FTDI converter and the USB-C hub.
7. Let go of the button after 1-2 seconds.
8. Attempt to flash.

### Procedure 3

1. Connect a cable between FTDI GND and Sonoff GND.
2. Connect a cable between FTDI RX and Sonoff TX.
3. Connect a cable between FTDI TX and Sonoff RX.
4. Connect a cable to FTDI VCC and leave the male lead alone.
5. Connect the Micro USB cable between the FTDI converter and the USB-C hub.
6. Hold down the button on the Sonoff (bringing GPIO0 to GND).
7. Insert the VCC male lead to 3v3.
8. Attempt to flash with the button held throughout.

### Procedure 4

1. Connect a cable between FTDI GND and Sonoff GND.
2. Connect a cable between FTDI RX and Sonoff TX.
3. Connect a cable between FTDI TX and Sonoff RX.
4. Connect a cable between FTDI VCC and Sonoff 3v3.
5. Hold down the button on the Sonoff (bringing GPIO0 to GND).
6. Connect the Micro USB cable between the FTDI converter and the USB-C hub.
7. Attempt to flash with the button held throughout.

– – –

In all cases, the Sonoff LED never came on (flashing or otherwise).

# Flashing Attempts

## 1. Arduino IDE

Followed the Tasmota instructions for the Arduino IDE. After 9-10 seconds, the IDE returns the following error:

```
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
```

## 2. PlatformIO

Followed the PlatformIO instructions for the Arduino IDE. After 9-10 seconds, the IDE returns the following error:

```
|   |-- <ESP8266HTTPClient> v1.1
|   |   |-- <ESP8266WiFi> v1.0
|-- <ESP8266WiFi> v1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> v1.0
|-- <ESP8266WebServer> v1.0
|   |-- <ESP8266WiFi> v1.0

Compiling /Users/abach/Git/vsc/Sonoff-Tasmota-5.12.0/.pioenvs/sonoff/src/sonoff.ino.o
Linking /Users/abach/Git/vsc/Sonoff-Tasmota-5.12.0/.pioenvs/sonoff/firmware.elf
Checking program size
text       data     bss     dec     hex filename
445299    10668   35136  491103   77e5f /Users/abach/Git/vsc/Sonoff-Tasmota-5.12.0/.pioenvs/sonoff/firmware.elf
Building /Users/abach/Git/vsc/Sonoff-Tasmota-5.12.0/.pioenvs/sonoff/firmware.bin
Looking for upload port...
Auto-detected: COM5
Uploading /Users/abach/Git/vsc/Sonoff-Tasmota-5.12.0/.pioenvs/sonoff/firmware.bin
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error -1
 [ERROR] Took 17.86 seconds
```

## 3. `esptool.py` Version 1

```
$ esptool.py --port /dev/tty.usbserial-00000000 -b 115200 write_flash 0x0 ~/Git/vsc/Sonoff-Tasmota-5.12.0/.pioenvs/sonoff/firmware.bin
esptool.py v2.3.1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
```

## 4. `esptool.py` Version 2

Using an idea I'd read elsewhere, I started `esptool.py`, let it attempt to connect, _then_ connected voltage to the Sonoff. As soon as voltage was connected, the stack trace below occurred.

```
$ esptool.py --port /dev/tty.usbserial-00000000 -b 115200 write_flash 0x0 ~/Git/vsc/Sonoff-Tasmota-5.12.0/.pioenvs/sonoff/firmware.bin
esptool.py v2.3.1
Connecting........_____....._____....._____.....
Traceback (most recent call last):
  File "/usr/local/bin/esptool.py", line 2637, in <module>
    _main()
  File "/usr/local/bin/esptool.py", line 2630, in _main
    main()
  File "/usr/local/bin/esptool.py", line 2349, in main
    esp = ESPLoader.detect_chip(args.port, initial_baud, args.before, args.trace)
  File "/usr/local/bin/esptool.py", line 223, in detect_chip
    detect_port.connect(connect_mode)
  File "/usr/local/bin/esptool.py", line 410, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=True)
  File "/usr/local/bin/esptool.py", line 373, in _connect_attempt
    self._port.setDTR(True)   # IO0=LOW
  File "/Library/Python/2.7/site-packages/serial/serialutil.py", line 599, in setDTR
    self.dtr = value
  File "/Library/Python/2.7/site-packages/serial/serialutil.py", line 469, in dtr
    self._update_dtr_state()
  File "/Library/Python/2.7/site-packages/serial/serialposix.py", line 634, in _update_dtr_state
    fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_DTR_str)
IOError: [Errno 6] Device not configured
```

# Alterations I _Have_ Tried

* Connecting `TX -> RX` and `RX -> TX`
* Connecting `TX -> TX` and `RX -> RX`
* Different cables

# Alterations I _Haven't_ Tried

* Using a different FTDI adapter (I can't imagine the one I have is bad)
* Using an Arduino _as_ the FTDI adapter

reloxx13, 2018-04-03:
If u plug in the ftdi, does the 2 green smds on the board flash once shortly while connecting to usb?
[image]

RX<->TX   TX<->RX always.

if u tried to flash without DOUT modus in arduino IDE, maybe the devices is damaged.

sync fail is a RX TX failure, you sid you tried another cables, but try another again.
- correct DATA Usb cable (**not one for phone charging**)
- another jumper cables between flasher and sonoff ( or check the the cables with a multimeter )

another note:

> Manually hold the male pins in place through the correct holes on the Sonoff

as soon as you do not solder them, you are leaved foryourself. its not safe that "holding" them in place is connected. you wrote by yourself:

> In all cases, the Sonoff LED never came on (flashing or otherwise). Anecdotally, if I connect only the VCC -> 3v3 and GND -> GND pins, the LED flashes for a bit, goes dark, goes solid for a bit, goes dark, etc.

which also says, the connection of vcc an gnd is bad as fuck. if vcc and GND is given, the LED should blink (if the orginal fw is not damaged till now).

you said you checked the vcc output, pls check the voltage of the VCC pin on the flasher and write it here.
low voltage cauzes the esp to crash, too.

best would be to use another pc, maybe a windows one. if you can say you can flash with the windows one, you need to find the failure on the mac side (usb hub, usb-c voltage too low, secruity issue, wrong driver or not correct installed). 
than you know your flashing hardware is working and its the OS fucking sth up.

reloxx13, 2018-04-03:
More to add:

connect the flasher with the sonoff, than hold the button down and plug in usb with the other hand.
After 2-3sec release the button.

The button could be defect too and not connection GPIO0 to gnd.
Solder cable on the btn pins and connect them before pluging in usb to be sure gpio0 is connected to gnd.
You have also to "feel" the btn press, sometimes you think its pressed but it is not!
leading to sync error while flashing, too.

reloxx13, 2018-04-03:
in your exptool description you did not use DOUT, too...

should be:
`esptool.py --port COM5 write_flash -fs 1MB -fm dout 0x0 sonoff.bin`

reloxx13, 2018-04-03:
Atom detected COM5

you used 
esptool.py --port /dev/tty.usbserial-00000000

bachya, 2018-04-03:
@reloxx13: Thanks so much for your follow up. Will respond to each point here.

>If u plug in the ftdi, does the 2 green smds on the board flash once shortly while connecting to usb?

Yes, there are 2-3 quick flashes of those LEDs when the cable is first plugged in.

>sync fail is a RX TX failure, you sid you tried another cables, but try another again.
>
>correct DATA Usb cable (not one for phone charging)
>another jumper cables between flasher and sonoff ( or check the the cables with a multimeter )

I've tried multiple Dupont cables and two different Mini-USB cables with the same result.

> as soon as you do not solder them, you are leaved foryourself. its not safe that "holding" them in place is connected. you wrote by yourself:
>
>
> which also says, the connection of vcc an gnd is bad [...]

Not sure what you mean by the first comment, but the voltage is being applied correctly (measured by multimeter). As I mentioned, the original firmware is likely gone due to the SonOTA experiment.

>you said you checked the vcc output, pls check the voltage of the VCC pin on the flasher and write it here.
>low voltage cauzes the esp to crash, too.

The voltage at the male end of the Dupont cables (with the female ends connected to the FTDI adapter) reads 3.35v.

> best would be to use another pc, maybe a windows one. if you can say you can flash with the windows one, you need to find the failure on the mac side (usb hub, usb-c voltage too low, secruity issue, [...]

That's a good point. Sadly, I don't have a Windows PC, but I'll see if I can drum one up. What about Linux?

>More to add:
>
>connect the flasher with the sonoff, than hold the button down and plug in usb with the other hand.
After 2-3sec release the button.
>
>The button could be defect too and not connecting GPIO0 to gnd.
>Solder cables on the btn pins and connect them before pluging in usb to be sure gpio0 is connected to gnd.
>You have also to "feel" the btn press, sometimes you think its pressed but it is not!
>leading to sync error while flashing, too.

I don't believe the Sonoff button is defective. This is a brand new device and prior to experimenting with SonOTA (i.e., with the original Sonoff firmware), the relay opened and closed with each button press. I can confirm that I am pressing the button straight down and hear it click.

>in your esptool description you did not use DOUT, too...
>
>should be:
>esptool.py --port COM5 write_flash -fs 1MB -fm dout 0x0 sonoff.bin
>
>https://github.com/arendst/Sonoff-Tasmota/wiki/Esptool
>
>Atom detected COM5
>
>you used
>esptool.py --port /dev/tty.usbserial-00000000
>
>i suggest to stay by atom with platformIO, its the userfriendliest solution, selects correct port and dout by itsself.

Apologies, the output from the original post had an incorrect port detected – I've updated the PlatformIO log and put it below. Note that `tty.usbserial-00000000` is the port offered:

```
|   |-- <ESP8266HTTPClient> v1.1
|   |   |-- <ESP8266WiFi> v1.0
|-- <ESP8266WiFi> v1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> v1.0
|-- <ESP8266WebServer> v1.0
|   |-- <ESP8266WiFi> v1.0

Compiling /Users/abach/Git/vsc/Sonoff-Tasmota-5.12.0/.pioenvs/sonoff/src/sonoff.ino.o
Linking /Users/abach/Git/vsc/Sonoff-Tasmota-5.12.0/.pioenvs/sonoff/firmware.elf
Checking program size
text       data     bss     dec     hex filename
445299    10668   35136  491103   77e5f /Users/abach/Git/vsc/Sonoff-Tasmota-5.12.0/.pioenvs/sonoff/firmware.elf
Building /Users/abach/Git/vsc/Sonoff-Tasmota-5.12.0/.pioenvs/sonoff/firmware.bin
Looking for upload port...
Auto-detected: tty.usbserial-00000000
Uploading /Users/abach/Git/vsc/Sonoff-Tasmota-5.12.0/.pioenvs/sonoff/firmware.bin
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error -1
 [ERROR] Took 17.86 seconds
```

Lastly, here's my `platformio.ini` (which I used in Flashing Attempt #2) – note that it uses `dout` as the flash mode:

```

[platformio]
src_dir = sonoff

; *** Uncomment one of the lines below to build/upload only one environment
;env_default = wemos-d1-mini
env_default = sonoff
;env_default = sonoff-CN
;env_default = sonoff-DE
;env_default = sonoff-ES
;env_default = sonoff-FR
;env_default = sonoff-IT
;env_default = sonoff-NL
;env_default = sonoff-PL
;env_default = sonoff-minimal
;env_default = sonoff-ds18x20

[env:sonoff]
;platform = espressif8266@1.5.0  ; v2.3.0
;platform = espressif8266@1.6.0  ; v2.4.0
platform = espressif8266
framework = arduino
board = esp01_1m
board_flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.1m0.ld -DMQTT_MAX_PACKET_SIZE=1000
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON
extra_scripts = pio/strip-floats.py

; *** Serial Monitor options
monitor_baud = 115200
```

bachya, 2018-04-04:
Ah, great catch, @Frogmore42.

I’ve flashed Tasmota onto several D1 Minis, so I’m familiar with the process.

Given your feedback, I think I’ll found another way to power the Sonoff (perhaps through a 12v barrel supply, converted down to 3.3v, off a breadboard) and just use TX and RX on the FTDI converter. I’ll report back with what I find!

bachya, 2018-04-04:
@Frogmore42 is the winner: my FTDI converter wasn't delivering enough current to power the Sonoff long enough to flash.

I thought about building a Rube Goldberg-esque contraption with a breadboard and 120v-3.3v barrel power converter, but then I remembered [this YouTube video](https://www.youtube.com/watch?v=qqSLwK1DP8Q) and built a simple FTDI converter (with amble current) from a spare Arduino UNO:

[image]

These were my steps:

1. Held down the Sonoff button.
2. Powered on the Arduino.
3. Let go of the button after about 2 seconds.
4. Successfully flashed via PlatformIO

And that's that! Thanks, all, for your help!

closed 2018-04-04 by bachya

i-me-mine, 2018-04-05:
i just got a batch of 4...  i thought they would flash just like the previous four, but there has been some variety of hardware change...  in the arduino settings options I had to set to dout, and in the software I removed all references to "Serial"...  everything works perfectly after that...  can't tell you why the serial monitor was causing the problem, but that is what it turned out to be...

McAllstar, 2018-04-23:
don't put the baud rate in. 
 esptool.py --port /dev/tty.usbserial-00000000 write_flash 0x0  flash_size=detect ~/Git/vsc/Sonoff-Tasmota-5.12.0/.pioenvs/sonoff/firmware.bin
esptool.py v2.3.1

ubeaut, 2018-05-05:
I just plug the pins temporarily for flashing in the holes and it works but the last batch of sonoff basics I received have the holes soldered up now for some unknown reason so I spent time de-soldering them and most worked except one and I had to solder header pins on that one. Why would they start soldering up the holes?

ubeaut, 2018-05-05:
> @ubeaut <https://github.com/ubeaut> Me too! Latest batch had the GPIO
> pins soldered in and required me to heat them up and suck the solder out.
> No way to prove this, but I wonder if they’re trying to prevent what we’re
> doing (flashing over their firmware).

If they do their sales will drop......dramatically!

al4085, 2018-06-04:
Same here ... I've got about 25 sonoff ( 15 Basic, 2 Th, 2 4Channel..., 3 Touch) and could easily flash all of them using an Arduino IDE...
Just received 3 Basic directly from itead
-1. Had to remove the solder from the holes.;;; tricky
-2. Could flash only 1 of the 3... unclear why the other 2 did not (poor contact of the pins ?) They dont even blink under 3.3V tension. I will soon try under 240V power 
-3. The one that did flash does not start / blink / send anything to serial

Questions for the team here:
- I've read that some Sonoff Basic come with 8285 on ..; how do I check for that ?
- How can I check that my pins are well connected to the board on the other two ?  
- I'm interested in purchasing spare older versions if indded these cannot be falshed anymore ;-) :-(( 
- Any way to give feedback to Itead  ?

MYLE-01, 2018-06-04:
My 5 cents
 I use Arduino and make the bin file with all my setting in it 

PLUG my FTD1232 into it Then I press the black button plug it into the USB into computer count to 10 take finger off button then I load FlashESP8266.exe  select the com port them select the bin which in the same dir as FlashESP8266.exe  and hit flash  done Have done for the last 6 sonoff and it always work for me

al4085, 2018-06-06:
Further (imperfect) progress
 I managed to flash 1 of the basics by putting the DOUT on. 
It works perfectly on FTDI, BUT doesnt when on 240V ??? Strange : has anyone encountered this ? Can I connect both 240 V AND FTDI to see the serial monitor or is it dangerous ?

ascillato, 2018-06-06:
It is VERY DANGEROUS!

**_DO NOT CONNECT FTDI WHILE MAINS IS ALSO CONNECTED!!!!_**

**_YOU CAN BE ELECTROCUTED!!!!_** 

FTDI provide the 3.3v needed for the esp8266

If your FTDI can not provide enough power, please use an extern 3.3v supply

xlcnd, 2018-11-14:
After much frustration, I was able to flash several of the new 'sonoff basic' (with date 2017-05-05 on board), by following these steps:
1. `pip install esptool`
2. `platformio --run`  (to make the firmware).
3. Insert the cables from the FTDI in the right Sonoff's pins!
4. Held down the Sonoff button.
5. Plug the usb cable (with the FTDI) to the computer.
6. Let go off the Sonoff button after about 3 seconds.
7. `esptool.py --port /dev/ttyUSB0 write_flash 0x0 .pioenvs/sonoff-basic/firmware.bin`

SUCCESS!

I wasn't able to flash them directly with Arduino or platformio!

gulliverrr, 2018-12-31:
Every now and then I get a new batch of various Sonoff devices and my previously working system does not flash them due to various reasons not directly controlled by me. I have a long txt file with black magic tricks to remind myself.
Today after numerous attempts I added this (with sudo in front)
sudo pip install esptool

And it worked! BTW I use an external 3.3V supply as I don't trust those FTDI adapters.

tedsluis, 2019-01-03:
I managed to flash my new Sonoff Basic RF R2 Power switches using vscode on a Fdora linux computer and a CH340G USB board.  

At first I followed the instructions on https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Basic, but I always ended with this same error message:
```
Compiling .pioenvs/sonoff-basic/src/sonoff.ino.cpp.o
Linking .pioenvs/sonoff-basic/firmware.elf
Retrieving maximum program size .pioenvs/sonoff-basic/firmware.elf
Building .pioenvs/sonoff-basic/firmware.bin
Checking size .pioenvs/sonoff-basic/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [======    ]  55.4% (used 45380 bytes from 81920 bytes)
PROGRAM: [====      ]  43.9% (used 449140 bytes from 1023984 bytes)
Configuring upload protocol...
Looking for upload port...
Use manually specified: /dev/ttyUSB0
Uploading .pioenvs/sonoff-basic/firmware.bin
error: cannot access /dev/ttyUSB0

error: espcomm_open failed
error: espcomm_upload_mem failed
```
 I found out that my vscode session (runs as a normal Linux user without root privileges) did not have the proper permissions to connect to the /dev/ttyUSB0 device. So I tied this:

```
[root@lenovo ~]# ls -lha /dev/ttyUSB*
crw-rw----. 1 root dialout 188, 0  3 jan 07:35 /dev/ttyUSB0
[root@lenovo ~]# chmod 777 /dev/ttyUSB0
[root@lenovo ~]# ls -lha /dev/ttyUSB*
crwxrwxrwx. 1 root dialout 188, 0  3 jan 07:36 /dev/ttyUSB0
```
I still did not work. I found out that the permissions got lost when I connected the 3v3 wire. The final solution was to grant the permissions again after connecting the 3v3 wire.

My Sonoff Basic RF R2 Power: 
[image] 

[image] 

The preparations I took to flash successfully:
 
* install platformIO plugin in vscode
* solder a 4 pin socket on the sonoff board
* connect the sonoff with the CH340G: GND-GND, Txd-RxD, RxD-TxD,  3v3-3v3
* download and unzipped [Sonoff-Tasmota Source code](https://github.com/arendst/Sonoff-Tasmota/releases) and opened it in vscode
* edit [platformio.ini](https://github.com/arendst/Sonoff-Tasmota/blob/master/platformio.ini)  and select the sonoff-basic environment and configure  `upload_port  = /dev/ttyUSB0`
* edit [my_user_config.h](https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/my_user_config.h) and configure WFI ssid/password, ip address, gateway, dns, etc

The steps I took to flash successfully: 

* disconnect the CH340G usb from the computer
* connect all the wires, except the 3v3. 
* press and hold down the button on the sonoff board.
* insert the CG340G USB into the computer
* execute: `sudo chmod 777 /dev/ttyUSB0`
* connect the 3v3
* again execute: `sudo chmod 777 /dev/ttyUSB0`
* let go the button on the sonoff board 
* wait 3 seconds
* Selected "Upload" from the menu to flash the firmware.

```
Compiling .pioenvs/sonoff-basic/src/sonoff.ino.cpp.o
Linking .pioenvs/sonoff-basic/firmware.elf
Retrieving maximum program size .pioenvs/sonoff-basic/firmware.elf
Building .pioenvs/sonoff-basic/firmware.bin
Checking size .pioenvs/sonoff-basic/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [======    ]  55.4% (used 45380 bytes from 81920 bytes)
PROGRAM: [====      ]  43.9% (used 449140 bytes from 1023984 bytes)
Configuring upload protocol...
Looking for upload port...
Use manually specified: /dev/ttyUSB0
Uploading .pioenvs/sonoff-basic/firmware.bin
Uploading 453280 bytes from .pioenvs/sonoff-basic/firmware.bin to flash at 0x00000000
................................................................................ [ 18% ]
................................................................................ [ 36% ]
................................................................................ [ 54% ]
................................................................................ [ 72% ]
................................................................................ [ 90% ]
...........................................                                      [ 100% ]
==== [SUCCESS] Took 57.81 seconds =====
```
 
My CH340G USB:
[image] 

[image] 

Cable connected to the sonoff:
[image] 

May it is useful for other to mention this /dev/ttyUSB0 permissions solution in the documentation somewhere?

Question: How can I resolve the  /dev/ttyUSB0 permissions issue permanent, without the need to apply the permissions every time I reconnect to a Sonoff using the USB CH340G?

ubeaut, 2019-01-03:
To make the /dev/ttyUSB0 permissions permanent.
Put your user in the dialout group.
vi /etc/group
dialout:x:20:youruserid
save the group file
log out and log back in
type id and hit returnYOu should see you are now in the dialout group

mtnbkr88, 2019-01-16:
I am sooo frustrated. I recently received six of these Sonoff Basic switches with the Sonoff RF R2 Power v1.0 boards as shown in tedsluis pictures above (they have the ESP8285 chip) and I have the same FTDI USB-TTL converter shown in the OPs (bachya) very first post. I am powering the Sonoff board using a separate variable power supply with a common ground with the FTDI converter like described in previous posts. I'm using a Windows 10 PC with the latest ESP.Easy.Flasher.exe version 0.03.002 and software from the ESPEasy developers. I know my wires are good. No matter what combinations to boot the boards into programming mode I try, they never reply to data transmits. When I'm holding the black button down on power up, the board pulls 70ma.  After I let go the black button, power drops to 30ma. Here is the output from the log:

######2019-01-16######
#######0.03.002#######
######FLASH INFO######
BIN file: ESP_Easy_mega-20181216_normal_ESP8285_1024.bin
COM port: (COM8) USB Serial Port
Baud rate: 115200
! FORCED DOUT ENABLED !
######POST FLASH######
No post flash information entered...
######FLASH LOG######
[esptool.exe -vv -bm dout -cd nodemcu -cb 115200 -cp COM8 -ca 0x00000 -cf "C:\Tmp\ESPEasy_2xx\bin\ESP_Easy_mega-20181216_normal_ESP8285_1024.bin"]
[1/16/2019 11:26:31 AM] esptool v0.4.12 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
[1/16/2019 11:26:31 AM] setting flash mode from qio to dout
[1/16/2019 11:26:31 AM] 	setting board to nodemcu
[1/16/2019 11:26:31 AM] 	setting baudrate from 115200 to 115200
[1/16/2019 11:26:31 AM] 	setting port from  to COM8
[1/16/2019 11:26:31 AM] 	setting address from 0x00000000 to 0x00000000
[1/16/2019 11:26:31 AM] 	espcomm_upload_file
[1/16/2019 11:26:31 AM] 	espcomm_upload_mem
[1/16/2019 11:26:31 AM] 	setting serial port timeouts to 1000 ms
[1/16/2019 11:26:31 AM] opening bootloader
[1/16/2019 11:26:31 AM] resetting board
[1/16/2019 11:26:31 AM] trying to connect
[1/16/2019 11:26:31 AM] 	flush start
[1/16/2019 11:26:31 AM] 	setting serial port timeouts to 1 ms
[1/16/2019 11:26:31 AM] 	setting serial port timeouts to 1000 ms
[1/16/2019 11:26:31 AM] 	flush complete
[1/16/2019 11:26:31 AM] 	espcomm_send_command: sending command header
[1/16/2019 11:26:31 AM] 	espcomm_send_command: sending command payload
[1/16/2019 11:26:31 AM] 	read 0, requested 1
[1/16/2019 11:26:31 AM] trying to connect
[1/16/2019 11:26:31 AM] 	flush start
[1/16/2019 11:26:31 AM] 	setting serial port timeouts to 1 ms
[1/16/2019 11:26:31 AM] 	setting serial port timeouts to 1000 ms
[1/16/2019 11:26:31 AM] 	flush complete
[1/16/2019 11:26:31 AM] 	espcomm_send_command: sending command header
[1/16/2019 11:26:31 AM] 	espcomm_send_command: sending command payload
[1/16/2019 11:26:31 AM] 	read 0, requested 1
[1/16/2019 11:26:31 AM] trying to connect
[1/16/2019 11:26:31 AM] 	flush start
[1/16/2019 11:26:31 AM] 	setting serial port timeouts to 1 ms
[1/16/2019 11:26:31 AM] 	setting serial port timeouts to 1000 ms
[1/16/2019 11:26:31 AM] 	flush complete
[1/16/2019 11:26:31 AM] 	espcomm_send_command: sending command header
[1/16/2019 11:26:31 AM] 	espcomm_send_command: sending command payload
[1/16/2019 11:26:31 AM] 	read 0, requested 1
[1/16/2019 11:26:31 AM] resetting board
[1/16/2019 11:26:31 AM] trying to connect
[1/16/2019 11:26:31 AM] 	flush start
[1/16/2019 11:26:31 AM] 	setting serial port timeouts to 1 ms
[1/16/2019 11:26:31 AM] 	setting serial port timeouts to 1000 ms
[1/16/2019 11:26:31 AM] 	flush complete
[1/16/2019 11:26:31 AM] 	espcomm_send_command: sending command header
[1/16/2019 11:26:31 AM] 	espcomm_send_command: sending command payload
[1/16/2019 11:26:31 AM] 	read 0, requested 1
[1/16/2019 11:26:31 AM] trying to connect
[1/16/2019 11:26:31 AM] 	flush start
[1/16/2019 11:26:31 AM] 	setting serial port timeouts to 1 ms
[1/16/2019 11:26:31 AM] 	setting serial port timeouts to 1000 ms
[1/16/2019 11:26:31 AM] 	flush complete
[1/16/2019 11:26:31 AM] 	espcomm_send_command: sending command header
[1/16/2019 11:26:31 AM] 	espcomm_send_command: sending command payload
[1/16/2019 11:26:31 AM] 	read 0, requested 1
[1/16/2019 11:26:31 AM] trying to connect
[1/16/2019 11:26:31 AM] 	flush start
[1/16/2019 11:26:31 AM] 	setting serial port timeouts to 1 ms
[1/16/2019 11:26:31 AM] 	setting serial port timeouts to 1000 ms
[1/16/2019 11:26:31 AM] 	flush complete
[... 11 lines cut ...]
[1/16/2019 11:26:31 AM] 	read 0, requested 1
[1/16/2019 11:26:31 AM] trying to connect
[1/16/2019 11:26:31 AM] 	flush start
[1/16/2019 11:26:31 AM] 	setting serial port timeouts to 1 ms
[1/16/2019 11:26:31 AM] 	setting serial port timeouts to 1000 ms
[1/16/2019 11:26:31 AM] 	flush complete
[1/16/2019 11:26:31 AM] 	espcomm_send_command: sending command header
[1/16/2019 11:26:31 AM] 	espcomm_send_command: sending command payload
[1/16/2019 11:26:31 AM] 	read 0, requested 1
[1/16/2019 11:26:31 AM] trying to connect
[1/16/2019 11:26:31 AM] 	flush start
[1/16/2019 11:26:31 AM] 	setting serial port timeouts to 1 ms
[1/16/2019 11:26:31 AM] 	setting serial port timeouts to 1000 ms
[1/16/2019 11:26:31 AM] 	flush complete
[1/16/2019 11:26:31 AM] 	espcomm_send_command: sending command header
[1/16/2019 11:26:31 AM] 	espcomm_send_command: sending command payload
[1/16/2019 11:26:31 AM] 	read 0, requested 1
[1/16/2019 11:26:31 AM] warning: espcomm_sync failed
[1/16/2019 11:26:31 AM] error: espcomm_open failed
[1/16/2019 11:26:31 AM] error: espcomm_upload_mem failed
[2019-01-16 11:26:31] STOPPED due to 2 errors! (try reset on the unit, then start a new flash attempt)

I've tried all I can think of with the tools I have, even switching TX and RX. I'm at my wits end. I don't want to build any Linux systems or start messing with Arduinos. The ESPEasy flasher tool provided by the ESPEasy developers should be enough. At this point I'm ready to ship these six boards to somebody and pay them to put ESPEasy on them.

Can anybody help me? Thanks.

gulliverrr, 2019-01-16:
@mtnbkr88 I don't know ESPEasy and I don't know windows. If you want Tasmota I can give it a try if you really find no other solution. Im in the EU. Contact URL on my profile

mtnbkr88, 2019-01-16:
gulliverrr, thanks for the reply. I'm in the US and I'm not familiar with Tasmota so I'm still hoping for an ESPEasy solution.

Thanks.

ascillato, 2019-01-16:
@mtnbkr88 do you know that this is Tasmota repository right? This is not ESPEasy. You are posting in Tasmota repository. 

The fact that you are in the US don't affect Tasmota. You can use it as a lot of other people in the US and many other people in others countries are doing. 

If you need help with Tasmota, please read the wiki and then find us in the Tasmota Support Chat. Link in the Tasmota readme (main page of this repository)

meingraham, 2019-01-16:
You really ought to strongly consider using TASMOTA.

That aside, if you have a binary (regardless of what firmware it is) you can use ESPEasy Mega to upload the binary to the device. I find the method described in [Dr. Zzs tutorial](https://www.youtube.com/watch?v=KMiP9Ku71To) pretty foolproof.

Mike

mtnbkr88, 2019-01-17:
FYI, I finally got flashing to work. Turns out I needed to downgrade my FTDI driver to 2.10.0.0. Thanks all.

tedder, 2019-05-02:
For posteriety (eg future googlers), I spent an hour or two on this, even replaced a device in case I let the smoke out of it with an accidental 5V connection and then an intentional 5V connection to a Sonoff Pow R2, as I thought maybe my FTDI wasn't providing enough power.

.. it wasn't providing enough power.

After getting a [Sparkfun Beefy 3](https://www.sparkfun.com/products/13746), it worked on the first try. I should have done it sooner and saved myself time.

lalo-uy, 2019-05-02:
A simple 100uF cap between 3.3V & GND helps a lot with FTDI.

tedder, 2019-05-02:
That may be, but if you're supplying power to a Sonoff through a FTDI, it's very likely to not have enough power, as others in this thread have already noted. For instance, the [FT232R](https://www.ftdichip.com/Support/Documents/DataSheets/ICs/DS_FT232R.pdf)'s built-in LDO supplies "up to 50mA". That's enough to run an 8266 at idle, but certainly not when transmitting.

meingraham, 2019-05-02:
Alas, increasing voltage doesn't necessarily mean more available power to the device. Usually available power (Watts) available from the FTDI is "capped". Increasing the voltage just means less current and vice versa. A CH340 based serial adapter will improve the available power situation.

tedder, 2019-05-02:
Yep, voltage was a mistake. It was a hope that I'd be (a) grabbing it from the USB power, not through the chip and (b) it wouldn't let any magic smoke out.

e8geny, 2019-05-08:
I had the same problem. I programmed 12 sonoffs and only a half of them had good connection to COM port. I tried a lot, that was written at forums. But the solution i found was - **to solder wires directly from USB-TTL card into Sonoff board**!!! **Do not use any connectors.** May be there is resistance in plastic case of connector, or may be there is bad conact.

stuzx61978, 2019-12-23:
> I had the same problem. I programmed 12 sonoffs and only a half of them had good connection to COM port. I tried a lot, that was written at forums. But the solution i found was - to solder wires [...]

I,ve just signed up. I am a complete newbie to sonoff and just wanted to say that after a day of trying various things to flash my sonoff , this turned out to be my issue too.
I replaced the cables for the tx and rx and it worked first time. 
I want to use to control my garage door. I have set gpio3 to a switch, and I plan to use this as a status switch, which is my next mission.. lol

moelfus, 2020-01-30:
I was also struggeling to flash a Sonoff Basic. I was using a cheap CH340 adaptor and connected the board with the 3.3V pin. (TX->RX, RX->TX, button pushed when powering up the device, first erase and then upload of the firmware) I used sveral tools for flashing, for example tasmotizer and pyflasher and the upload was even sometimes confirmed to be successfull, however the sonoff didn't do anything. In the end using the esptool.py helped. It showed that the MD5 checksum wasn't right. I then connected the sonoff board with the vcc pin instead of the 3.3V and it worked like a charm. Maybe the CH340 wasn't powerful enough or the cables were too long the mulitmeter showed less than 3V, and thus, using VCC (5V) did help in the end. Just wanted to let you guys know in case you are also struggeling

lalo-uy, 2020-01-30:
Never connect sonnov Vcc to 5V. unless your CH340 was 3.3v device or was
wrong, 5V on sonoff vcc will destroy the esp chip

moelfus, 2020-02-06:
@lalo-uy you are completly right! My mistake, I'm using one of the cheap CH340 which have three pins, 3V, 5V, and VCC. I used a multimeter and VCC delivers ~3.7 V. The device delivers enough power to flash a sonoff basic and sonoff touch, however is not sufficent enough for an S26

cicciopizza, 2020-04-16:
I also had several problems. In the end I realized that you need to have well-fixed connectors:
Initially I had simply 'placed' the plug on the sonoff WITHOUT soldering them and it didn't work.
Then I tried to hold them tight with my hand making the contact do well and it worked.
So in the end you have to make sure that the contacts on the sonoff are well made