arendst/Tasmota #2345: Add option for pulstime

mpuff, 2018-04-03:
Hello.

Is it possible to add an option to send an mqtt off command when the pulsetime command is set.
Because when the pulstime option set the item to the off state, in the mqtt broker is only the on command as last state, when the homeautomation software do a restart, it will check the last state of the item in the mqtt broker, and the item gets on again. so a motor will start again ore something else.

curzon01, 2018-04-03:
homeautomation should check the status topic, not the command topic

PulseTime20
```
18:03:12 MQT: stat/tasmota/esp08/RESULT = {"PulseTime1":"20 (Active 0)"}
```
Power1 ON
```
18:03:17 MQT: stat/tasmota/phesp08/RESULT = {"POWER1":"ON"}
18:03:17 MQT: stat/tasmota/phesp08/POWER1 = ON
18:03:19 MQT: stat/tasmota/phesp08/RESULT = {"POWER1":"OFF"}
18:03:19 MQT: stat/tasmota/phesp08/POWER1 = OFF
```

mpuff, 2018-04-03:
I have tasmota on an wemosd1 mini, i have checked now, i do not have an mqtt stat with POWER, only result messages from RESULT1 - RESULT11
In RESULT11 i have the message that the POWER state is OFF, but as mqtt command i have in the broker always ON

curzon01, 2018-04-03:
> Because i only send an ON command with the homeautomation software, not the status OFF

I did the same manual, doesn't matter if using web gui or MQTT message

> I think the broker does set the last command after a restart

Stop homeautomation sending retain messages, this always makes a lot of trouble. Also delete all current related retain message from broker.

- [ ] Provide the output of command ``status 0``

mpuff, 2018-04-03:
Ok , thank you for your help.
I will change the retain to 0 (off)

closed 2018-04-03 by mpuff