arendst/Tasmota #2346: sonoff rf basic learn mode

coolbombom, 2018-04-03:
Hi guys

Thank u for a great firmware ... i was wondering is there any way to make the sonoff rf basic enter rf learn mode from the web console?

I have looked at the wiki commands and could not find an answer!

Kind regards
c_bb

ErikAndren, 2018-04-16:
Are you talking about the sonoff rf bridge or basic?
Only the bridge supports RF commands.

https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#sonoff-rf-bridge-433:

RfKey<x> |   | Send learned or default RF data for RfKey1 to RfKey16
RfKey<x> | 1 | Send default RF data for RfKey1 to RfKey16 using RfSync, RfLow, RfHigh and RfHost parameters
**RfKey<x> | 2 | Learn RF data for RfKey1 to RfKey16**
RfKey<x> | 3 | Unlearn RF data for RfKey1 to RfKey16
RfKey<x> | 4 | Save RF data using RfSync, RfLow, RfHigh and last RfCode parameters
RfKey<x> | 5 | Show default or learnt RF data

stoinov, 2018-05-05:
So the RF basic is only able to learn when you physically press the button twice?

closed 2018-07-04 by stale[bot]