arendst/Tasmota #2347: XXL Build Error

reloxx13, 2018-04-03:
If 

user_config.h
 

       // #define USE_MHZ19                                // Add support for MH-Z19 CO2 sensor (+2k code)
        // #define USE_SENSEAIR                             // Add support for SenseAir K30, K70 and S8 CO2 sensor (+2k3 code)

is commented out, the XXL Build fails with

    C:/xampp/htdocs/Sonoff-Tasmota/sonoff/xsns_15_mhz19.ino: In function 'void MhzShow(boolean)':
    C:/xampp/htdocs/Sonoff-Tasmota/sonoff/xsns_15_mhz19.ino:543:46: error: 'HTTP_SNS_CO2' was not declared in this scope
    C:/xampp/htdocs/Sonoff-Tasmota/sonoff/xsns_17_senseair.ino: In function 'void SenseairShow(boolean)':
    C:/xampp/htdocs/Sonoff-Tasmota/sonoff/xsns_17_senseair.ino:439:46: error: 'HTTP_SNS_CO2' was not declared in this scope
    
    Compiling .pioenvs\sonoff-xxl\libca4\ESP8266WebServer\Parsing.cpp.o
    *** [.pioenvs\sonoff-xxl\src\sonoff.ino.cpp.o] Error 1

In the sonoff_post.h those variables are defined if its the xxl build.
sonoff_post.h

        #ifdef USE_ALL_SENSORS  // ===================== Configure sonoff-xxl.bin =========================
    ...
        #define USE_MHZ19                             // Add support for MH-Z19 CO2 sensor (+2k code)
        #define USE_SENSEAIR                          // Add support for SenseAir K30, K70 and S8 CO2 sensor (+2k3 code)
    ...
        #endif  // USE_ALL_SENSORS =====================

guess its including too late.

sonoff.ino
   

     #include "i18n.h"                           // Language support configured by user_config.h
        #include "sonoff_template.h"                // Hardware configuration
        #include "sonoff_post.h"                    // Configuration overrides for all previous includes

in the i18n.h the variable HTTP_SNS_CO2 will never ever be defined if using the xxl build

i18n.h
  

      #if defined(USE_MHZ19) || defined(USE_SENSEAIR)
        const char HTTP_SNS_CO2[] PROGMEM = "%s{s}%s " D_CO2 "{m}%d " D_UNIT_PARTS_PER_MILLION "{e}";                // {s} = <tr><th>, {m} = </th><td>, {e} = </td></tr>
        #endif  // USE_WEBSERVER

//low prio for me, ill just comment out the whole xxl environment block.

reloxx13, 2018-04-03:
Short Solution:
add  || defined(USE_ALL_SENSORS)

i18n.h

    #if defined(USE_MHZ19) || defined(USE_SENSEAIR) || defined(USE_ALL_SENSORS)
    const char HTTP_SNS_CO2[] PROGMEM = "%s{s}%s " D_CO2 "{m}%d " D_UNIT_PARTS_PER_MILLION "{e}";                // {s} = <tr><th>, {m} = </th><td>, {e} = </td></tr>
    #endif  // USE_WEBSERVER

arendst, 2018-04-04:
Thnx for testing 👍 

I suggest to move include sonoff_post.h in sonoff.ino up to line 37 just below the user_config_override.h define test.

arendst, 2018-04-05:
Should be fixed with release 5.12.0j.

closed 2018-04-05 by arendst