arendst/Tasmota #2353: state MQTT topic not changing when using the MQTT command only

jraydoran, 2018-04-04:
Hardware: Sonoff Basic
Firmware Version: 5.12.0
Topic: sonoff01
Full Topic: tasmota/%topic%/%prefix%/

There is no state change on topic:
tasmota/sonoff01/stat/POWER

only when using MQTT command "ON" and "OFF" on topic:
tasmota/sonoff01/cmnd/POWER

The actual load is being switched 100%

When using the manual switch or using the web interface to control the load, the state topic changes as expected.

I am monitoring the messages using MQTT client "MQTT.fx version 1.5.2

Will be using openhab eventually.

Thank you!

Kind Regards
Jason

jraydoran, 2018-04-04:
When monitoring the console. I see the state change on

topic:
stat/sonoff01/POWER

not on 'tasmota/sonoff01/stat/POWER' as expected

Once again, this only happens with using MQTT command.

amercer86, 2018-04-05:
I'm having the same issue with MQTT commands-

When I subscribe to all topics on my broker, I see this payload from the sonoff:

stat/sonoff/POWER:"ON"

It, of course, toggles to "OFF" when I press the button. I see the same messages on the console. The topic doesn't match my user_conf.h.

curzon01, 2018-04-05:
Do not play with MqttClient ID, leave MqttClient as is:
`MqttClient DVES_%06X`

it's only a technical name to identify a client on MQTT server and typically not neccessary to change it or make it human readable - see also #2337, #2291, #1893

jraydoran, 2018-04-05:
@curzon,

Thank you so much for you curtious reply!
This did in-fact resolve my issue. Appreciate it ALOT!!

Kind Regards
Jason

amercer86, 2018-04-06:
I haven 't changed my MqttClient ID:
`#define MQTT_CLIENT_ID         "DVES_%06X"  `

Other bits of user_config.h:
```
#define PROJECT                "Lamp"   
#define MQTT_TOPIC             PROJECT  
#define MQTT_FULLTOPIC         "tasmota/bedroom/%topic%/%prefix%/"
#define PUB_PREFIX             "TEST" 
```
Still, the topic coming through MQTT is: stat/sonoff/POWER

amercer86, 2018-04-08:
Sorry for the delay.

I started tinkering a bit more this morning and turns out I was having this issue described in the Troubleshooting wiki. I was able to resolve by pushing the reset command over MQTT.

> Config problems (can cause boot loops, items set in user_config to not appear, etc)
> By default, this firmware tries to preserve the existing config (to support automated updates via OTA upgrades), but various things can happen that cause the existing config to be a problem.
> 
> When you change the settings in the code, that doesn't directly change the settings on the running machine when you load it. What happens is that when it
> boots up, the firmware looks to see if it has a valid config (is it an upgrade
> to an older tasmota version), and if the CFG_HOLDER value is in the right place,it assumes that the existing config is valid. If it doesn't find the right value, it assumes that this is not an upgrade and takes the compiled-in config
> data and writes it out to the config area.
> 
> There are multiple ways to force the config to what's set in user_config to recover a system.
> 
> hold button1 down for 4 seconds
> issue a reset command (reset 1 via the web console, /cmnd/sonoff/reset 1 via mqtt)
> change the value of CFG_HOLDER in user_config and re-flash the device (CFG_HOLDER is a 0xYYYYDDMM setup. Just change it to today. Any future updates should not be hindered by that).

jraydoran, 2018-04-09:
Thanks for sharing!

closed 2018-04-09 by jraydoran